PX4/PX4-Autopilot #16229: Dumped GPS communication does not get logged

jbeyerstedt, 2020-11-19:
**Describe the bug**
When enabling the parameter `GPS_DUMP_COMM`, the raw communication with the GPS module is dumped to the uORB, but not written to the log file. I would expect, that the log is enabled automatically.

**To Reproduce**
Steps to reproduce the behavior:
1. Set `GPS_DUMP_COMM` to true
2. Turn on the drone and wait for a GPS fix
3. Arm (to start the log) and wait for a few seconds
4. Disarm and analyze the logged topics (e.g. with pyulog's ulog_info command)

**Expected behavior**
When `GPS_DUMP_COMM` is enabled, the dumped data should also be written to the log file.

dk7xe, 2020-11-19:
Indeed, that is the expectet behaviour to have the data in the log file at the end.

bkueng, 2020-11-20:
See https://github.com/PX4/PX4-Autopilot/pull/16247

jbeyerstedt, 2020-11-20:
Thanks, looks fine for me.
By the way: `transponder_report` (for UTM_GLOBAL_POSITION and ADS-B messages) is also not logged anywhere.

closed 2020-11-20 by dagar

kaklik, 2020-11-20:
Hi!  
I want to test this feature, but the format of the logged GPS_DUMP_COMM messages is unclear. There are many gps_dump/data.X topics without obvious explanation and meaning.

[image]

Is it something like NMEA or ubx messages enclosed in uOrb topics?

jbeyerstedt, 2020-11-20:
The messages just contain a dump of the raw (binary) UART communication. Depending on the GPS receiver, it might be NMEA or a custom protocol (e.g. when using an u-blox receiver).

You can recover the binary data stream with [pyulog](https://github.com/PX4/pyulog/blob/master/pyulog/extract_gps_dump.py) if you like. But it's really just the bare UART communication encapsulated in ulog data points.
I adapted the script linked above to get the individual ublox protocol message from our GPS receiver (Pixhawk 4 GPS). I realized, that it's really not even assured, that one ulog message just contains one UART message. Each ulog message is just filled with a bunch of bytes and the data stream just continues in the next message.

dk7xe, 2020-11-21:
Thank you @bkueng and @dagar !