PX4/PX4-Autopilot #16230: Regression: pixhawk 4 out of service on master

MaEtUgR, 2020-11-19:
**Describe the bug**
Regression: Flashing the IO chip fails on a pixhawk 4 flashed with latest master.

**Found by @priseborough** 🙏 
FYI @dagar / @davids5 

**To Reproduce**
Steps to reproduce the behavior:
1. Fetch latest master f18e70029ab873fca52c950eb860009c05e3c03e
2. Flash a pixhawk 4 `make fmu-v5 upload`
3. When it boots hear fast beeping all the time, IO pins don't come up
4. MAVLink shell
[image]

**Expected behavior**
Flashes IO chip, stopps beeping and IO pins come up.

**Log Files and Screenshots**
https://photos.app.goo.gl/YB4pTSUkK7eiz3FN7

**Drone (please complete the following information):**
- Reproducable with bare board hooked up over USB

**Additional context**
Flashing back stable 1.11.2 recovers it all.

priseborough, 2020-11-19:
For reference, when I also encountered this behaviour, the relevant airframe parameters in use at the time were:
 
[image]

MaEtUgR, 2020-11-19:
I have another clue: dmesg repeatedly shows "px4io bad sync" 💡 
<details><summary>dmesg output</summary>
<p>

#### yes, even hidden code blocks!

[image]

</p>
</details>

davids5, 2020-11-19:
I see the issue......

davids5, 2020-11-19:
I suspected the timing had changed...but it was not that simple.

closed 2020-11-19 by dagar

MaEtUgR, 2020-11-19:
Thanks @davids5 !