PX4/PX4-Autopilot #16232: Make px4_sitl gazebo crashes whole system.

LightningPORTO, 2020-11-19:
After a clean installation, with all the dependencies and submodules, the PX4 SITL Gazebo make is crashing the hole system.

**To Reproduce**
Steps to reproduce the behavior:
1. Clean installation;
2. Download current branch & dependencies
3. run "make px4_sitl gazebo"
4. See error

**Expected behavior**
Successful make of SITL gazebo

Jaeyoung-Lim, 2020-11-19:
@LightningPORTO Have you looked at your memory? Probably you don't have enough RAM?

julianoes, 2020-11-19:
Try `make px4_sitl gazebo -j2`

LightningPORTO, 2020-11-19:
> @LightningPORTO Have you looked at your memory? Probably you don't have enough RAM?

I have 16 Gb, it goes wild when compiling. Im using htop to check it

LightningPORTO, 2020-11-19:
> make px4_sitl gazebo -j2

Same problem

Jaeyoung-Lim, 2020-11-20:
> 4. See error

Can you share the error?

LightningPORTO, 2020-11-30:
There is no error, it just freezes the PC

dagar, 2020-11-30:
@LightningPORTO try adding `-- -j1` here. https://github.com/PX4/PX4-Autopilot/blob/aa80167b476957f546a33b4626752958e9d30518/platforms/posix/cmake/sitl_target.cmake#L48

LightningPORTO, 2020-12-03:
Still doesnt fix it. Im closing this issue, but the error is still there. Something to to with the gazebo Sitl submodules

closed 2020-12-03 by LightningPORTO