PX4/PX4-Autopilot #16242: Gazebo SITL : fatal error: google/protobuf/port_def.inc: No such file or directory

FlavioTonelli, 2020-11-19:
I'm trying to run a gazebo SITL simulation over my Ubuntu 20.04 workstation.
The px4 repo I'm using is the current master one.
I set up the environment running the dedicated ubuntu.sh script (Tools/setup/ubuntu.sh).

When I do make px4_sitl gazebo I have a ProtoBuf error.
Here is the output of "make px4_sitl gazebo | grep ProtoBuf" :

> CMake Warning at /usr/share/cmake-3.16/Modules/FindProtobuf.cmake:499 (message):
>   Protobuf compiler version 3.13.0 doesn't match library version 3.6.1
> 
> Call Stack (most recent call first):
>   /usr/lib/x86_64-linux-gnu/cmake/gazebo/gazebo-config.cmake:166 (find_package)
>   CMakeLists.txt:47 (find_package)
> 
> -- Found Protobuf: /usr/lib/x86_64-linux-gnu/libprotobuf.so;-lpthread (found version "3.6.1") 
> CMake Warning at /usr/share/cmake-3.16/Modules/FindProtobuf.cmake:499 (message):
>   Protobuf compiler version 3.13.0 doesn't match library version 3.6.1
> Call Stack (most recent call first):
>   /usr/share/cmake/ignition-cmake2/cmake2/FindIgnProtobuf.cmake:34 (find_package)
>   /usr/lib/x86_64-linux-gnu/cmake/ignition-transport8/ignition-transport8-config.cmake:92 (find_package)
>   /usr/lib/x86_64-linux-gnu/cmake/gazebo/gazebo-config.cmake:198 (find_package)
>   CMakeLists.txt:47 (find_package)
> 
> 
> -- Found Protobuf: /usr/lib/x86_64-linux-gnu/libprotobuf.so;-lpthread (found suitable version "3.6.1", minimum required is "3") 
> CMake Warning at /usr/share/cmake-3.16/Modules/FindProtobuf.cmake:499 (message):
>   Protobuf compiler version 3.13.0 doesn't match library version 3.6.1
> Call Stack (most recent call first):
>   /usr/share/cmake/ignition-cmake2/cmake2/FindIgnProtobuf.cmake:34 (find_package)
>   /usr/lib/x86_64-linux-gnu/cmake/ignition-msgs5/ignition-msgs5-config.cmake:92 (find_package)
>   /usr/lib/x86_64-linux-gnu/cmake/ignition-transport8/ignition-transport8-config.cmake:96 (find_package)
>   /usr/lib/x86_64-linux-gnu/cmake/gazebo/gazebo-config.cmake:198 (find_package)
>   CMakeLists.txt:47 (find_package)
> 
> 
> CMake Warning at /usr/share/cmake-3.16/Modules/FindProtobuf.cmake:499 (message):
>   Protobuf compiler version 3.13.0 doesn't match library version 3.6.1
> Call Stack (most recent call first):
>   CMakeLists.txt:137 (find_package)
> 
> 
> -- Found Protobuf: /usr/lib/x86_64-linux-gnu/libprotobuf.so;-lpthread (found version "3.6.1") 
> -- Checking for module 'protobuf'
> --   Found protobuf, version 3.6.1
> SITLGps.pb.h:10:10: fatal error: google/protobuf/port_def.inc: No such file or directory
>    10 | #include <google/protobuf/port_def.inc>
> Wind.pb.h:10:10: fatal error: google/protobuf/port_def.inc: No such file or directory
>    10 | #include <google/protobuf/port_def.inc>
> Airspeed.pb.h:10:10: fatal error: google/protobuf/port_def.inc: No such file or directory
>    10 | #include <google/protobuf/port_def.inc>
> Int32.pb.h:10:10: fatal error: google/protobuf/port_def.inc: No such file or directory
>    10 | #include <google/protobuf/port_def.inc>
> Groundtruth.pb.h:10:10: fatal error: google/protobuf/port_def.inc: No such file or directory
>    10 | #include <google/protobuf/port_def.inc>
> Float.pb.h:10:10: fatal error: google/protobuf/port_def.inc: No such file or directory
>    10 | #include <google/protobuf/port_def.inc>
> Imu.pb.h:10:10: fatal error: google/protobuf/port_def.inc: No such file or directory
>    10 | #include <google/protobuf/port_def.inc>
> IRLock.pb.h:10:10: fatal error: google/protobuf/port_def.inc: No such file or directory
>    10 | #include <google/protobuf/port_def.inc>
> Range.pb.h:10:10: fatal error: google/protobuf/port_def.inc: No such file or directory
>    10 | #include <google/protobuf/port_def.inc>
> ./Wind.pb.h:10:10: fatal error: google/protobuf/port_def.inc: No such file or directory
>    10 | #include <google/protobuf/port_def.inc>**

FlavioTonelli, 2020-11-20:
For some reason on my ubuntu 20.04 the protoc compiler version used was the 3.13(I shall have better investigate on that warning).
Now that I'm pointing to the right bin(3.6.1) the issue is over. It seems not related to px4 itself , closing.

closed 2020-11-20 by FlavioTonelli

david-guardian, 2023-06-14:
> For some reason on my ubuntu 20.04 the protoc compiler version used was the 3.13(I shall have better investigate on that warning). Now that I'm pointing to the right bin(3.6.1) the issue is over. It seems not related to px4 itself , closing.

Could you clarify the steps taken to resolve this locally?

whyscience, 2024-11-18:
Check your protoc --version, if it is not 3.6.1, rm it, or `sudo apt-get install libprotobuf-dev protobuf-compiler --reinstall`, 
if this still not work, reinstall from src, 

```
wget https://github.com/protocolbuffers/protobuf/releases/download/v3.6.1/protobuf-all-3.6.1.tar.gz \
    && tar xzf protobuf-all-3.6.1.tar.gz \
    && cd protobuf-3.6.1 \
    && ./autogen.sh \
    && ./configure \
    && make -j${proc_n} && sudo make install
```

then rm -rf Firmware/build, build again, it should work.