PX4/PX4-Autopilot #16246: PX4 v1.11.2 alt-hold flight problem with full pitch

lovettchris, 2020-11-20:
**Describe the bug**
Full pitch forwards on a flight in alt-hold mode creates weird flapping motion.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect Pixhawk 4 hardware to your PC over serial port.
2. Ensure it has firmware version 1.11.2 installed.
3. Run this Windows Binary of latest AirSim bits (or build latest bits yourself).
4. Now fly with ALT-HOLD mode turned on with full pitch forwards

**Expected behavior**
Should fly smoothly.

**Log Files and Screenshots**
*Always* provide a link to the flight log file:
- https://logs.px4.io/plot_app?log=96127077-1bb0-4a6c-ae72-84d92efb1a31.
- See demo video: http://lovettsoftware.com/bugs/px4_alt_hold_bug.mp4

Add screenshots to help explain your problem.

**Drone (please complete the following information):**
- HITL quadrocopter X
- https://shop.holybro.com/pixhawk-4beta-launch_p1089.html.

**Additional context**
This did not happen with v1.9.2 of the firmware on the same hardware.

dagar, 2020-11-20:
Could you give this commit a try? https://github.com/PX4/PX4-Autopilot/pull/16030

It's currently in the `release/1.11` branch which will be v1.11.3 soon enough.

FYI @MaEtUgR

lovettchris, 2020-11-20:
I just tried release/1.11 and I see the same problem.

dagar, 2020-11-20:
Try reducing `MPC_THR_MAX` below 100% (say 95%).

lovettchris, 2020-11-20:
MPC_THR_MAX of 0.95 does not help.  On 1.9.2 is MPC_THR_MAX 1 due to PR 9359 and it flies smoothly there, but I notice it does lose altitude until it hits the ground, so perhaps the new code is trying to stop that from happening, But the flappy oscillation is weird, I think it should instead automaticallly turn down the pitch until it can hold the altitude, but not create the unstable oscillation.  Perhaps it needs some pid smoothing or something.

dagar, 2020-11-20:
Thanks for testing those quick ideas, I'll have to take a closer look at the log.

lovettchris, 2020-11-20:
No problem, I also tried MPC_THR_MAX of 0.90 and same problem, not even a little bit better.

bresch, 2020-11-24:
The cause of this behavior is the prioritization of the altitude hold over the XY axes. The limitation is geometric so rapid switching can occur: tilt more -> altitude loss -> limit tilt -> altitude regained -> tilt more -> ...
One possible workaround is to reduce the maximum tilt to never reach that condition ([MPC_MAN_TILT_MAX](https://dev.px4.io/master/en/advanced/parameter_reference.html#MPC_MAN_TILT_MAX)). 

FYI @MaEtUgR

lovettchris, 2020-11-24:
I figured something like that was going on, but I still claim this bug should be used as an excuse to implement the "proper fix"
 which is to auto-compute the max tilt that will achieve alt-hold and not create this unstable oscillation.