PX4/PX4-Autopilot #16248: which PX4 firmware should I use?

daydayupcg137, 2020-11-20:
Hello, I have made a six-rotor with a wheelbase of 550mm, which can adjust the tilt Angle around the arm. May I ask if your pX4 firmware can be used for control?Is "px4_fmu_v2_multicopter"?The other question is how many degrees can you achieve with a normal tilt Angle?
[image]

dagar, 2020-11-20:
Which flight controller do you have? You should be able to flash directly through QGC and it will automatically pick the correct board and latest stable version of PX4.

daydayupcg137, 2020-11-23:
I use the  pixhawk2.4.8  with 2MB Flash memory,so ,I should use "make px4_fmu-v3_default"?

dagar, 2020-11-23:
Yes, use `make px4_fmu-v3_default`. You may need to update the bootlooader (set SYS_BL_UPDATE 1 and reboot).

daydayupcg137, 2020-11-24:
Hello，I have a new problem when I "make px4_fmu-v3_default". Can you help me？

Error: the git tag 'f33b118dfb' does not match the expected format.

The expected format is 'v<PX4 version>[-<custom version>]'

  <PX4 version>: v<major>.<minor>.<patch>[-rc<rc>|-beta<beta>]

  <custom version>: <major>.<minor>.<patch>

Examples:

  v1.9.0rc3 (deprecated)

  v1.9.0-rc3 (preferred)

  v1.9.0-beta1

  v1.9.0-1.0.0

See also https://dev.px4.io/master/en/setup/building_px4.html#firmware_version

[90/704] Building CXX object msg/CMake.../topics_sources/sensor_selection.cpp.o

ninja: build stopped: subcommand failed.

Makefile:200: recipe for target 'px4_sitl' failed

make: *** [px4_sitl] Error 1

The source code is as follows:https://github.com/castacks/PX4-fully-actuated

closed 2021-01-01 by LorenzMeier