PX4/PX4-Autopilot #16251: canutils - long int in fprintf not passing compiler check

igalloway, 2020-11-20:
Error when compiling. Long int vs unsigned int.

`-- Build files have been written to: /home/iain/PX4-firmware/build/nxp_ucans32k146_default
[39/321] Generating apps/libapps.a
FAILED: NuttX/apps/libapps.a 
cd /home/iain/PX4-firmware/build/nxp_ucans32k146_default/NuttX/apps && find /home/iain/PX4-firmware/build/nxp_ucans32k146_default/NuttX/apps -name *.o -delete && make --quiet --no-print-directory TOPDIR="/home/iain/PX4-firmware/build/nxp_ucans32k146_default/NuttX/nuttx" > nuttx_apps.log
candump.c: In function 'candump_main':
candump.c:711:30: error: format '%ld' expects argument of type 'long int', but argument 3 has type 'time_t' {aka 'unsigned int'} [-Werror=format=]
  711 |      fprintf(logfile, "(%010ld.%06ld) %*s %s\n",
      |                         ~~~~~^
      |                              |
      |                              long int
      |                         %010d
  712 |       tv.tv_sec, tv.tv_usec,
      |       ~~~~~~~~~               
      |         |
      |         time_t {aka unsigned int}
cc1: all warnings being treated as errors
make[3]: *** [/home/iain/PX4-firmware/build/nxp_ucans32k146_default/NuttX/apps/Application.mk:193: candump.home.iain.PX4-firmware.build.nxp_ucans32k146_default.NuttX.apps.canutils.candump.o] Error 1
make[2]: *** [Makefile:59: /home/iain/PX4-firmware/build/nxp_ucans32k146_default/NuttX/apps/canutils/candump_all] Error 2
make[1]: *** [Makefile:53: all] Error 2
[40/321] Generating nuttx/libs/libc/libc.a
ninja: build stopped: subcommand failed.
make: *** [Makefile:226: nxp_ucans32k146] Error 1
`

dagar, 2020-11-20:
As a temporary unblocker while this is fixed in NuttX you can turn off `-Werror` here. https://github.com/PX4/PX4-Autopilot/blob/615333badb289aa77285bcfad071cc6a70b13632/platforms/nuttx/NuttX/Make.defs.in#L104-L115

igalloway, 2020-11-20:
compiles ok with -Werror removed. thanks @dagar 
@PetervdPerk-NXP  is looking at an appropriate solution.