PX4/PX4-Autopilot #16252: Would love to be able to run Simulator TCP over something other than localhost

lovettchris, 2020-11-20:
For general questions please use [PX4 Discuss](http://discuss.px4.io/) or [Slack](http://slack.px4.io/).

**Describe problem solved by the proposed feature**
Thanks to how Windows "WSL2" works, "localhost" communication from wsl to windows no longer works as WSL2 is a running in a completely separate virtual machine.

**Describe your preferred solution**
Allow void Simulator::run() to specify a destination ip address other than INADDR_ANY which only works on local host for tcp connections.  For example, I tested this to connect AirSim running on my windows machine and it works: "_myaddr.sin_addr.s_addr = inet_addr("172.24.144.1");"

**Describe possible alternatives**
Perhaps there is a way to bridge localhost on WSL2 to localhost on my host PC but I couldn't find an easy way to do it.

**Additional context**
But being able to communicate with simulator running on a different machine has other good advantages - for example, load balancing so PX4 and AirSim for example are running on different machines.  For example, it would be cool if I could run PX4 SITL on my raspberry pi and AirSim on my big box with 2080 graphics card.

lberridge1992, 2020-11-23:
Would also love to be able to do this Chris, I managed to use SSH port forwarding as a workaround to talk from a PC running Airsim to a VM running PX4 that was on Virtual Box with a bridged network connection but not sure if WSL supports this. Seems to work without any dodgy side effects other than needing to make sure Airsim is launched before PX4 SITL.

Jaeyoung-Lim, 2020-11-23:
@lovettchris I think https://github.com/PX4/PX4-Autopilot/pull/15965 would already be sufficient for your needs

lovettchris, 2020-11-23:
Close, I added a comment to #15965, for this to work in the WSL situation it would need to support explicit ip address instead of just host name, because hostname is ambiguous in the WSL case.

lovettchris, 2021-04-17:
Fix is in master, closing the issue.

closed 2021-04-17 by lovettchris