PX4/PX4-Autopilot #16254: STM32 F4 DMA on SPI1

lukegluke, 2020-11-22:
I don't know if there is an issue, but I would like to draw your attention and recheck if smt32f4 boards configs are really correct.

For example according to defconfig Omnibus F4 SD board is STM32F405RG board and use DMA on SPI1. But for STM32F4 family heap is allocated from CCM RAM (_if CONFIG_STM32_CCMEXCLUDE is not set_) that cannot be used for DMA. And another STM32F4 board - crazyflie explicitly specify this in comment and don't config SPI1 DMA:
https://github.com/PX4/PX4-Autopilot/blob/615333badb289aa77285bcfad071cc6a70b13632/boards/bitcraze/crazyflie/nuttx-config/include/board.h#L216-L219
I have custom STM32F405RG board with MTD on SPI1 and can confirm that block writing and reading get hanged up if DMA is enabled.

So I wonder why DMA issue is configured on Omnibus F4 SD board? Maybe IMU drivers (MPU6000 and ICM20602 on SPI1 bus) in fact internally don't use a DMA and everything is ok (especially take into account other STM32F4 fmu boards v2-v4), and SP1 DMA config is just usless and can be cleared? While crazyflie has SD card on SPI1?
If this is the point you can easily close this issue, because it is just for a precaution.

dagar, 2020-11-22:
To avoid various problems and keep each driver simple we added support for DMA capable buffers directly within NuttX SPI. https://github.com/PX4/PX4-Autopilot/blob/615333badb289aa77285bcfad071cc6a70b13632/boards/px4/fmu-v4/nuttx-config/nsh/defconfig#L177

lukegluke, 2020-11-22:
Oh, I see. In fact I had this option along with SPI1_DMA, but it didn't help for MTD NuttX driver, I didn't investigate why it get hanging on writes/reads while successfully attached.

closed 2021-01-01 by LorenzMeier