PX4/PX4-Autopilot #16256: wrong idle typemask in offboard mode using mavros

dssdyx, 2020-11-23:
[PX4/Firmware/blob/master/src/modules/mavlink/mavlink_receiver.cpp#L825-L840](url)
`bool is_idle_sp = (bool)(set_position_target_local_ned.type_mask & 0x4000);`
I want to idle my drone in offboard mode, and I send idle setpoint through mavros at 10 Hz then switch to offboard. I use 0x4000 as the typemask of setpoint massage according to mavlink_receiver.cpp, but in QGC, it shows "INFO Takeoff Detected" directly and motors spin very fast. My FCU version is v1.11.1.There is the same error in v1.11.2 and v1.11.0-beta. However, it can success idle in v1.9.2.
I find a discuss about it.
[https://discuss.px4.io/t/offboard-takeoff-land-loiter-via-set-position-target-local-ned/2648](url)
`enum TypeMasks {`
  `        MASK_POSITION =         0b0000111111111000,`
  `        MASK_VELOCITY =         0b0000111111000111,`
  `        MASK_TAKEOFF_POSITION = 0b0001111111111000,`
  `        MASK_TAKEOFF =          0b0001111111111111,`
  `        MASK_LAND_VELOCITY =    0b0010111111000111,`
  `        MASK_LAND_POSITION =    0b0010111111111000,`
  `        MASK_LAND =             0b0010111111111111,`
  `        MASK_LOITER_POSITION =  ### 0b0100111111111000,`
  `        MASK_IDLE_POSITION =    0b1000111111111000,`
`};`
Then I use MASK_IDLE_POSITION =    0b1000111111111000 as the idle typemask and try again in v1.11.1. It can idle by 0b1000111111111000. I feel confused because this is different with mavlink_receiver.cpp.

Jaeyoung-Lim, 2020-11-28:
@dssdyx 
```
enum TypeMasks {
MASK_POSITION = 0b0000111111111000,
MASK_VELOCITY = 0b0000111111000111,
MASK_TAKEOFF_POSITION = 0b0001111111111000,
MASK_TAKEOFF = 0b0001111111111111,
MASK_LAND_VELOCITY = 0b0010111111000111,
MASK_LAND_POSITION = 0b0010111111111000,
MASK_LAND = 0b0010111111111111,
MASK_LOITER_POSITION = ### 0b0100111111111000,
MASK_IDLE_POSITION = 0b1000111111111000,
};
```
These enums are probably outdated. It is not defined in PX4, so what you have in mavlink_receiver is accurate

Jaeyoung-Lim, 2021-01-08:
Addressed in https://github.com/PX4/PX4-Autopilot/pull/16296

closed 2021-01-08 by Jaeyoung-Lim