PX4/PX4-Autopilot #16260: make px4_sitl_default gazebo  error

machenxiang, 2020-11-23:
**Describe the bug**
When I  make px4_sitl_default gazebo , it occurs below
```
-- PX4 version: v1.11.0-beta1
-- PX4 config file: /home/y/PX4_Firmware/boards/px4/sitl/default.cmake
-- PX4 config: px4_sitl_default
-- PX4 platform: posix
-- PX4 lockstep: enabled
-- cmake build type: RelWithDebInfo
-- The CXX compiler identification is GNU 7.5.0
-- The C compiler identification is GNU 7.5.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/cc
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Building for code coverage
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.6.9", minimum required is "3") 
-- build type is RelWithDebInfo
-- PX4 ECL: Very lightweight Estimation & Control Library v1.9.0-rc1-255-g47624a0
-- Configuring done
-- Generating done
-- Build files have been written to: /home/y/PX4_Firmware/build/px4_sitl_default
[1/740] git submodule src/drivers/gps/devices
[9/740] git submodule src/lib/ecl
[14/740] git submodule mavlink/include/mavlink/v2.0
[16/740] git submodule Tools/sitl_gazebo
[21/740] Performing configure step for 'sitl_gazebo'
-- install-prefix: /usr/local
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Performing Test COMPILER_SUPPORTS_CXX17
-- Performing Test COMPILER_SUPPORTS_CXX17 - Success
-- Performing Test COMPILER_SUPPORTS_CXX14
-- Performing Test COMPILER_SUPPORTS_CXX14 - Success
-- Performing Test COMPILER_SUPPORTS_CXX11
-- Performing Test COMPILER_SUPPORTS_CXX11 - Success
-- Performing Test COMPILER_SUPPORTS_CXX0X
-- Performing Test COMPILER_SUPPORTS_CXX0X - Success
-- Using C++17 compiler
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
[... 145 lines cut ...]
[49/100] Building CXX object CMakeFiles/gazebo_multirotor_base_plugin.dir/src/gazebo_multirotor_base_plugin.cpp.o
FAILED: CMakeFiles/gazebo_multirotor_base_plugin.dir/src/gazebo_multirotor_base_plugin.cpp.o 
/usr/bin/c++  -DLIBBULLET_VERSION=2.87 -DLIBBULLET_VERSION_GT_282 -Dgazebo_multirotor_base_plugin_EXPORTS -isystem /usr/include/gazebo-9 -isystem /usr/include/bullet -isystem /usr/include/simbody -isystem /usr/include/sdformat-6.0 -isystem /usr/include/ignition/math4 -isystem /usr/include/OGRE -isystem /usr/include/OGRE/Terrain -isystem /usr/include/OGRE/Paging -isystem /usr/include/ignition/transport4 -isystem /usr/include/ignition/msgs1 -isystem /usr/include/ignition/common1 -isystem /usr/include/ignition/fuel_tools1 -isystem /usr/include/x86_64-linux-gnu/qt5 -isystem /usr/include/x86_64-linux-gnu/qt5/QtCore -isystem /usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++ -I/home/y/PX4_Firmware/Tools/sitl_gazebo/include -I. -I/usr/include/eigen3 -I/usr/include/eigen3/eigen3 -I/usr/include/gazebo-9/gazebo/msgs -I/home/y/PX4_Firmware/mavlink/include -isystem /usr/include/opencv -I/home/y/PX4_Firmware/Tools/sitl_gazebo/external/OpticalFlow/include -I/usr/include/gstreamer-1.0 -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -isystem /usr/include/uuid -isystem /usr/include/x86_64-linux-gnu -Wno-deprecated-declarations -Wno-address-of-packed-member -fPIC   -I/usr/include/uuid -I/usr/include/x86_64-linux-gnu -std=gnu++1z -MD -MT CMakeFiles/gazebo_multirotor_base_plugin.dir/src/gazebo_multirotor_base_plugin.cpp.o -MF CMakeFiles/gazebo_multirotor_base_plugin.dir/src/gazebo_multirotor_base_plugin.cpp.o.d -o CMakeFiles/gazebo_multirotor_base_plugin.dir/src/gazebo_multirotor_base_plugin.cpp.o -c /home/y/PX4_Firmware/Tools/sitl_gazebo/src/gazebo_multirotor_base_plugin.cpp
In file included from /usr/include/gazebo-9/gazebo/msgs/MessageTypes.hh:82:0,
                 from /usr/include/gazebo-9/gazebo/msgs/msgs.hh:35,
                 from /usr/include/gazebo-9/gazebo/gazebo_core.hh:20,
                 from /usr/include/gazebo-9/gazebo/gazebo.hh:20,
                 from /home/y/PX4_Firmware/Tools/sitl_gazebo/include/gazebo_multirotor_base_plugin.h:26,
                 from /home/y/PX4_Firmware/Tools/sitl_gazebo/src/gazebo_multirotor_base_plugin.cpp:21:
/usr/include/gazebo-9/gazebo/msgs/pose_animation.pb.h:60:3: internal compiler error: Segmentation fault
   }
   ^
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
[53/100] Building CXX object CMakeFiles/gazebo_opticalflow_plugin.dir/src/gazebo_opticalflow_plugin.cpp.o
ninja: build stopped: subcommand failed.
[736/740] Linking CXX executable bin/px4
FAILED: external/Stamp/sitl_gazebo/sitl_gazebo-build 
cd /home/y/PX4_Firmware/build/px4_sitl_default/build_gazebo && /usr/bin/cmake --build .
ninja: build stopped: subcommand failed.
Makefile:198: recipe for target 'px4_sitl_default' failed
make: *** [px4_sitl_default] Error 1

```

Jaeyoung-Lim, 2020-11-23:
Try
```
make clean
make px4_sitl gazebo
```

machenxiang, 2020-11-23:
> Try
> 
> ```
> make clean
> make px4_sitl gazebo
> ```

Thanks for your reply Jaeyoung-Lim. I have tried this but it has the same error.

machenxiang, 2020-11-24:
I have solved this problem. This error is because of running out of swap space. When I install Ubuntu 18.04,  I set my swap space 8g. So I should add my swap space.
```
#The first step
sudo dd if=/dev/zero of=/home/swap bs=64M count=64
#The second step
 sudo mkswap /home/swap 
#The third step
sudo swapon /home/swap
#After all above finished ,make clean and build again
make clean
make px4_sitl_default gazebo
```

Jaeyoung-Lim, 2020-11-24:
@machenxiang Thanks for sharing your solution, and glad that you manage to get it working.

Closing as resolved

closed 2020-11-24 by Jaeyoung-Lim

Amna-Smaoui, 2022-04-27:
I have the same problem and I don't know how to fix it . I tried everything but still doesn't work 
[image]