PX4/PX4-Autopilot #16261: make px4_sitl gazebo fail in v1.11.2

mzahana, 2020-11-23:
**Describe the bug**
I am not able to compile the Gazebo SITL target on release  `v1.11.2` using
```
make px4_sitl gazebo
```

I get the following error(s)

```sh
[20/749] Generating parameters.xml
FAILED: parameters.xml parameters.json 
cd /home/arrow/Firmware/build/px4_sitl_default/src/lib/parameters && /usr/bin/python3 /home/arrow/Firmware/src/lib/parameters/px_process_params.py --src-path /home/arrow/Firmware/src/lib/airspeed /home/arrow/Firmware/src/lib/airspeed_validator /home/arrow/Firmware/src/lib/avoidance /home/arrow/Firmware/src/lib/battery /home/arrow/Firmware/src/lib/bezier /home/arrow/Firmware/src/lib/cdev /home/arrow/Firmware/src/lib/cdev/test /home/arrow/Firmware/src/lib/circuit_breaker /home/arrow/Firmware/src/lib/collision_prevention /home/arrow/Firmware/src/lib/controllib /home/arrow/Firmware/src/lib/controllib/controllib_test /home/arrow/Firmware/src/lib/conversion /home/arrow/Firmware/src/lib/drivers/accelerometer /home/arrow/Firmware/src/lib/drivers/airspeed /home/arrow/Firmware/src/lib/drivers/barometer /home/arrow/Firmware/src/lib/drivers/device /home/arrow/Firmware/src/lib/drivers/gyroscope /home/arrow/Firmware/src/lib/drivers/led /home/arrow/Firmware/src/lib/drivers/magnetometer /home/arrow/Firmware/src/lib/drivers/rangefinder /home/arrow/Firmware/src/lib/drivers/smbus /home/arrow/Firmware/src/lib/flight_tasks /home/arrow/Firmware/src/lib/flight_tasks/tasks/FlightTask /home/arrow/Firmware/src/lib/flight_tasks/tasks/Utility /home/arrow/Firmware/src/lib/flight_tasks/tasks/Auto /home/arrow/Firmware/src/lib/flight_tasks/tasks/AutoMapper /home/arrow/Firmware/src/lib/flight_tasks/tasks/ManualAltitude /home/arrow/Firmware/src/lib/flight_tasks/tasks/ManualAltitudeSmooth /home/arrow/Firmware/src/lib/flight_tasks/tasks/ManualAltitudeSmoothVel /home/arrow/Firmware/src/lib/flight_tasks/tasks/ManualPosition /home/arrow/Firmware/src/lib/flight_tasks/tasks/ManualPositionSmooth /home/arrow/Firmware/src/lib/flight_tasks/tasks/ManualPositionSmoothVel /home/arrow/Firmware/src/lib/flight_tasks/tasks/AutoLineSmoothVel /home/arrow/Firmware/src/lib/flight_tasks/tasks/AutoFollowMe /home/arrow/Firmware/src/lib/flight_tasks/tasks/Offboard /home/arrow/Firmware/src/lib/flight_tasks/tasks/Failsafe /home/arrow/Firmware/src/lib/flight_tasks/tasks/Descend /home/arrow/Firmware/src/lib/flight_tasks/tasks/Transition /home/arrow/Firmware/src/lib/flight_tasks/tasks/Orbit /home/arrow/Firmware/src/lib/hysteresis /home/arrow/Firmware/src/lib/l1 /home/arrow/Firmware/src/lib/landing_slope /home/arrow/Firmware/src/lib/led /home/arrow/Firmware/src/lib/mag_compensation /home/arrow/Firmware/src/lib/mathlib /home/arrow/Firmware/src/lib/mixer_module /home/arrow/Firmware/src/lib/output_limit /home/arrow/Firmware/src/lib/pid /home/arrow/Firmware/src/lib/rc/rc_tests /home/arrow/Firmware/src/lib/systemlib /home/arrow/Firmware/src/lib/tecs /home/arrow/Firmware/src/lib/terrain_estimation /home/arrow/Firmware/src/lib/tunes /home/arrow/Firmware/src/lib/weather_vane /home/arrow/Firmware/platforms/posix/src/px4/common/px4_daemon /home/arrow/Firmware/platforms/posix/src/px4/common/lockstep_scheduler /home/arrow/Firmware/platforms/posix/src/px4/generic/generic/tone_alarm /home/arrow/Firmware/platforms/common/px4_work_queue /home/arrow/Firmware/platforms/common/px4_work_queue/test /home/arrow/Firmware/src/modules/uORB /home/arrow/Firmware/src/modules/uORB/uORB_tests /home/arrow/Firmware/src/drivers/camera_capture /home/arrow/Firmware/src/drivers/camera_trigger /home/arrow/Firmware/src/drivers/gps /home/arrow/Firmware/src/drivers/pwm_out_sim /home/arrow/Firmware/src/drivers/rpm/rpm_simulator /home/arrow/Firmware/src/drivers/tone_alarm /home/arrow/Firmware/src/modules/airship_att_control /home/arrow/Firmware/src/modules/airspeed_selector /home/arrow/Firmware/src/modules/attitude_estimator_q /home/arrow/Firmware/src/modules/camera_feedback /home/arrow/Firmware/src/modules/commander/failure_detector /home/arrow/Firmware/src/modules/commander/Arming/PreFlightCheck /home/arrow/Firmware/src/modules/commander/Arming/ArmAuthorization /home/arrow/Firmware/src/modules/commander/Arming/HealthFlags /home/arrow/Firmware/src/modules/commander /home/arrow/Firmware/src/modules/commander/commander_tests /home/arrow/Firmware/src/modules/dataman /home/arrow/Firmware/src/modules/ekf2/Utility /home/arrow/Firmware/src/modules/ekf2 /home/arrow/Firmware/src/modules/events /home/arrow/Firmware/src/modules/fw_att_control /home/arrow/Firmware/src/modules/fw_pos_control_l1/launchdetection /home/arrow/Firmware/src/modules/fw_pos_control_l1/runway_takeoff /home/arrow/Firmware/src/modules/fw_pos_control_l1 /home/arrow/Firmware/src/modules/land_detector /home/arrow/Firmware/src/modules/landing_target_estimator /home/arrow/Firmware/src/modules/local_position_estimator /home/arrow/Firmware/src/modules/logger /home/arrow/Firmware/src/modules/mavlink /home/arrow/Firmware/src/modules/mavlink/mavlink_tests /home/arrow/Firmware/src/modules/mc_att_control/AttitudeControl /home/arrow/Firmware/src/modules/mc_att_control /home/arrow/Firmware/src/modules/mc_hover_thrust_estimator /home/arrow/Firmware/src/modules/mc_pos_control/Takeoff /home/arrow/Firmware/src/modules/mc_pos_control/PositionControl /home/arrow/Firmware/src/modules/mc_pos_control /home/arrow/Firmware/src/modules/mc_rate_control/RateControl /home/arrow/Firmware/src/modules/mc_rate_control /home/arrow/Firmware/src/modules/navigator /home/arrow/Firmware/src/modules/rc_update /home/arrow/Firmware/src/modules/replay /home/arrow/Firmware/src/modules/rover_pos_control /home/arrow/Firmware/src/modules/sensors/data_validator /home/arrow/Firmware/src/modules/sensors/sensor_calibration /home/arrow/Firmware/src/modules/sensors/vehicle_acceleration /home/arrow/Firmware/src/modules/sensors/vehicle_angular_velocity /home/arrow/Firmware/src/modules/sensors/vehicle_air_data /home/arrow/Firmware/src/modules/sensors/vehicle_imu /home/arrow/Firmware/src/modules/sensors /home/arrow/Firmware/src/modules/simulator /home/arrow/Firmware/src/modules/simulator/battery_simulator /home/arrow/Firmware/src/modules/temperature_compensation /home/arrow/Firmware/src/modules/vmount /home/arrow/Firmware/src/modules/vtol_att_control /home/arrow/Firmware/src/modules/uuv_att_control /home/arrow/Firmware/src/systemcmds/dyn /home/arrow/Firmware/src/systemcmds/esc_calib /home/arrow/Firmware/src/systemcmds/led_control /home/arrow/Firmware/src/systemcmds/mixer /home/arrow/Firmware/src/systemcmds/motor_ramp /home/arrow/Firmware/src/systemcmds/motor_test /home/arrow/Firmware/src/systemcmds/param /home/arrow/Firmware/src/systemcmds/perf /home/arrow/Firmware/src/systemcmds/pwm /home/arrow/Firmware/src/systemcmds/sd_bench /home/arrow/Firmware/src/systemcmds/shutdown /home/arrow/Firmware/src/systemcmds/tests /home/arrow/Firmware/src/systemcmds/tests/hrt_test /home/arrow/Firmware/src/systemcmds/topic_listener /home/arrow/Firmware/src/systemcmds/tune_control /home/arrow/Firmware/src/systemcmds/ver /home/arrow/Firmware/src/systemcmds/work_queue /home/arrow/Firmware/src/examples/fake_magnetometer /home/arrow/Firmware/src/examples/fixedwing_control /home/arrow/Firmware/src/examples/hello /home/arrow/Firmware/src/examples/px4_mavlink_debug /home/arrow/Firmware/src/examples/px4_simple_app /home/arrow/Firmware/src/examples/rover_steering_control /home/arrow/Firmware/src/examples/uuv_example_app /home/arrow/Firmware/src/examples/work_item /home/arrow/Firmware/build/px4_sitl_default/generated_params --xml /home/arrow/Firmware/build/px4_sitl_default/parameters.xml --json /home/arrow/Firmware/build/px4_sitl_default/parameters.json --compress --inject-xml /home/arrow/Firmware/src/lib/parameters/parameters_injected.xml --overrides {} --board px4_sitl
Traceback (most recent call last):
  File "/home/arrow/Firmware/src/lib/parameters/px_process_params.py", line 192, in <module>
    main()
  File "/home/arrow/Firmware/src/lib/parameters/px_process_params.py", line 188, in main
    SaveCompressed(f)
  File "/home/arrow/Firmware/src/lib/parameters/px_process_params.py", line 62, in SaveCompressed
    f.write(content_file.read())
  File "/usr/lib/python3.6/encodings/ascii.py", line 26, in decode
    return codecs.ascii_decode(input, self.errors)[0]
UnicodeDecodeError: 'ascii' codec can't decode byte 0xc2 in position 2925: ordinal not in range(128)
[21/749] Generating uORB topic headers
[... cut ...]
    text = f.read()
  File "/usr/lib/python3.6/encodings/ascii.py", line 26, in decode
    return codecs.ascii_decode(input, self.errors)[0]
UnicodeDecodeError: 'ascii' codec can't decode byte 0xc2 in position 3349: ordinal not in range(128)
ninja: build stopped: subcommand failed.
Makefile:224: recipe for target 'px4_sitl' failed
make: *** [px4_sitl] Error 1
```

**To Reproduce**
* clone the firmware
* checkout `v1.11.2`
* `git submodule update --recursive`
* `make px4_sitl gazebo`

@Jaeyoung-Lim any ideas?

dagar, 2020-11-23:
Looks like python versioning.

mzahana, 2020-11-23:
@dagar Same in master. I am using a docker container with Ubuntu 18 and ROS melodic

mzahana, 2020-11-24:
@dagar If it compiles fine with you, could you mention the Python version you use?
For me, I am using Python 3.6.9

Jaeyoung-Lim, 2020-11-24:
@mzahana We run build tests in the CI server using the containers https://github.com/PX4/PX4-containers, so it "should" work if everything is configured correctly

mzahana, 2020-11-25:
Problem solved after using clean Ubuntu 18 environment.

closed 2020-11-25 by mzahana

bigonestep, 2021-01-10:
Encountered the same problem, besides changing to a clean environment (although I am using a clean docker ubuntu18.04 environment with a desktop), what other solutions are there?

mcelhennyi, 2021-01-20:
> Encountered the same problem, besides changing to a clean environment (although I am using a clean docker ubuntu18.04 environment with a desktop), what other solutions are there?

https://github.com/PX4/PX4-Autopilot/pull/13978#issuecomment-575884123

This helped me^^