PX4/PX4-Autopilot #16263: SITL Build Error on Mac OS 'readarray' not supported

canberkdurmus, 2020-11-24:
**Description of the bug**
While building the PX4-Autopilot with `make px4_sitl gazebo`, in the `PX4-Autopilot/Tools/sitl_run.sh`file `readarray` is not supported by the Mac OS X (Catalina 10.15.7).

**To Reproduce**
Steps to reproduce the behavior:
1. Clone the repository PX4-Autopilot from the master branch
2. Go to the PX4-Autopilot folder
3. Execute `make px4_sitl gazebo`
4. See error

**Expected behavior**
Gazebo SITL launch.

**Log Files and Screenshots**
[image]

**Drone (please complete the following information):**
- SITL Default Iris Quadcopter

**Additional context**
Output As Text:
``` 
[0/5] Performing build step for 'sitl_gazebo'
ninja: no work to do.
[4/5] cd /Users/canberk/Desktop/PX4-Auto...top/PX4-Autopilot/build/px4_sitl_defaul
SITL ARGS
sitl_bin: /Users/canberk/Desktop/PX4-Autopilot/build/px4_sitl_default/bin/px4
debugger: none
program: gazebo
model: none
world: none
src_path: /Users/canberk/Desktop/PX4-Autopilot
build_path: /Users/canberk/Desktop/PX4-Autopilot/build/px4_sitl_default
empty model, setting iris as default
GAZEBO_PLUGIN_PATH :/Users/canberk/Desktop/PX4-Autopilot/build/px4_sitl_default/build_gazebo
GAZEBO_MODEL_PATH :/Users/canberk/Desktop/PX4-Autopilot/Tools/sitl_gazebo/models
LD_LIBRARY_PATH :/Users/canberk/Desktop/PX4-Autopilot/build/px4_sitl_default/build_gazebo
empty world, setting empty.world as default
/Users/canberk/Desktop/PX4-Autopilot/Tools/sitl_run.sh: line 131: readarray: command not found
FAILED: platforms/posix/CMakeFiles/gazebo 
cd /Users/canberk/Desktop/PX4-Autopilot/build/px4_sitl_default/tmp && /Users/canberk/Desktop/PX4-Autopilot/Tools/sitl_run.sh /Users/canberk/Desktop/PX4-Autopilot/build/px4_sitl_default/bin/px4 none gazebo none none /Users/canberk/Desktop/PX4-Autopilot /Users/canberk/Desktop/PX4-Autopilot/build/px4_sitl_default
ninja: build stopped: subcommand failed.
make: *** [px4_sitl] Error 1
```

Jaeyoung-Lim, 2020-11-26:
This is a regression caused by https://github.com/PX4/PX4-Autopilot/pull/16214

Jaeyoung-Lim, 2020-11-26:
@canberkdurmus Fixed in https://github.com/PX4/PX4-Autopilot/pull/16287 

Thanks for reporting!

canberkdurmus, 2020-11-27:
Thank you for the support, I am closing the issue.

closed 2020-11-27 by canberkdurmus