PX4/PX4-Autopilot #16275: I am not being able to run 'make px4_sitl gazebo'

ricardorodrigues09, 2020-11-25:
Hello! So I get on my PX4-Autopilot directory, I run 'make px4_sitl gazebo' and I get the following output:

[0/4] Performing build step for 'sitl_gazebo'
ninja: no work to do.
[3/4] cd /home/ricardorodrigues/PX4-Au...s/PX4-Autopilot/build/px4_sitl_default
SITL ARGS
sitl_bin: /home/ricardorodrigues/PX4-Autopilot/build/px4_sitl_default/bin/px4
debugger: none
program: gazebo
model: none
world: none
src_path: /home/ricardorodrigues/PX4-Autopilot
build_path: /home/ricardorodrigues/PX4-Autopilot/build/px4_sitl_default
empty model, setting iris as default
GAZEBO_PLUGIN_PATH :/home/ricardorodrigues/PX4-Autopilot/build/px4_sitl_default/build_gazebo:/home/ricardorodrigues/PX4-Autopilot/build/px4_sitl_default/build_gazebo
GAZEBO_MODEL_PATH :/home/ricardorodrigues/PX4-Autopilot/Tools/sitl_gazebo/models:/home/ricardorodrigues/PX4-Autopilot/Tools/sitl_gazebo/models
LD_LIBRARY_PATH /home/ricardorodrigues/catkin_ws/devel/lib:/opt/ros/melodic/lib:/home/ricardorodrigues/PX4-Autopilot/build/px4_sitl_default/build_gazebo:/home/ricardorodrigues/PX4-Autopilot/build/px4_sitl_default/build_gazebo
empty world, setting empty.world as default
/home/ricardorodrigues/PX4-Autopilot/Tools/sitl_run.sh: line 134: possibleModelPath: command not found
FAILED: platforms/posix/CMakeFiles/gazebo 
cd /home/ricardorodrigues/PX4-Autopilot/build/px4_sitl_default/tmp && /home/ricardorodrigues/PX4-Autopilot/Tools/sitl_run.sh /home/ricardorodrigues/PX4-Autopilot/build/px4_sitl_default/bin/px4 none gazebo none none /home/ricardorodrigues/PX4-Autopilot /home/ricardorodrigues/PX4-Autopilot/build/px4_sitl_default
ninja: build stopped: subcommand failed.
Makefile:224: recipe for target 'px4_sitl' failed
make: *** [px4_sitl] Error 1

Can someone please explain me what's the problem here?
Thank you in advance.

Jaeyoung-Lim, 2020-11-26:
Which system are you running this on?

ricardorodrigues09, 2020-11-26:
Ubuntu 18.04.5 LTS (Bionic Beaver)

Jaeyoung-Lim, 2020-11-26:
@ricardorodrigues09 You are missing the models in the submodule
```
make submodulesupdate
```

ricardorodrigues09, 2020-11-26:
I did that and now I'm getting:

[0/4] Performing build step for 'sitl_gazebo'
ninja: no work to do.
[3/4] cd /home/ricardorodrigues/PX4-Au...s/PX4-Autopilot/build/px4_sitl_default
SITL ARGS
sitl_bin: /home/ricardorodrigues/PX4-Autopilot/build/px4_sitl_default/bin/px4
debugger: none
program: gazebo
model: none
world: none
src_path: /home/ricardorodrigues/PX4-Autopilot
build_path: /home/ricardorodrigues/PX4-Autopilot/build/px4_sitl_default
empty model, setting iris as default
GAZEBO_PLUGIN_PATH :/home/ricardorodrigues/PX4-Autopilot/build/px4_sitl_default/build_gazebo
GAZEBO_MODEL_PATH :/home/ricardorodrigues/PX4-Autopilot/Tools/sitl_gazebo/models
LD_LIBRARY_PATH /home/ricardorodrigues/catkin_ws/devel/lib:/opt/ros/melodic/lib:/home/ricardorodrigues/PX4-Autopilot/build/px4_sitl_default/build_gazebo
empty world, setting empty.world as default
/home/ricardorodrigues/PX4-Autopilot/Tools/sitl_run.sh: line 134: possibleModelPath: command not found

which is basically a smaller portion of what I had before

Jaeyoung-Lim, 2020-11-27:
@ricardorodrigues09 Could you maybe try with https://github.com/PX4/PX4-Autopilot/pull/16287 ?

ricardorodrigues09, 2020-11-28:
Thank you very much @Jaeyoung-Lim, it worked! Now something strange is happening, the robot is not loading, contrary to the world. This is what I get:

[0/4] Performing build step for 'sitl_gazebo'
ninja: no work to do.
[3/4] cd /home/ricardorodrigues/PX4-Au...s/PX4-Autopilot/build/px4_sitl_default
SITL ARGS
sitl_bin: /home/ricardorodrigues/PX4-Autopilot/build/px4_sitl_default/bin/px4
debugger: none
program: gazebo
model: none
world: none
src_path: /home/ricardorodrigues/PX4-Autopilot
build_path: /home/ricardorodrigues/PX4-Autopilot/build/px4_sitl_default
empty model, setting iris as default
GAZEBO_PLUGIN_PATH :/home/ricardorodrigues/PX4-Autopilot/build/px4_sitl_default/build_gazebo:/home/ricardorodrigues/PX4-Autopilot/build/px4_sitl_default/build_gazebo:/home/ricardorodrigues/PX4-Autopilot/build/px4_sitl_default/build_gazebo
GAZEBO_MODEL_PATH :/home/ricardorodrigues/PX4-Autopilot/Tools/sitl_gazebo/models:/home/ricardorodrigues/PX4-Autopilot/Tools/sitl_gazebo/models:/home/ricardorodrigues/PX4-Autopilot/Tools/sitl_gazebo/models
LD_LIBRARY_PATH /home/ricardorodrigues/catkin_ws/devel/lib:/opt/ros/melodic/lib:/home/ricardorodrigues/PX4-Autopilot/build/px4_sitl_default/build_gazebo:/home/ricardorodrigues/PX4-Autopilot/build/px4_sitl_default/build_gazebo:/home/ricardorodrigues/PX4-Autopilot/build/px4_sitl_default/build_gazebo
empty world, setting empty.world as default
Using: /iris/iris.sdf
SITL COMMAND: "/home/ricardorodrigues/PX4-Autopilot/build/px4_sitl_default/bin/px4" "/home/ricardorodrigues/PX4-Autopilot/build/px4_sitl_default"/etc -s etc/init.d-posix/rcS -t "/home/ricardorodrigues/PX4-Autopilot"/test_data
INFO  [px4] Creating symlink /home/ricardorodrigues/PX4-Autopilot/build/px4_sitl_default/etc -> /home/ricardorodrigues/PX4-Autopilot/build/px4_sitl_default/tmp/rootfs/etc

______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 0
Info: found model autostart file as SYS_AUTOSTART=10016
INFO  [param] selected parameter default file eeprom/parameters_10016
[param] Loaded: eeprom/parameters_10016
INFO  [dataman] Unknown restart, data manager file './dataman' size is 11798680 bytes
INFO  [simulator] Waiting for simulator to accept connection on TCP port 4560
Gazebo multi-robot simulator, version 9.15.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 172.16.113.145

Everything seems fine right?

Jaeyoung-Lim, 2020-11-28:
It is not calling the model
```
[3/4] cd /home/ricardorodrigues/PX4-Au...s/PX4-Autopilot/build/px4_sitl_default
SITL ARGS
sitl_bin: /home/ricardorodrigues/PX4-Autopilot/build/px4_sitl_default/bin/px4
debugger: none
program: gazebo
model: none
world: none
```
Are you running the following command?
```
make px4_sitl gazebo_none
```

ricardorodrigues09, 2020-11-28:
@Jaeyoung-Lim, I am running : 

make px4_sitl gazebo

For example, with "make px4_sitl gazebo_plane" I get this as output:

[0/4] Performing build step for 'sitl_gazebo'
ninja: no work to do.
[3/4] cd /home/ricardorodrigues/PX4-Au...s/PX4-Autopilot/build/px4_sitl_default
SITL ARGS
sitl_bin: /home/ricardorodrigues/PX4-Autopilot/build/px4_sitl_default/bin/px4
debugger: none
program: gazebo
model: plane
world: none
src_path: /home/ricardorodrigues/PX4-Autopilot
build_path: /home/ricardorodrigues/PX4-Autopilot/build/px4_sitl_default
GAZEBO_PLUGIN_PATH :/home/ricardorodrigues/PX4-Autopilot/build/px4_sitl_default/build_gazebo
GAZEBO_MODEL_PATH :/home/ricardorodrigues/PX4-Autopilot/Tools/sitl_gazebo/models
LD_LIBRARY_PATH /home/ricardorodrigues/catkin_ws/devel/lib:/opt/ros/melodic/lib:/home/ricardorodrigues/PX4-Autopilot/build/px4_sitl_default/build_gazebo
empty world, setting empty.world as default
Using: /plane/plane.sdf
SITL COMMAND: "/home/ricardorodrigues/PX4-Autopilot/build/px4_sitl_default/bin/px4" "/home/ricardorodrigues/PX4-Autopilot/build/px4_sitl_default"/etc -s etc/init.d-posix/rcS -t "/home/ricardorodrigues/PX4-Autopilot"/test_data
INFO  [px4] Creating symlink /home/ricardorodrigues/PX4-Autopilot/build/px4_sitl_default/etc -> /home/ricardorodrigues/PX4-Autopilot/build/px4_sitl_default/tmp/rootfs/etc

______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 0
Info: found model autostart file as SYS_AUTOSTART=1030
INFO  [param] selected parameter default file eeprom/parameters_1030
[param] parameter file not found, creating eeprom/parameters_1030
  SYS_AUTOCONFIG: curr: 0 -> new: 1
  SYS_AUTOSTART: curr: 0 -> new: 1030
  BAT_N_CELLS: curr: 0 -> new: 4
  CAL_ACC0_ID: curr: 0 -> new: 1310988
  CAL_ACC1_ID: curr: 0 -> new: 1310996
  CAL_ACC2_ID: curr: 0 -> new: 1311004
  CAL_GYRO0_ID: curr: 0 -> new: 1310988
  CAL_GYRO1_ID: curr: 0 -> new: 1310996
  CAL_GYRO2_ID: curr: 0 -> new: 1311004
  CAL_MAG0_ID: curr: 0 -> new: 197388
  CAL_MAG1_ID: curr: 0 -> new: 197644
  CBRK_SUPPLY_CHK: curr: 0 -> new: 894281
  COM_RC_IN_MODE: curr: 0 -> new: 1
  EKF2_REQ_GPS_H: curr: 10.0000 -> new: 0.5000
  EKF2_MULTI_IMU: curr: 0 -> new: 3
  SENS_IMU_MODE: curr: 1 -> new: 0
  EKF2_MULTI_MAG: curr: 0 -> new: 2
  SENS_MAG_MODE: curr: 1 -> new: 0
  SDLOG_MODE: curr: 0 -> new: 1
  SDLOG_PROFILE: curr: 1 -> new: 131
  SDLOG_DIRS_MAX: curr: 0 -> new: 7
  SENS_BOARD_X_OFF: curr: 0.0000 -> new: 0.0000
  SENS_DPRES_OFF: curr: 0.0000 -> new: 0.0010
  TRIG_INTERFACE: curr: 4 -> new: 3
  COM_CPU_MAX: curr: 90.0000 -> new: -1.0000
  IMU_INTEG_RATE: curr: 200 -> new: 250
  COM_POS_FS_DELAY: curr: 1 -> new: 5
  COM_POS_FS_EPH: curr: 5.0000 -> new: 15.0000
  COM_POS_FS_EPV: curr: 10.0000 -> new: 30.0000
  COM_POS_FS_GAIN: curr: 10 -> new: 0
  COM_POS_FS_PROB: curr: 30 -> new: 1
  COM_VEL_FS_EVH: curr: 1.0000 -> new: 5.0000
  EKF2_ARSP_THR: curr: 0.0000 -> new: 8.0000
  EKF2_FUSE_BETA: curr: 0 -> new: 1
  EKF2_GPS_CHECK: curr: 245 -> new: 21
  EKF2_MAG_ACCLIM: curr: 0.5000 -> new: 0.0000
  EKF2_MAG_YAWLIM: curr: 0.2500 -> new: 0.0000
  EKF2_REQ_EPH: curr: 3.0000 -> new: 10.0000
  EKF2_REQ_EPV: curr: 5.0000 -> new: 10.0000
  EKF2_REQ_HDRIFT: curr: 0.1000 -> new: 0.5000
  EKF2_REQ_SACC: curr: 0.5000 -> new: 1.0000
  EKF2_REQ_VDRIFT: curr: 0.2000 -> new: 1.0000
  RTL_TYPE: curr: 0 -> new: 1
  RTL_RETURN_ALT: curr: 60.0000 -> new: 100.0000
  RTL_DESCEND_ALT: curr: 30.0000 -> new: 100.0000
  MIS_LTRMIN_ALT: curr: -1.0000 -> new: 25.0000
  MIS_TAKEOFF_ALT: curr: 2.5000 -> new: 25.0000
  PWM_RATE: curr: 400 -> new: 50
  GPS_UBX_DYNMODEL: curr: 7 -> new: 8
  FW_LND_AIRSPD_SC: curr: 1.3000 -> new: 1.0000
  FW_LND_ANG: curr: 5.0000 -> new: 8.0000
  FW_THR_LND_MAX: curr: 1.0000 -> new: 0.0000
  FW_L1_PERIOD: curr: 20.0000 -> new: 15.0000
  FW_P_TC: curr: 0.4000 -> new: 0.5000
  FW_PR_FF: curr: 0.5000 -> new: 0.4000
  FW_PR_I: curr: 0.1000 -> new: 0.0500
  FW_PR_P: curr: 0.0800 -> new: 0.0500
  FW_R_TC: curr: 0.4000 -> new: 0.7000
  FW_RR_FF: curr: 0.5000 -> new: 0.2000
  FW_RR_I: curr: 0.1000 -> new: 0.0200
  FW_RR_P: curr: 0.0500 -> new: 0.2200
* FW_L1_PERIOD: curr: 15.0000 -> new: 12.0000
  FW_W_EN: curr: 0 -> new: 1
* MIS_LTRMIN_ALT: curr: 25.0000 -> new: 30.0000
* MIS_TAKEOFF_ALT: curr: 25.0000 -> new: 30.0000
  NAV_ACC_RAD: curr: 10.0000 -> new: 15.0000
  NAV_DLL_ACT: curr: 0 -> new: 2
  RWTO_TKOFF: curr: 0 -> new: 1
* SYS_AUTOCONFIG: curr: 1 -> new: 0
INFO  [dataman] Unknown restart, data manager file './dataman' size is 11798680 bytes
INFO  [simulator] Waiting for simulator to accept connection on TCP port 4560
Gazebo multi-robot simulator, version 9.15.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 172.16.113.146

and the robot is also not loaded

guanbin123, 2020-12-08:
I am running : make px4_sitl_default gazebo

[0/4] Performing build step for 'sitl_gazebo'
ninja: no work to do.
[3/4] cd /home/guanbin/PX4-Autopilot/b...n/PX4-Autopilot/build/px4_sitl_default
SITL ARGS
sitl_bin: /home/guanbin/PX4-Autopilot/build/px4_sitl_default/bin/px4
debugger: none
program: gazebo
model: none
world: none
src_path: /home/guanbin/PX4-Autopilot
build_path: /home/guanbin/PX4-Autopilot/build/px4_sitl_default
empty model, setting iris as default
GAZEBO_PLUGIN_PATH :/home/guanbin/PX4-Autopilot/build/px4_sitl_default/build_gazebo
GAZEBO_MODEL_PATH :/home/guanbin/PX4-Autopilot/Tools/sitl_gazebo/models
LD_LIBRARY_PATH /home/guanbin/catkin_ws/devel_isolated/test_mavros/lib:/home/guanbin/catkin_ws/devel_isolated/offb/lib:/home/guanbin/catkin_ws/devel_isolated/mavros_extras/lib:/home/guanbin/catkin_ws/devel_isolated/mavros/lib:/home/guanbin/catkin_ws/devel_isolated/mavros_msgs/lib:/home/guanbin/catkin_ws/devel_isolated/libmavconn/lib:/home/guanbin/catkin_ws/devel_isolated/learning_topic/lib:/opt/ros/melodic/lib:/home/guanbin/catkin_ws/devel_isolated/mavlink/lib:/home/guanbin/PX4-Autopilot/build/px4_sitl_default/build_gazebo
empty world, setting empty.world as default
SITL COMMAND: "/home/guanbin/PX4-Autopilot/build/px4_sitl_default/bin/px4" "/home/guanbin/PX4-Autopilot/build/px4_sitl_default"/etc -s etc/init.d-posix/rcS -t "/home/guanbin/PX4-Autopilot"/test_data
INFO  [px4] Creating symlink /home/guanbin/PX4-Autopilot/build/px4_sitl_default/etc -> /home/guanbin/PX4-Autopilot/build/px4_sitl_default/tmp/rootfs/etc

______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 0
Error: Unknown model 'iris'
**ERROR [px4] Startup script returned with return value: 256**

Jaeyoung-Lim, 2020-12-08:
@guanbin123 Your issue is not related to this issue

Jaeyoung-Lim, 2020-12-08:
@ricardorodrigues09 Have you maybe tried with gazebo11? It is very strange to fail like that

guanbin123, 2020-12-08:
Thank you, gazebo version is 9.16. I plan to reinstall it and try

发自我的iPhone

ricardorodrigues09, 2020-12-08:
@Jaeyoung-Lim I decided to install everything all over again and it is working now, thank you for your help. I have another problem but it is unrelated with this one so I'm going to create a new thread. thanks again

Jaeyoung-Lim, 2020-12-08:
Awesome! Glad that you found a way to get it work

closed 2020-12-08 by Jaeyoung-Lim