PX4/PX4-Autopilot #16277: Makefile:224: recipe for target 'px4_sitl' failed make: *** [px4_sitl] Error 1

Rich2006, 2020-11-26:


Jaeyoung-Lim, 2021-01-17:
Looks like you are missing some dependencies

Did you go through the toolchain installation as documented in http://docs.px4.io/master/en/dev_setup/dev_env_linux_ubuntu.html?

You can execute the script
```
Tools/setup/ubuntu.sh
```

asha176, 2022-10-14:
@Rich2006 Hey how did it get solved ?

Brkozshn, 2023-05-06:
[1/2] launching px4 jmavsim_iris (SYS_AUTOSTART=10017)

______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] startup script: /bin/sh etc/init.d-posix/rcS 0
env SYS_AUTOSTART: 10017
INFO  [param] selected parameter default file parameters.bson
INFO  [param] selected parameter backup file parameters_backup.bson
  SYS_AUTOCONFIG: curr: 0 -> new: 1
ERROR [param] Parameter UXRCE_DDS_KEY not found.
  SYS_AUTOSTART: curr: 0 -> new: 10017
  CAL_ACC0_ID: curr: 0 -> new: 1310988
  CAL_GYRO0_ID: curr: 0 -> new: 1310988
  CAL_ACC1_ID: curr: 0 -> new: 1310996
  CAL_GYRO1_ID: curr: 0 -> new: 1310996
  CAL_ACC2_ID: curr: 0 -> new: 1311004
  CAL_GYRO2_ID: curr: 0 -> new: 1311004
  CAL_MAG0_ID: curr: 0 -> new: 197388
  CAL_MAG0_PRIO: curr: -1 -> new: 50
  CAL_MAG1_ID: curr: 0 -> new: 197644
  CAL_MAG1_PRIO: curr: -1 -> new: 50
* SENS_BOARD_X_OFF: curr: 0.0000 -> new: 0.0000
* SENS_DPRES_OFF: curr: 0.0000 -> new: 0.0010
INFO  [dataman] data manager file './dataman' size is 7866640 bytes
etc/init.d-posix/rcS: 39: [: Illegal number: 
INFO  [init] jMAVSim simulator
INFO  [simulator_mavlink] using TCP on remote host localhost port 4560
WARN  [simulator_mavlink] Please ensure port 4560 is not blocked by a firewall.
INFO  [simulator_mavlink] Resolved host 'localhost' to address: 127.0.0.1
INFO  [simulator_mavlink] Waiting for simulator to accept connection on TCP port 4560
Buildfile: /home/burak/src/PX4-Autopilot/Tools/simulation/jmavsim/jMAVSim/build.xml

make_dirs:

compile:

create_run_jar:

copy_res:

BUILD SUCCESSFUL
Total time: 3 seconds
Options parsed, starting Sim.
Starting GUI...
Exception in thread "main" java.lang.reflect.InvocationTargetException
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.base/java.lang.reflect.Method.invoke(Method.java:566)
	at org.eclipse.jdt.internal.jarinjarloader.JarRsrcLoader.main(JarRsrcLoader.java:61)
Caused by: java.lang.UnsatisfiedLinkError: Can't load library: /usr/lib/jvm/java-11-openjdk-amd64/lib/libawt_xawt.so
	at java.base/java.lang.ClassLoader.loadLibrary(ClassLoader.java:2633)
	at java.base/java.lang.Runtime.load0(Runtime.java:768)
	at java.base/java.lang.System.load(System.java:1837)
	at java.base/java.lang.ClassLoader$NativeLibrary.load0(Native Method)
	at java.base/java.lang.ClassLoader$NativeLibrary.load(ClassLoader.java:2445)
	at java.base/java.lang.ClassLoader$NativeLibrary.loadLibrary(ClassLoader.java:2501)
	at java.base/java.lang.ClassLoader.loadLibrary0(ClassLoader.java:2700)
	at java.base/java.lang.ClassLoader.loadLibrary(ClassLoader.java:2651)
	at java.base/java.lang.Runtime.loadLibrary0(Runtime.java:830)
	at java.base/java.lang.System.loadLibrary(System.java:1873)
	at java.desktop/java.awt.Toolkit$3.run(Toolkit.java:1399)
	at java.desktop/java.awt.Toolkit$3.run(Toolkit.java:1397)
	at java.base/java.security.AccessController.doPrivileged(Native Method)
	at java.desktop/java.awt.Toolkit.loadLibraries(Toolkit.java:1396)
	at java.desktop/java.awt.Toolkit.<clinit>(Toolkit.java:1429)
	at java.desktop/java.awt.Component.<clinit>(Component.java:621)
	at me.drton.jmavsim.Simulator.<init>(Simulator.java:193)
	at me.drton.jmavsim.Simulator.main(Simulator.java:944)
	... 5 more

I have an error like this and It always keep getting when try to run $make px4_sitl jmavsim. Can anybody help ?

andres040193ec, 2023-05-11:
what if the error is still:

make: *** [Makefile:232: px4_sitl] Error 1

but you are using your own drone model and it is already in the /Tools/simulation/gazebo/sitl_gazebo/models file? any help would be great

Brkozshn, 2023-05-11:
Can anyone help me with the issue ? I'm still suffering !!! @Jaeyoung-Lim @Rich2006