PX4/PX4-Autopilot #16278: sensor_combined not always published in SITL

jlaitine0, 2020-11-26:
**Describe the bug**

I noticed that in SITL, around 75% of the simulator starts lead to sensor_combined topic not being published at all. Sometimes it works just fine.

It might be related to Multi EKF support, because disabling that seems to solve the issue completely:

	# Disable Multi-EKF
	param set EKF2_MULTI_IMU 0
	param set SENS_IMU_MODE 1
	param set EKF2_MULTI_MAG 0
	param set SENS_MAG_MODE 1

**To Reproduce**
Steps to reproduce the behavior:
1. Start SITL: HEADLESS=1 make px4_sitl_rtps gazebo_iris_rtps
2. pxh> uorb status
3. look for sensor_combined topic, many times not found at all

If the sensor_combined topic is not being published, just try to set these parameters:
	param set EKF2_MULTI_IMU 0
	param set SENS_IMU_MODE 1

and the publishing will start.

**Expected behavior**
sensor_combined topic should be present in uorb statuts

**Log Files and Screenshots**
N/A

**Drone (please complete the following information):**
SITL

**Additional context**
Add any other context about the problem here.

dagar, 2020-11-26:
What's SITL doing in these cases where `sensor_combined` isn't found? Do you see other data available and updating? Can you takeoff?

jlaitine0, 2020-11-26:
Yes, everything appears to be working normally, I can takeoff. Only reason I ever noticed this is, that I need the sensor data in ROS side (over RTPS), but that never came. Then I figured out that it is never even published on PX4 side. Here is a snippet of all sensor_* topics show with "uorb status" in failure case:

sensor_accel                          0    4  8   48 /obj/sensor_accel0
sensor_accel                          1    3  8   48 /obj/sensor_accel1
sensor_accel                          2    3  8   48 /obj/sensor_accel2
sensor_baro                           0    2  1   32 /obj/sensor_baro0
sensor_baro                           1    2  1   32 /obj/sensor_baro1
sensor_gps                            0    5  1  104 /obj/sensor_gps0
sensor_gyro                           0    4  8   40 /obj/sensor_gyro0
sensor_gyro                           1    3  8   40 /obj/sensor_gyro1
sensor_gyro                           2    3  8   40 /obj/sensor_gyro2
sensor_mag                            0    2  4   48 /obj/sensor_mag0
sensor_mag                            1    2  4   48 /obj/sensor_mag1
sensor_preflight_mag                  0    1  1   16 /obj/sensor_preflight_mag0
sensors_status_imu                    0    2  1   88 /obj/sensors_status_imu0

dagar, 2020-11-26:
Can you confirm with one of the cases where it's missing?

`listener sensor_combined`

`listener sensor_selection`

jlaitine0, 2020-11-26:
pxh> listener sensor_combined
never published
pxh> listener sensor_selection
never published
pxh> ekf2 status
INFO  [ekf2] available instances: 6
INFO  [ekf2] 0: ACC: 1310988, GYRO: 1310988, MAG: 197388, healthy, test ratio: 0.0082988 (0.00000) *
INFO  [ekf2] 1: ACC: 1310996, GYRO: 1310996, MAG: 197388, healthy, test ratio: 0.0082988 (0.07160) 
INFO  [ekf2] 2: ACC: 1311004, GYRO: 1311004, MAG: 197388, healthy, test ratio: 0.0082757 (0.81846) 
INFO  [ekf2] 3: ACC: 1310988, GYRO: 1310988, MAG: 197644, healthy, test ratio: 0.0082765 (0.84748) 
INFO  [ekf2] 4: ACC: 1310996, GYRO: 1310996, MAG: 197644, healthy, test ratio: 0.0082765 (0.84748) 
INFO  [ekf2] 5: ACC: 1311004, GYRO: 1311004, MAG: 197644, healthy, test ratio: 0.0082711 (0.28035) 

ekf2:0 attitude: 1, local position: 1, global position: 1
ekf2: ECL update: 18864 events, 0us elapsed, 0.00us avg, min 0us max 0us 0.000us rms
ekf2: ECL full update: 12258 events, 0us elapsed, 0.00us avg, min 0us max 0us 0.000us rms

ekf2:1 attitude: 1, local position: 1, global position: 1
ekf2: ECL update: 18864 events, 0us elapsed, 0.00us avg, min 0us max 0us 0.000us rms
ekf2: ECL full update: 12258 events, 0us elapsed, 0.00us avg, min 0us max 0us 0.000us rms

ekf2:2 attitude: 1, local position: 1, global position: 1
ekf2: ECL update: 18863 events, 0us elapsed, 0.00us avg, min 0us max 0us 0.000us rms
ekf2: ECL full update: 12259 events, 0us elapsed, 0.00us avg, min 0us max 0us 0.000us rms

ekf2:3 attitude: 1, local position: 1, global position: 1
ekf2: ECL update: 18863 events, 0us elapsed, 0.00us avg, min 0us max 0us 0.000us rms
ekf2: ECL full update: 12259 events, 0us elapsed, 0.00us avg, min 0us max 0us 0.000us rms

ekf2:4 attitude: 1, local position: 1, global position: 1
ekf2: ECL update: 18863 events, 0us elapsed, 0.00us avg, min 0us max 0us 0.000us rms
ekf2: ECL full update: 12259 events, 0us elapsed, 0.00us avg, min 0us max 0us 0.000us rms

ekf2:5 attitude: 1, local position: 1, global position: 1
ekf2: ECL update: 18863 events, 0us elapsed, 0.00us avg, min 0us max 0us 0.000us rms
ekf2: ECL full update: 12259 events, 0us elapsed, 0.00us avg, min 0us max 0us 0.000us rms

dagar, 2020-11-26:
Thanks, I'll think I know where to debug from here.

dagar, 2020-12-07:
This should be fixed with https://github.com/PX4/PX4-Autopilot/pull/16348.

closed 2020-12-08 by dagar