PX4/PX4-Autopilot #16290: Delay in response when sending setpoint velocities in Offboard mode

tlin442, 2020-11-27:
**Describe the bug**
When operating in OFFBOARD mode with velocity setpoints, there appears to be a delay in the order of ~150ms between publishing velocity setpoints (via /mavros/setpoint_velocity/cmd_vel_unstamped) and the command being executed by the craft. This delay is present only when reviewing timestamps from rosbag data, and does not appear in the flight logs. The same behavior also appears when running under SITL with gazebo.

I'm not 100% sure if the issue is in PX4 or MAVROS, but messages published on /diagnostics gives an average time sync RTT of 4.5ms and no TX queue overflow messages were seen. I have tested the latency between publishing of the velocity setpoint and subscription callback of MAVROS to be approx. 1ms.

**To Reproduce**
Steps to reproduce the behavior:
1. Start flying, I tested with "make px4_sitl_default gazebo" and a real craft
2. Send stream of velocity setpoints
3. Switch to offboard and log mavros messages with rosbag
4. See delay in rosbag data

**Expected behavior**
The craft should immediately start executing the velocity setpoint.

**Log Files and Screenshots**
PX4 Flight log as attached, rosbag log starts around the 1:57 mark.
https://logs.px4.io/plot_app?log=abdb7516-45a5-4c40-a75e-0a19f4b41c72

The following graphs are all time synced with the same T=0 point.
Screenshots of velocity response from rosbag (note, X in LOCAL_ENU frame), there is a very apparent delay of approx. 150ms between the first change in setpoint and the first response of the craft, local velocity estimates from PX4 seem to agree about this delay.
[image]

Pitch response of craft (forward direction of craft initially aligned with pitch), note the very sharp change in pitch at T=0.15s:
[image]

The same maneuver from the PX4 flight log, there doesn't appear to be any delays here:
[image]

Velocity response using the 3DR Iris in SITL:
[image]

Please let me know if you would like the rosbag(s) and how I should get them to you!

**Drone (please complete the following information):**
PixRacer flight controller + Onboard Jetson Nano
Custom 250 Frame
Intel Realsense T265 sending Odometry information via MAVROS

PX4 v1.11.1

**Additional context**

Jaeyoung-Lim, 2020-12-07:
@tlin442 What is the communication delay on your link? How is the flight controller connected to the Jetson Nano?

tlin442, 2020-12-07:
@Jaeyoung-Lim I've tried measuring the ping on the link by enabling gcs comms on MAVROS, and checking the ping via the console. The ping through this route (QGC->Ethernet->Jetson Nano->MAVROS->Hardware Serial Port (/dev/ttyTHS1)->Pixracer) is about 15ms on average with all software running.

The flight controller is connected through the Jetson's hardware serial port (921600 8N1, Onboard mode, max data rate set to 0 for link speed/2).

The issue is still present when running SITL with Gazebo on an i7-3630QM laptop.

Jaeyoung-Lim, 2020-12-07:
@tlin442 The ping is not related to your latency between `MAVROS->Hardware Serial Port (/dev/ttyTHS1)->Pixracer)`

> The issue is still present when running SITL with Gazebo on an i7-3630QM laptop.

Are you saying you have the exact same latency also in SITL?

tlin442, 2020-12-07:
@Jaeyoung-Lim Oh, sorry, what would be the best way to test for serial port latency

And yes, the same latency exists in SITL, I believe it's actually slightly longer at ~200ms in SITL.

Jaeyoung-Lim, 2020-12-07:
@tlin442 I still think those two are different issues.

You can try and send something and reply back and check the time delay.

programmargorp, 2020-12-07:
@Jaeyoung-Lim I will investigate the serial link latency tomorrow.

All graphs above are timestamped to the commanded time from when my node emits a velocity setpoint. I do expect delays in this up to maybe 50-80ms, but not 100+ms.

Jaeyoung-Lim, 2020-12-07:
@programmargorp Okay, then first you need to verify that the delay is not being caused by mavros. You need to profile every step to reach the conclusion that the delay is coming from the PX4 Firmware itself. Then we can narrow it down and try and investigate.

tlin442, 2020-12-07:
@Jaeyoung-Lim Thanks for your help so far. I've just profiled the latency of each component using chrono::high_resolution_clock on the Jetson in both my code and MAVROS, the latencies in each step of the link are as follows:
My Node -> MAVROS - ~1ms publish to subscribe.
MAVROS -> Serial TX queue empty - ~1.5ms from subscribe to tx queue empty
MAVROS -> PX4 RTT (via /mavros/timesync_status) - anywhere between 1 to 15ms, hovering mostly around the 5ms mark.

Please let me know if I'm missing anything.