PX4/PX4-Autopilot #16300: Quat_reset large attitude estimate jumps

JMshare, 2020-11-29:
I am seeing some large attitude estimate jumps during our fixed-wing flights, see bellow.
[image]

I found out that they match the `quat_reset_counter` changes in the `vehicle_attitude` topic, but I can't say what triggers them.

Now, I can live with the small jumps in pitch and roll, but 50 degree yaw jump if you are using differential thrust for yaw control is not  great. The next few seconds after each jump is the reaction of the plane trying to get back to stable.

Nothing extraordinary seen in the `vehicle_magnetometer` data that should be triggering this I think.
Log [https://review.px4.io/plot_app?log=6a0e6bec-c709-4d34-9ba2-031072c61098](url)

I am using the Pixhawk4 with the default UBLOX NEO M8N GPS module. Firmware v1.9.0.

Do you know why this is happening and how to prevent it?

JMshare, 2020-12-07:
Hello,

is this really not considered an issue here? I have at least one of these jumps in every single flight log from May 2019
[image]

dagar, 2020-12-07:
There are quite a few faults in that log.

[image]

The magnetic heading innovations also look questionable.

[image]

You said your based on PX4 v1.9.0? As a start I'd highly recommend updating to v1.11.2 or even current master (soon to be v1.12). The main estimator (ekf2) has improved considerably in the past couple years.

JMshare, 2020-12-07:
Thanks Daniel, I will try that.

mrpollo, 2020-12-16:
Let us know how it goes @JMshare feel free to re-open the issue once you check with the latest PX4

closed 2020-12-16 by mrpollo