PX4/PX4-Autopilot #16305: [VTOL] 0° yaw/heading after back transition in mission

ThomasRigi, 2020-11-30:
**Describe the bug**
In mission mode, just after a back transition the VTOL (quadplane) always yaws towards due north. This 0° yaw setpoint only changes once the drone moves on to the next waypoint. 

Present in v1.11.2 and on current master. In SITL and on a real drone.

**To Reproduce**
Steps to reproduce the behavior:
1. Drone switched on. Both `make px4_sitl gazebo_standard_vtol` and a real drone
2. Uploaded mission: 
[RC club many transitions.zip](https://github.com/PX4/PX4-Autopilot/files/5616756/RC.club.many.transitions.zip)
3. Start the mission
4. Observe the yaw after back transitions.

**Expected behavior**
The drone should either
- keep its current heading 
- use the heading as calculated normally in the MPC trajectory generation
- weathervane
like in v1.10.1 : https://logs.px4.io/plot_app?log=9ba19728-7ba2-4250-8884-e39e3c6a1d7f

**Log Files and Screenshots**
-Real quadplane: https://logs.px4.io/plot_app?log=3b4e3749-8b0e-44a2-9e9b-74b329b3dbab (ignore after 6:00)
-SITL, v1.11.2: https://logs.px4.io/plot_app?log=21bdc8c9-9cfc-4964-90aa-2f123ebb70f6
-SITL, master: https://logs.px4.io/plot_app?log=cc6301a2-0a6f-4100-a973-128cb7784ddf
-SITL, v1.11.2, other flight modes: https://logs.px4.io/plot_app?log=74ae49e6-1d82-4a2f-81ee-2fb387cacd1c (no enforced 0° yaw)

Useless 360 after back transition: 
[image]

**Drone (please complete the following information):**
- Quadplane with Pixhawk 4, v1.11.2
- Gazebo Standard VTOL

**Additional context**
I could only reproduce this in mission mode (see 4th log).

moreba1, 2020-12-03:
In my test with 1.11.1 version MPC_YAW_MODE don't works for standard quadplane VTOL in multirotor mode. in my opinion it is better that this parameter activated for VTOL , so everyone can set it.
Yaw mode
Values:
    0: towards waypoint
    1: towards home
    2: away from home
    3: along trajectory
    4: towards waypoint (yaw first)

ThomasRigi, 2020-12-04:
After some git bisect I could pin down the change to https://github.com/PX4/PX4-Autopilot/commit/574b482fdb18662c9eabd5a4ace79fe20a9c94bd @sfuhrer @dagar 

The weathervane was only reactivated after a VTOL Takeoff & Transition command, not after a plain VTOL transition command.

Fix in https://github.com/PX4/PX4-Autopilot/pull/16335

moreba1, 2020-12-04:
yes , it is correct.

closed 2020-12-09 by sfuhrer