PX4/PX4-Autopilot #16308: Problem mavlink communication between FMU and a companion computer

dmdobrea, 2020-12-01:
Based on the MAVLink Messaging tutorial (https://dev.px4.io/v1.10/en/middleware/mavlink.html), I followed all the step, but when I compile the code I get the error:

error: ‘MavlinkOrbSubscription’ does not name a type 

in mavlin_messages.c.

Searching I found that the MavlinkOrbSubscription class is included in "the mavlink_orb_subscription.h" file. But, I do not have a such file in the PX4 master release.

The file can be founded in 1.10.2 version. 

So, plese can you update the 1.11.2 version with all the components? 

Thank you very much!!!

Dobrea Dan

antonio-sc66, 2020-12-01:
Looks like the documentation hasn't been updated. Use uORB::Subscription instead of MavlinkOrbSubscription

dmdobrea, 2020-12-02:
Thank you, very much! I am working now.

igalloway, 2020-12-02:
@Hamish Willee <hamishwillee@gmail.com>

hamishwillee, 2020-12-02:
Thanks @igalloway - this is being tracked in https://github.com/PX4/PX4-user_guide/pull/932

antonio-sc66, 2020-12-03:
@dmdobrea The tutorial should be working now, check https://docs.px4.io/master/en/middleware/mavlink.html

closed 2021-01-01 by Jaeyoung-Lim