PX4/PX4-Autopilot #16309: Problem about px4 controlling the servo

joker-csee, 2020-12-01:
Hi there,

Here comes a problem when I try to control the servo with px4. 

I just modified the .main.mix file related to the mixer and added one more mainout channel. Then plugged the servo cable into the corresponding output port. But once the power is connected, the steering gear is fully deflected, and no matter how I shake the remote control lever, the steering gear remains in a dead state.

Can someone please explain me what's the problem here? 

Thanks a lot!!!

yuthikasagarage, 2020-12-01:
do you want to do a rc pass-through or use it on the mission?

joker-csee, 2020-12-01:
> do you want to do a rc pass-through or use it on the mission?

I want to do a rc pass-through. Specifically, the remote control is used to control the swing angle of the servo.

yuthikasagarage, 2020-12-01:
did you enable the rc pass through channel in qgc? aux1 or aux2 is supported for rc-passthrough with mixers like the one below.
also (select RC channel with RC_MAP_AUX1 param or others if you are using a custom mixer)
https://github.com/PX4/PX4-Autopilot/blob/master/ROMFS/px4fmu_common/mixers/quad_x.main.mix

joker-csee, 2020-12-01:
> did you enable the rc pass through channel in qgc? aux1 or aux2 is supported for rc-passthrough with mixers like the one below.
> also (select RC channel with RC_MAP_AUX1 param or others if you are [...]

No I didn't. But I tried to connect the servo with aux and it worked, while not changing any param on qgc. How to enable the rc pass through in qgc?

Is it because that the defult params of mainout channel is set to fit motors, not servo?

yuthikasagarage, 2020-12-01:
in quad_x_main.mix the aux1 and aux2 is configured as passthrough. what i meant in qgc was the 2 tick boxes to enable aux1, aux2 passthrough. yeah main channels usually accommodate motors.  if the problem is resolved you can close it.

joker-csee, 2020-12-02:
> in quad_x_main.mix the aux1 and aux2 is configured as passthrough. what i meant in qgc was the 2 tick boxes to enable aux1, aux2 passthrough. yeah main channels usually accommodate motors. if the problem is resolved you can close it.

Thanks for that, I saw the options in qgc.

But I need to control servos with main channels.What should I do for that? Do I need to change somewhere in the source code?

bresch, 2020-12-14:
@joker-csee Could you please describe a bit more what you are trying to do? Is it for a multirotor, a plane, a VTOL plane, a rover? Which autopilot hardware are you using? (Pixhawk4? Pixracer? ..) Can you show how you modified the mixer file?

joker-csee, 2020-12-15:
> @joker-csee Could you please describe a bit more what you are trying to do? Is it for a multirotor, a plane, a VTOL plane, a rover? Which autopilot hardware are you using? (Pixhawk4? Pixracer? ..) Can you show how you modified the mixer file?

Thanks for your reply.

I'm trying to control a multirotor with servos on it to make motors swinging like 'tilt quadrotor', using **pixhawk1** with latest v1.11.2 firmware px4_fmu-v2_default.px4. Because the num of servos is **more** than aux channels', I have to use **mainout** channels to control the **servos** out of range. To test this feature, I added the following codes into PX4-Autopilot/ROMFS/px4fmu_common/mixers/**tilt_quad.main.mix** to response merely the yaw channel singal. (But 'tilt quadrotor' model is not supported on pixhwak1, so I used another 'dji f450' model with the same codes instead.) And I also chose the corresponding model on QGC.
```
servo1
M: 1
S: 0 2   10000   10000      0 -10000  10000

servo2
M: 1
S: 0 2  -10000  -10000      0 -10000  10000

servo3
M: 1
S: 0 2  -10000  -10000      0 -10000  10000

servo4
M: 1
S: 0 2   10000   10000      0 -10000  10000
```

The RESULT is that once the power is connected, all servos are fully deflected, and out of control. Though it works when I add the changes into tilt_quad.**aux**.mix using aux channels instead, it's not what I want.

mrpollo, 2020-12-15:
@dagar do you have any suggestions here?

RicardoM17, 2021-01-18:
@joker-csee do you still need help with this? The tilt quad frame that you mentioned was done by me. Unfortunately I only had to use 4 servos.

What exactly is your issue? Can you share a log?

joker-csee, 2021-01-18:
> @joker-csee do you still need help with this? The tilt quad frame that you mentioned was done by me. Unfortunately I only had to use 4 servos.
> 
> What exactly is your issue? Can you share a log?

Hi, Marques! I am reading your dissertation nowadays, it's really helpful!

I tried to use mainout channels to control servos like what you have done with aux channels. But problem came up that the servos were fully deflected once powered on. Do I need to change the signal frequency to fit servos in source code, or change other parameters to fix the problem?

Unfortunately, there is no saved log.

Thanks for reply. Any help would be greatful.

RicardoM17, 2021-01-18:
Hi @joker-csee it's really interesting to see other people doing projects similar to mine.

This is probably a mixing issue, or some other parameter that you have to change. I'm happy to help but there probably needs to be quite a lot of back and forth so maybe reach out to me on the PX4 slack directly and we can discuss there.

Name: Ricardo Marques
User ID: U94PHPLQM

Then once (or if) the problem is fixed you can post the solution here so the issue can be closed, but other people can find it, if they need it.

joker-csee, 2021-01-21:
one can control servos on main channels 5-8 with this mixer code in a '+' quadrotor airframe, responding to the change of yaw.

# quad +
R: 4+ 10000 10000 10000 0
# mixer of other four mainouts to respond yaw
M: 1
S: 0 2  10000  10000      0 -10000  10000
M: 1
S: 0 2 -10000 -10000      0 -10000  10000
M: 1
S: 0 2 -10000 -10000      0 -10000  10000
M: 1
S:  2  10000  10000      0 -10000  10000

pay attention not to add the servo-control channels into parameter 'PWN_OUT' in airframe file, which will set the signal rate to 400hz on these channels.

closed 2021-01-21 by joker-csee

ahmedfarouk99, 2024-06-29:
@joker-csee @bresch @RicardoM17 
here is my setup:
1-i'm using f450 generic quad-rotor with pixhawk-4 and px4 firmware
2-i'm using aux pwm instead of main to decrease letancy according to px4 documentation
3- i connected my 4 motors to first 4 pwm aux pins
4- i changed parameter to use actuator tab instead of motor tab and i assigned the 4 motors to pwm aux (1-4)
5- i tested my f450 and everthing works great
6- i need to use an external servo, so i assigned it to pwm aux pin-5 and 5v external battery
7- i added servo-1 in the actuator tab as pwm aux-5
8- once i added this servo , it appears in the scroller tab next to my 4 motors used ,and i'm able to manually control the 4 motors and the servo through the scroller to check if they are working good, and all of them were working good
9- i needed to assign servo-1 used to rc channel 5 , so i modified " rc_map_aux5=channel 5" ,but when i used rc channel 5 , nothing happened....
can anyone help me please...
thanks in advance