PX4/PX4-Autopilot #16316: px4 non-existent target "platforms__common"

weiyuhe, 2020-12-02:
I was following the tutorial _Sending a Custom Message from MAVROS to PX4_ from [https://dev.px4.io/master/en/ros/mavros_custom_messages.html](url)

I am getting an error **px4 non-existent target "platforms__common"** when I catkin build.

I am using melodic and ROS 18.04. I am running px4 in SITL.

Any suggestion?

Jaeyoung-Lim, 2020-12-06:
@weiyuhe Do you have the firmware under your catkin_ws? If you do, this is not the correct location to put the firmware

weiyuhe, 2020-12-06:
@Jaeyoung-Lim Yes, I did have the Firmware folder inside my catkin_ws. Could you explain why that is a bad idea? Thank you.

Jaeyoung-Lim, 2020-12-06:
@weiyuhe catkin assumes that you have a specific directory structure, which the PX4-Autopilot repo doesn't follow. 

That is why the documentation points you to not clone it inside the catkin workspace: https://dev.px4.io/master/en/simulation/ros_interface.html

closed 2020-12-15 by Jaeyoung-Lim