PX4/PX4-Autopilot #16328: Ambiguous reference frame for accelerometer measurements in /msg/sensor_accel.msg

rblilja, 2020-12-03:
File: /msg/sensor_accel.msg 

Description:

The comments of the x, y, z variables makes little sense in terms of the frame of reference, e.g. for the x axis: "# acceleration in the NED X board axis in m/s^2"

1. Unclear if NED refers to the actual navigation frame. Then mentioning of "board" makes no sense.
2. If NED is incorrect. Unclear if "board" refers to the sensor frame or body frame of a circuit board?

LorenzMeier, 2020-12-03:
The sensor measurements get transformed into the board's NED frame and so the description is correct and accurate.

closed 2020-12-03 by LorenzMeier

rblilja, 2020-12-03:
To me it makes little sense that the board can have a NED frame. Possibly a NED frame attached in e.g. the COG of the board. What I'm getting at is that it shall simply be the NED frame (which in literature often is called the navigation frame). 

Looking in https://dev.px4.io/v1.11/en/contribute/notation.html, it mentions FRD and NED. So cleaner to just refer to those?

It makes me wonder. It is actually not the FRD frame the accelerations are transformed to? Then I understand your argument.

bresch, 2020-12-04:
> It makes me wonder. It is actually not the FRD frame the accelerations are transformed to?

@rblilja Yes, it's FRD. Could you please make a PR to fix that please?

rblilja, 2020-12-04:
@bresch Sure. No problem. Thanks for ironing this one out!

rblilja, 2020-12-06:
@bresch Made a PR. It is my first time contributing - so please correct me if I did any mistakes. Not sure how I link a PR with an issue, but the system seems to have picked it up above :)