PX4/PX4-Autopilot #16334: PixRacer and HereFlow UAVCAN driver

taileron, 2020-12-04:
**Describe the bug**
PixRacer doesn´t recognise UAVCAN devices like HereFlow
various devices work perfect with X2.1-777 (e.g. HereFlow as range and flow on id 125)

**To Reproduce**
Steps to reproduce the behavior:
UAVCAN parameters successfully enabled 
dmesg enabled to display start procedure
uavcan status displayed
firstly I thought the "china" PixRacer has a pcb design issue but a new r15 one directly from MRO behaves identical
the cirquit diagram shows CL and CH pins exchanged between X2.1 and PixRacer so I tried it to change them in the sensor  

**Expected behavior**
flow and range/data should appear in QGC

**Log Files and Screenshots**
nsh> uavcan status
Pool allocator status:
	Capacity hard/soft: 500/250 blocks
	Reserved:  30 blocks
	Allocated: 4 blocks

UAVCAN node status:
	Internal failures: 0
	Transfer errors:   0
	RX transfers:      0
	TX transfers:      1653

CAN1 status:
	HW errors: 795
	IO errors: 833
	RX frames: 87
	TX frames: 1892

control latency: 0 events, 0us elapsed, 0.00us avg, min 0us max 0us 0.000us rms
INFO  [mixer_module] Switched to rate_ctrl work queue: 0
INFO  [mixer_module] Mixer loaded: no
INFO  [mixer_module] Driver instance: 0
INFO  [mixer_module] Channel Configuration:
INFO  [mixer_module] Channel 0: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
INFO  [mixer_module] Channel 1: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
INFO  [mixer_module] Channel 2: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
INFO  [mixer_module] Channel 3: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
INFO  [mixer_module] Channel 4: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
INFO  [mixer_module] Channel 5: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
INFO  [mixer_module] Channel 6: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
INFO  [mixer_module] Channel 7: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
INFO  [mixer_module] Channel 8: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
INFO  [mixer_module] Channel 9: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
INFO  [mixer_module] Channel 10: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
INFO  [mixer_module] Channel 11: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
INFO  [mixer_module] Channel 12: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
INFO  [mixer_module] Channel 13: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
INFO  [mixer_module] Channel 14: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
INFO  [mixer_module] Channel 15: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191

Sensor 'baro':
devname: /dev/uavcan/baro
channel 0: empty
channel 1: empty
channel 2: empty
channel 3: empty

Sensor 'mag':
devname: /dev/uavcan/mag
channel 0: empty
channel 1: empty
channel 2: empty
channel 3: empty

Sensor 'gnss':
devname: /dev/gnss
channel 0: empty
channel 1: empty
channel 2: empty
channel 3: empty
uavcan: gnss: rtcm pub: 0 events, 0.00us avg, min 0us max 0us 0.000us rms

Sensor 'flow':
devname: /dev/flow
channel 0: empty
channel 1: empty
channel 2: empty
channel 3: empty

Sensor 'battery':
devname: /dev/battery
channel 0: empty
channel 1: empty
channel 2: empty
channel 3: empty

Sensor 'airspeed':
devname: /dev/airspeed
channel 0: empty
channel 1: empty
channel 2: empty
channel 3: empty

Sensor 'differential_pressure':
devname: /dev/airspeed
channel 0: empty
channel 1: empty
channel 2: empty
channel 3: empty

Sensor 'rangefinder':
devname: /dev/range_finder
channel 0: empty
channel 1: empty
channel 2: empty
channel 3: empty

Online nodes (Node ID, Health, Mode):

uavcan: cycle time: 24877 events, 706708us elapsed, 28.41us avg, min 14us max 11330us 106.940us rms
uavcan: cycle interval: 24877 events, 2769.66us avg, min 32us max 155828us 1326.074us rms

(here is GCC 7.2.1 but with GCC 9.3.1 it behaves identical)
nsh> dmesg
sercon: Registering CDC/ACM serial driver
sercon: Successfully registered the CDC/ACM serial driver
HW arch: PX4_FMU_V4
FW git-hash: 615333badb289aa77285bcfad071cc6a70b13632
FW version: 1.11.0 0 (17498112)
FW git-branch: master
OS: NuttX
OS version: Release 10.0.0 (167772415)
OS git-hash: 904a602c74dc08a100b5c2bd490807de19e73e10
Build datetime: Nov 20 2020 19:08:18
Build uri: localhost
Toolchain: GNU GCC, 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]
PX4GUID: 00010000000036313737333951050051002d
MCU: STM32F42x, rev. V
INFO  [param] selected parameter default file /fs/mtd_params
Board defaults: /etc/init.d/rc.board_defaults
INFO  [dataman] Unknown restart, data manager file '/fs/microsd/dataman' size is 362560 bytes
Board sensors: /etc/init.d/rc.board_sensors
ms5611 #0 on SPI bus 2 (devid=0x3d)
WARN  [SPI_I2C] hmc5883: no instance started (no device on bus?)
lis3mdl #0 on SPI bus 1 (devid=0x5)
icm20602 #0 on SPI bus 1 (devid=0x38)
mpu9250 #0 on SPI bus 1 (devid=0x24)
INFO  [uavcan] Node ID 1, bitrate 1000000
INFO  [uavcan] sensor bridge 'baro' init ok
INFO  [uavcan] sensor bridge 'mag' init ok
INFO  [uavcan] sensor bridge 'gnss' init ok
INFO  [uavcan] sensor bridge 'flow' init ok
INFO  [uavcan] sensor bridge 'battery' init ok
INFO  [uavcan] sensor bridge 'airspeed' init ok
INFO  [uavcan] sensor bridge 'differential_pressure' init ok
INFO  [uavcan] sensor bridge 'rangefinder' init ok
ERROR [uavcan] CentralizedServer init: -13
ERROR [uavcan] Node init failed: -13
ERROR [uavcan] Firmware Server Failed to Start -13
Board extras: /etc/init.d/rc.board_mavlink
INFO  [mavlink] mode: Config, data rate: 800000 B/s on /dev/ttyACM0 @ 57600B
INFO  [mavlink] mode: Normal, data rate: 20000 B/s on /dev/ttyS0 @ 921600B
Starting Main GPS on /dev/ttyS3
Starting MAVLink on /dev/ttyS1
INFO  [mavlink] mode: Normal, data rate: 20000 B/s on /dev/ttyS1 @ 921600B
INFO  [init] Mixer: /etc/mixers/vtol_convergence.main.mix on /dev/pwm_output0
ekf2 [581:100]
Board extras: /etc/init.d/rc.board_extras
INFO  [logger] logger started (mode=all)

NuttShell (NSH) NuttX-10.0.0
nsh> INFO  [ecl/EKF] reset position to last known position
INFO  [ecl/EKF] reset velocity to zero
INFO  [ecl/EKF] 13508753: EKF aligned, (baro hgt, IMU buf: 18, OBS buf: 14)
INFO  [mavlink] Starting mavlink shell

LorenzMeier, 2021-01-01:
Do you have an update on this?

taileron, 2021-01-02:
@LorenzMeier I was able to do some more research, the problem arises as soon as telemetry 1 or 2 is used ... I always had one of them in use, of course. The PixRacer only recognises UAVCAN when TEL 1 and TEL 2 are switched off. If one of the telemetry UARTS is activated, UAVCAN also blocks the nsh shell.

taileron, 2021-01-03:
@LorenzMeier For me, the deactivation of the ESP 01, which can be plugged into the PixRacer, was the final solution (mavlink start in rc.board_mavlink instead of TTXS0 -> TTYS2). A single Mavlink instance is tolerated by UAVCAN and it does not matter whether the telemetry unit is connected or not. Maybe there are not enough resources to use everything at the same time.

LorenzMeier, 2021-01-03:
This basically means that you are running out of RAM.

taileron, 2021-01-13:
@LorenzMeier Unfortunately, another round is now coming ... with the (China) PixRacer, uavcan works when the autostart Mavlink instances are removed. Unfortunately, with the current MRO PixRacer, uavcan can no longer be made to work at all.
In principle, only the sensors and the MCU version are different. 

PixRacer (China)

PX4GUID: 0001000000003131373130365117004f0031
MCU: STM32F42x, rev. 3

ms5611 #0 on SPI bus 2 (devid=0x3d)
hmc5883 #0 on SPI bus 1 (devid=0x1)
icm20608g #0 on SPI bus 1 (devid=0x3a)
mpu9250 #0 on SPI bus 1 (devid=0x24)
INFO  [uavcan] Node ID 1, bitrate 1000000
uavcan_flow channel 124 class instance 0 ok
uavcan_rangefinder adding channel 124...
uavcan_rangefinder channel 124 class instance 0 ok

free
             total       used       free    largest
Umem:       223360     210688      12672      10816
Prog:      2097152    1835008     262144          0

PixRacer (MRO)

PX4GUID: 00010000000036313737333951050051002d
MCU: STM32F42x, rev. V

ms5611 #0 on SPI bus 2 (devid=0x3d)
lis3mdl #0 on SPI bus 1 (devid=0x5)
icm20602 #0 on SPI bus 1 (devid=0x38)
mpu9250 #0 on SPI bus 1 (devid=0x24)
INFO  [uavcan] Node ID 1, bitrate 1000000
ERROR [uavcan] CentralizedServer init: -13
ERROR [uavcan] Node init failed: -13
ERROR [uavcan] Firmware Server Failed to Start -13

free
             total       used       free    largest
Umem:       223360     195664      27696      24832
Prog:      2097152    1835008     262144          0

taileron, 2021-01-19:
I found it after all - it was a small file error on the SD media and when I repaired it on the Mac, the usual index folders were created again. Checked the SD card in a Windows computer for file errors and repaired it and deleted the Mac index folders. 
After that, UAVCAN worked.  
Please remove lines 113 - 137 from RCS again, as they cause the real problem to be overlooked.

taileron, 2021-04-06:
In the end, there is always just not enough free ram. Depending on the problems with the file system or my larger mixers. If it is reduced at some point, it works. However, something always has to be removed in the code, because only the Mavlink USB connection or the Mavlink shell is sufficient for nothing to work and in order to be able to use only USB for monitoring the UAVCAN components, many other things have to be removed in the code beforehand.

dagar, 2021-04-06:
The uavcan driver/module size is increasing because it always allocates everything. We should add a parameter so that you can be more selective about which subscribers (sensors) and publishers (ESCs, beep, etc) are active. https://github.com/PX4/PX4-Autopilot/tree/master/src/drivers/uavcan/sensors

For mavlink USB autostart potentially helps in some situations. https://github.com/PX4/PX4-Autopilot/pull/16180
Without USB connected it saves a ton of memory by not running the mavlink USB instance or the lower level support (sercon, etc). Obviously you'll still get into trouble when trying to use both Mavlink USB + heavy UAVCAN simultaneously, but at least it makes real usage viable.

taileron, 2021-04-16:
flow is only monitored by USB not by wifi telemetry If that were not the case, USB would not be necessary at all.
Is it intentional that optical flow via telemetry is not monitored?
UAVCAN_ENABLE = Sensors Autom Conf still blocks further functions like logging or pwm output so does´t left enough resources for the rest.

Would it help to start individual sensors manually e.g. for testing in the mvlnk console to save ram-space.
How would be the complete syntax to start flow and rangefinder in this case (HereFlow):
Sensor 'flow':
name: uavcan_flow
channel 0: node id 125 --> instance 0
Sensor 'rangefinder':
name: uavcan_rangefinder
channel 0: node id 125 --> instance 0

taileron, 2021-04-28:
@dagar with only one mavlink instance I could do flight tests with UAVCAN and PixRacer. But then there is no way to get the logging simultaneously. With 4 k log buffer the log process doesn´t start at all, with 9 k it starts but always stops at some point and then the SD file system is no longer there. #17442 
The buffer scheme of 1.12 also requires somewhat faster SD cards, the 1Gb that came with the PicRacers that were good enough for years definitely don't work anymore.

dagar, 2021-04-28:
> @dagar with only one mavlink instance I could do flight tests with UAVCAN and PixRacer.

We should make UAVCAN more configurable so that it's not handling everything by default. https://github.com/PX4/PX4-Autopilot/blob/d62204856cf0711af9ca82a399aac529783c412b/src/drivers/uavcan/sensors/sensor_bridge.cpp#L58-L78

How much memory does `uavcan shrink` save?

taileron, 2021-05-12:
#17533 solved all issues about losing the file system. The logging procedure itself now works again with all SD cards.
There are now several successful ways to free up enough of the F4xx's little ram.
However, there remains an insurmountable problem, with 4K log buffer (reduced when UAVCAN is activated), the logger does not start at all:
logger status
INFO  [logger] Running in mode: all
INFO  [logger] Number of subscriptions: 150 (4800 bytes)
INFO  [logger] Full File Logging Running:
INFO  [logger] Log file: /fs/microsd/log/2021-05-12/16_55_26.ulg
INFO  [logger] Wrote 17.17 KiB (avg  0.14 KiB/s)
INFO  [logger] Since last status: dropouts: 0 (max len: 0.000 s), max used buffer: 0 / 4396 B
nsh> logger status
INFO  [logger] Running in mode: all
INFO  [logger] Number of subscriptions: 150 (4800 bytes)
INFO  [logger] Full File Logging Running:
INFO  [logger] Log file: /fs/microsd/log/2021-05-12/16_55_26.ulg
INFO  [logger] Wrote 17.17 KiB (avg  0.13 KiB/s)
INFO  [logger] Since last status: dropouts: 0 (max len: 0.000 s), max used buffer: 0 / 4396 B
nsh> 

It starts to work with the setting 5k and with 6k there are almost no dropouts:

taileron, 2021-05-14:
https://review.px4.io/plot_app?log=8f164e5c-26b1-41d0-a0db-658fdab419ac

UAVCAN reduced (sensor bridge) to flow and range, log buffer set to 6K 
All uavcan battery has to be removed in sensor bridge manually as well. As soon as uavcan is enabled, the (analog) power module gives wrong values. 
With this conditions it is still needed to remove USB (serdis) to have enough ram.
2k console buffer is used for dmesg
uart 1 and 3 tx buffer 1200 (some of my Picracers use ESP8266 at telemetry 2)

taileron, 2021-05-14:
very similar with:
PixFalcon UAVCAN disabled:
free
             total       used       free    largest
Umem:       230176     181952      48224      46752
Prog:      2097152    1966080     131072          0

PixFalcon UAVCAN enabled(2)shrinked to range+flow:
free
             total       used       free    largest
Umem:       230176     217120      13056      12720
Prog:      2097152    1966080     131072          0

uavcan status
Pool allocator status:
    Capacity hard/soft: 500/250 blocks
    Reserved:  151 blocks
    Allocated: 55 blocks

UAVCAN node status:
    Internal failures: 0
    Transfer errors:   0
    RX transfers:      0
    TX transfers:      1440

CAN1 status:
    HW errors: 21402
    IO errors: 22768
    RX frames: 0
    TX frames: 224
CAN2 status:
    HW errors: 21398
    IO errors: 22764
    RX frames: 0
    TX frames: 224

control latency: 0 events, 0us elapsed, 0.00us avg, min 0us max 0us 0.000us rms
INFO  [mixer_module] Switched to rate_ctrl work queue: 0
INFO  [mixer_module] Mixer loaded: no
INFO  [mixer_module] Driver instance: 0
INFO  [mixer_module] Channel Configuration:
INFO  [mixer_module] Channel 0: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
INFO  [mixer_module] Channel 1: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
INFO  [mixer_module] Channel 2: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
INFO  [mixer_module] Channel 3: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
INFO  [mixer_module] Channel 4: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
INFO  [mixer_module] Channel 5: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
INFO  [mixer_module] Channel 6: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
INFO  [mixer_module] Channel 7: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
INFO  [mixer_module] Channel 8: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
INFO  [mixer_module] Channel 9: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
INFO  [mixer_module] Channel 10: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
INFO  [mixer_module] Channel 11: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
INFO  [mixer_module] Channel 12: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
INFO  [mixer_module] Channel 13: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
INFO  [mixer_module] Channel 14: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
INFO  [mixer_module] Channel 15: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191

Sensor 'flow':
name: uavcan_flow
channel 0: empty
channel 1: empty
channel 2: empty
channel 3: empty

Sensor 'rangefinder':
name: uavcan_rangefinder
channel 0: empty
channel 1: empty
channel 2: empty
channel 3: empty

Online nodes (Node ID, Health, Mode):

uavcan: cycle time: 19504 events, 1418993us elapsed, 72.75us avg, min 15us max 6220us 159.212us rms
uavcan: cycle interval: 19504 events, 2967.72us avg, min 44us max 14303us 717.062us rms

PixFalcon UAVCAN enabled(3)range+flow+actuators:
free
             total       used       free    largest
Umem:       230176     220016      10160       9328
Prog:      2097152    1966080     131072          0

taileron, 2021-07-04:
#16180 brings by far the best result 25k free ram with all necessary modules active.
https://review.px4.io/plot_app?log=df3b8469-d0b6-42a1-ad26-29a6ddc9744f

taileron, 2021-07-05:
#16180 hereflow flight test
https://review.px4.io/plot_app?log=b10a8303-e03b-486d-9fa3-8adecaee8bfe

taileron, 2021-09-28:
with #16180 merged this issue can be closed

taileron, 2021-10-05:
#16180 now merged

dagar, 2021-10-05:
Thanks for the followup @taileron.

taileron, 2021-10-20:
pr-uavcan_parameters currently doesn't compile successful - but I think the idea for it is great

After sensor_bridge.cpp manual disabled unused sensors, reduce all uart buffers expect the wifi telemetry to 300
disabled autotune, osd, dshot, sd-stress it can be flown again

free                    total       used       free    largest  nused  nfree         Umem:     231584     216688      14896      14576    905     19         Prog:    2097152    1835008     262144          0     23      2

uavcan status Pool allocator status:     Capacity hard/soft: 500/250 blocks     Reserved:  79 blocks     Allocated: 13 blocks  UAVCAN node status:     Internal failures: 0     Transfer errors:   0     RX transfers:      269     TX transfers:      3360  CAN1 status:     HW errors: 43147     IO errors: 46141     RX frames: 1034     TX frames: 933  ESC outputs: INFO  [mixer_module] Param prefix: UAVCAN_EC control latency: 0 events, 0us elapsed, 0.00us avg, min 0us max 0us 0.000us rms INFO  [mixer_module] Mixer loaded: no INFO  [mixer_module] Driver instance: 0 Channel Configuration: Channel 0: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191 Channel 1: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191 Channel 2: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191 Channel 3: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191 Channel 4: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191 Channel 5: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191 Channel 6: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191 Channel 7: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191 Servo outputs: INFO  [mixer_module] Param prefix: UAVCAN_SV control latency: 0 events, 0us elapsed, 0.00us avg, min 0us max 0us 0.000us rms INFO  [mixer_module] Mixer loaded: no INFO  [mixer_module] Driver instance: 0 Channel Configuration: Channel 0: value: 0, failsafe: 0, disarmed: 0, min: 0, max: 0 Channel 1: value: 0, failsafe: 0, disarmed: 0, min: 0, max: 0 Channel 2: value: 0, failsafe: 0, disarmed: 0, min: 0, max: 0 Channel 3: value: 0, failsafe: 0, disarmed: 0, min: 0, max: 0 Channel 4: value: 0, failsafe: 0, disarmed: 0, min: 0, max: 0 Channel 5: value: 0, failsafe: 0, disarmed: 0, min: 0, max: 0 Channel 6: value: 0, failsafe: 0, disarmed: 0, min: 0, max: 0 Channel 7: value: 0, failsafe: 0, disarmed: 0, min: 0, max: 0  Sensor 'flow': name: uavcan_flow channel 0: node id 125 --> instance 0  Sensor 'rangefinder': name: uavcan_rangefinder channel 0: node id 125 --> instance 0  Online nodes (Node ID, Health, Mode):      125 OK         OPERAT      uavcan: cycle time: 48247 events, 2907623us elapsed, 60.27us avg, min 13us max 4487us 147.270us rms uavcan: cycle interval: 48247 events, 2900.32us avg, min 34us max 281373us 1646.370us rms

taileron, 2021-10-20:
The rest of the free ram is identical to #18471 but the great thing about it is that it can be set by parameter in the respective FC as a reduction does not necessarily have to take place with an H7.

taileron, 2021-10-20:
There could still be some hidden potential for further savings in Ram.

farzadsw, 2021-12-08:
I have a similar problem with pixracer and uavcan (optical flow), 
the uavcan does not start at the boot and when I start it manually it doesn't detect the hereflow sensor.

`NuttShell (NSH) NuttX-10.2.0
nsh> uavcan status
ERROR [uavcan] application not running
nsh> free
                   total       used       free    largest  nused  nfree
        Umem:     227440     169872      57568      57424    721     10
        Prog:    2097152    1966080     131072          0     24      1
nsh> uavcan start fw
INFO  [uavcan] Node ID 1, bitrate 1000000
nsh> free
                   total       used       free    largest  nused  nfree
        Umem:     227440     185872      41568      40048    795     10
        Prog:    2097152    1966080     131072          0     24      1
nsh> uavcan status
Pool allocator status:
    Capacity hard/soft: 500/250 blocks
    Reserved:  47 blocks
    Allocated: 30 blocks

UAVCAN node status:
    Internal failures: 0
    Transfer errors:   0
    RX transfers:      0
    TX transfers:      228

CAN1 status:

ESC outputs:
INFO  [mixer_module] Param prefix: UAVCAN_EC
control latency: 0 events, 0us elapsed, 0.00us avg, min 0us max 0us 0.000us rms
INFO  [mixer_module] Mixer loaded: no
INFO  [mixer_module] Driver instance: 0
Channel Configuration:
Channel 0: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
Channel 1: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
Channel 2: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
Channel 3: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
Channel 4: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
Channel 5: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
Channel 6: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
Channel 7: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
Servo outputs:
INFO  [mixer_module] Param prefix: UAVCAN_SV
control latency: 0 events, 0us elapsed, 0.00us avg, min 0us max 0us 0.000us rms
INFO  [mixer_module] Mixer loaded: no
INFO  [mixer_module] Driver instance: 0
Channel Configuration:
Channel 0: value: 0, failsafe: 0, disarmed: 0, min: 0, max: 0
Channel 1: value: 0, failsafe: 0, disarmed: 0, min: 0, max: 0
Channel 2: value: 0, failsafe: 0, disarmed: 0, min: 0, max: 0
Channel 3: value: 0, failsafe: 0, disarmed: 0, min: 0, max: 0
Channel 4: value: 0, failsafe: 0, disarmed: 0, min: 0, max: 0
Channel 5: value: 0, failsafe: 0, disarmed: 0, min: 0, max: 0
Channel 6: value: 0, failsafe: 0, disarmed: 0, min: 0, max: 0
Channel 7: value: 0, failsafe: 0, disarmed: 0, min: 0, max: 0

Sensor 'gnss':
name: uavcan_gnss
uavcan: gnss: rtcm pub: 0 events, 0.00us avg, min 0us max 0us 0.000us rms

Sensor 'mag':
name: uavcan_mag

Online nodes (Node ID, Health, Mode):

uavcan: cycle time: 2949 events, 102759us elapsed, 34.85us avg, min 17us max 1743us 87.697us rms
uavcan: cycle interval: 2949 events, 2998.91us avg, min 1475us max 4557us 410.817us rms`

my hw:
`nsh> ver all
HW arch: PX4_FMU_V4
FW git-hash: ad70f5eef92671df35e9614d0574d30c114afb18
FW version: 1.13.0 0 (17629184)
FW git-branch: kame.v2
OS: NuttX
OS version: Release 8.2.0 (134349055)
OS git-hash: d4c06e9dfb7bb14d0f19ef47b94d838b1caf50f3
Build datetime: Dec  9 2021 00:00:41
Build uri: localhost
Build variant: default
Toolchain: GNU GCC, 6.3.1 20170620 (release) [ARM/embedded-6-branch revision 249437]
PX4GUID: 00010000000020343856434e5003004a0042
MCU: STM32F42x, rev. 5`

Also, I already diabled mavlinks (except for USB) and some unused modules to free the memory (SD_STRESS,  MC_AUTOTUNE_ATTITUDE_CONTROL, OSD, IRLOCK)

I'm relatively new to the code base, what am I missing?

taileron, 2021-12-08:
In the PX4 version used, the individual UAVCAN modules can be explicitly activated in the sub-parameters, as GPS and MAG (is the default) can be deactivated but FLOW and RANGE can be activated.

farzadsw, 2021-12-08:
Thank you for the quick reply and the tip!
I activated sub-parameters, but still, it doesn't start at boot and also when run manually, has no data:

```
nsh> uavcan start fw
nsh> uavcan status
Pool allocator status:
    Capacity hard/soft: 500/250 blocks
    Reserved:  57 blocks
    Allocated: 24 blocks

UAVCAN node status:
    Internal failures: 0
    Transfer errors:   0
    RX transfers:      0
    TX transfers:      295

CAN1 status:

ESC outputs:
INFO  [mixer_module] Param prefix: UAVCAN_EC
control latency: 0 events, 0us elapsed, 0.00us avg, min 0us max 0us 0.000us rms
INFO  [mixer_module] Mixer loaded: no
INFO  [mixer_module] Driver instance: 0
Channel Configuration:
Channel 0: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
Channel 1: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
Channel 2: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
Channel 3: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
Channel 4: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
Channel 5: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
Channel 6: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
Channel 7: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
Servo outputs:
INFO  [mixer_module] Param prefix: UAVCAN_SV
control latency: 0 events, 0us elapsed, 0.00us avg, min 0us max 0us 0.000us rms
INFO  [mixer_module] Mixer loaded: no
INFO  [mixer_module] Driver instance: 0
Channel Configuration:
Channel 0: value: 0, failsafe: 0, disarmed: 0, min: 0, max: 0
Channel 1: value: 0, failsafe: 0, disarmed: 0, min: 0, max: 0
Channel 2: value: 0, failsafe: 0, disarmed: 0, min: 0, max: 0
Channel 3: value: 0, failsafe: 0, disarmed: 0, min: 0, max: 0
Channel 4: value: 0, failsafe: 0, disarmed: 0, min: 0, max: 0
Channel 5: value: 0, failsafe: 0, disarmed: 0, min: 0, max: 0
Channel 6: value: 0, failsafe: 0, disarmed: 0, min: 0, max: 0
Channel 7: value: 0, failsafe: 0, disarmed: 0, min: 0, max: 0

Sensor 'flow':
name: uavcan_flow

Sensor 'gnss':
name: uavcan_gnss
uavcan: gnss: rtcm pub: 0 events, 0.00us avg, min 0us max 0us 0.000us rms

Sensor 'mag':
name: uavcan_mag

Sensor 'rangefinder':
name: uavcan_rangefinder

Online nodes (Node ID, Health, Mode):

uavcan: cycle time: 3768 events, 277577us elapsed, 73.67us avg, min 17us max 105503us 1720.974us rms
uavcan: cycle interval: 3768 events, 3083.66us avg, min 364us max 121697us 3100.292us rms
nsh> free
                   total       used       free    largest  nused  nfree
        Umem:     227440     199616      27824      23024    815     12
        Prog:    2097152    1966080     131072          0     24      1
```

I also already tried formatting the SD card and swapping the H/L wires.
(and also the sensor is fine, it works with cube-orange)

taileron, 2021-12-10:
Sorry, I have only now been able to test it again, but the sensor starts to stop working when the ram is less than 4k free. If only Flow and Range are activated, UAVCAN (Sensors automatic config) consumes approx. 32k, a Mavlink instance (also the USB connection) approx. 15k. This (>51000) must be free beforehand. With "uavcan start fw / all" the sensors are not recognised here either, but approx. 17k less ram is required. I need it for VTOL and for me it was necessary to reduce some UART buffers in defconfig. 

uavcan status
Pool allocator status:
    Capacity hard/soft: 500/250 blocks
    Reserved:  27 blocks
    Allocated: 17 blocks

UAVCAN node status:
    Internal failures: 0
    Transfer errors:   0
    RX transfers:      1014
    TX transfers:      1103

CAN1 status:
    HW errors: 432
    IO errors: 432
    RX frames: 3831
    TX frames: 1309
.
.
Channel 7: value: 0, failsafe: 0, disarmed: 0, min: 0, max: 0

Sensor 'flow':
name: uavcan_flow
channel 0: node id 125 --> instance 0

Sensor 'rangefinder':
name: uavcan_rangefinder
channel 0: node id 125 --> instance 0

Online nodes (Node ID, Health, Mode):
     125 OK         OPERAT    

uavcan: cycle time: 19138 events, 1406326us elapsed, 73.48us avg, min 14us max 9584us 203.516us rms
uavcan: cycle interval: 19138 events, 2465.77us avg, min 32us max 29747us 1366.323us rms
free
                   total       used       free    largest  nused  nfree
        Umem:     232736     222736      10000       5712    931     17
        Prog:    2097152    1966080     131072          0     24      1

farzadsw, 2021-12-13:
Thank you for the reply,
Since I need Mavlink over USB, I tried to free up more ram by disabling some other modules (logger, camera,...). But still didn't manage to make it work:
```
nsh> free
                   total       used       free    largest  nused  nfree
        Umem:     227840     173392      54448      54224    776     15
        Prog:    2097152    1835008     262144          0     23      2
nsh> uavcan status
ERROR [uavcan] application not running
nsh> uavcan start fw
INFO  [uavcan] Node ID 1, bitrate 1000000
ERROR [uavcan] CAN driver init failed -1006
nsh> uavcan start fw
INFO  [uavcan] Node ID 1, bitrate 1000000
nsh> uavcan status
Pool allocator status:
    Capacity hard/soft: 500/250 blocks
    Reserved:  44 blocks
    Allocated: 30 blocks

UAVCAN node status:
    Internal failures: 0
    Transfer errors:   0
    RX transfers:      0
    TX transfers:      180

CAN1 status:

ESC outputs:
INFO  [mixer_module] Param prefix: UAVCAN_EC
control latency: 0 events, 0us elapsed, 0.00us avg, min 0us max 0us 0.000us rms
INFO  [mixer_module] Mixer loaded: no
INFO  [mixer_module] Driver instance: 0
Channel Configuration:
Channel 0: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
Channel 1: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
Channel 2: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
Channel 3: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
Channel 4: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
Channel 5: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
Channel 6: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
Channel 7: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
Servo outputs:
INFO  [mixer_module] Param prefix: UAVCAN_SV
control latency: 0 events, 0us elapsed, 0.00us avg, min 0us max 0us 0.000us rms
INFO  [mixer_module] Mixer loaded: no
INFO  [mixer_module] Driver instance: 0
Channel Configuration:
Channel 0: value: 0, failsafe: 0, disarmed: 0, min: 0, max: 0
Channel 1: value: 0, failsafe: 0, disarmed: 0, min: 0, max: 0
Channel 2: value: 0, failsafe: 0, disarmed: 0, min: 0, max: 0
Channel 3: value: 0, failsafe: 0, disarmed: 0, min: 0, max: 0
Channel 4: value: 0, failsafe: 0, disarmed: 0, min: 0, max: 0
Channel 5: value: 0, failsafe: 0, disarmed: 0, min: 0, max: 0
Channel 6: value: 0, failsafe: 0, disarmed: 0, min: 0, max: 0
Channel 7: value: 0, failsafe: 0, disarmed: 0, min: 0, max: 0

Sensor 'flow':
name: uavcan_flow

Sensor 'rangefinder':
name: uavcan_rangefinder

Online nodes (Node ID, Health, Mode):

uavcan: cycle time: 2259 events, 75690us elapsed, 33.51us avg, min 16us max 1822us 75.294us rms
uavcan: cycle interval: 2259 events, 2998.80us avg, min 1350us max 4669us 392.968us rms
nsh> free
                   total       used       free    largest  nused  nfree
        Umem:     227840     189616      38224      36432    846     16
        Prog:    2097152    1835008     262144          0     23      2
```

I think it would be better for me to switch to another FC with more ram for now.

I will put my parameters here, just in case it is helpful:
```
x   UAVCAN_BITRATE [715,1575] : 1000000
    UAVCAN_EC_FAIL1 [-1,1576] : -1
    UAVCAN_EC_FAIL2 [-1,1577] : -1
    UAVCAN_EC_FAIL3 [-1,1578] : -1
    UAVCAN_EC_FAIL4 [-1,1579] : -1
    UAVCAN_EC_FAIL5 [-1,1580] : -1
    UAVCAN_EC_FAIL6 [-1,1581] : -1
    UAVCAN_EC_FAIL7 [-1,1582] : -1
    UAVCAN_EC_FAIL8 [-1,1583] : -1
    UAVCAN_EC_FUNC1 [-1,1584] : 0
    UAVCAN_EC_FUNC2 [-1,1585] : 0
    UAVCAN_EC_FUNC3 [-1,1586] : 0
    UAVCAN_EC_FUNC4 [-1,1587] : 0
    UAVCAN_EC_FUNC5 [-1,1588] : 0
    UAVCAN_EC_FUNC6 [-1,1589] : 0
    UAVCAN_EC_FUNC7 [-1,1590] : 0
    UAVCAN_EC_FUNC8 [-1,1591] : 0
    UAVCAN_EC_MAX1 [-1,1592] : 8191
    UAVCAN_EC_MAX2 [-1,1593] : 8191
    UAVCAN_EC_MAX3 [-1,1594] : 8191
    UAVCAN_EC_MAX4 [-1,1595] : 8191
    UAVCAN_EC_MAX5 [-1,1596] : 8191
    UAVCAN_EC_MAX6 [-1,1597] : 8191
    UAVCAN_EC_MAX7 [-1,1598] : 8191
    UAVCAN_EC_MAX8 [-1,1599] : 8191
    UAVCAN_EC_MIN1 [-1,1600] : 1
    UAVCAN_EC_MIN2 [-1,1601] : 1
    UAVCAN_EC_MIN3 [-1,1602] : 1
    UAVCAN_EC_MIN4 [-1,1603] : 1
    UAVCAN_EC_MIN5 [-1,1604] : 1
    UAVCAN_EC_MIN6 [-1,1605] : 1
    UAVCAN_EC_MIN7 [-1,1606] : 1
    UAVCAN_EC_MIN8 [-1,1607] : 1
x   UAVCAN_ENABLE [716,1608] : 0
x   UAVCAN_ESC_IDLT [717,1609] : 1
x   UAVCAN_LGT_ANTCL [718,1610] : 2
x   UAVCAN_LGT_LAND [719,1611] : 0
x   UAVCAN_LGT_NAV [720,1612] : 3
x   UAVCAN_LGT_STROB [721,1613] : 1
x   UAVCAN_NODE_ID [722,1614] : 1
x   UAVCAN_RNG_MAX [723,1615] : 200.0000
x   UAVCAN_RNG_MIN [724,1616] : 0.3000
x   UAVCAN_SUB_ASPD [725,1617] : 0
x   UAVCAN_SUB_BARO [726,1618] : 0
x   UAVCAN_SUB_BAT [727,1619] : 0
x   UAVCAN_SUB_DPRES [728,1620] : 0
x + UAVCAN_SUB_FLOW [729,1621] : 1
x + UAVCAN_SUB_GPS [730,1622] : 0
x   UAVCAN_SUB_ICE [731,1623] : 0
x   UAVCAN_SUB_IMU [732,1624] : 0
x + UAVCAN_SUB_MAG [733,1625] : 0
x + UAVCAN_SUB_RNG [734,1626] : 1
    UAVCAN_SV_DIS1 [-1,1627] : 500
    UAVCAN_SV_DIS2 [-1,1628] : 500
    UAVCAN_SV_DIS3 [-1,1629] : 500
    UAVCAN_SV_DIS4 [-1,1630] : 500
    UAVCAN_SV_DIS5 [-1,1631] : 500
    UAVCAN_SV_DIS6 [-1,1632] : 500
    UAVCAN_SV_DIS7 [-1,1633] : 500
    UAVCAN_SV_DIS8 [-1,1634] : 500
    UAVCAN_SV_FAIL1 [-1,1635] : -1
    UAVCAN_SV_FAIL2 [-1,1636] : -1
    UAVCAN_SV_FAIL3 [-1,1637] : -1
```

farzadsw, 2021-12-13:
Thank you for the reply,
Since I need Mavlink over USB, I tried to free up more ram by disabling some other modules (logger, camera,...). But still didn't manage to make it work:

```
NuttShell (NSH) NuttX-10.2.0
nsh> free
                   total       used       free    largest  nused  nfree
        Umem:     227840     174272      53568      52368    777     15
        Prog:    2097152    1835008     262144          0     23      2
nsh> uavcan status
ERROR [uavcan] application not running
nsh> uavcan start fw
INFO  [uavcan] Node ID 1, bitrate 1000000
nsh> uavcan status
Pool allocator status:
    Capacity hard/soft: 500/250 blocks
    Reserved:  44 blocks
    Allocated: 26 blocks

UAVCAN node status:
    Internal failures: 0
    Transfer errors:   0
    RX transfers:      0
    TX transfers:      131

CAN1 status:

ESC outputs:
INFO  [mixer_module] Param prefix: UAVCAN_EC
control latency: 0 events, 0us elapsed, 0.00us avg, min 0us max 0us 0.000us rms
INFO  [mixer_module] Mixer loaded: no
INFO  [mixer_module] Driver instance: 0
Channel Configuration:
Channel 0: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
Channel 1: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
Channel 2: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
Channel 3: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
Channel 4: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
Channel 5: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
Channel 6: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
Channel 7: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
Servo outputs:
INFO  [mixer_module] Param prefix: UAVCAN_SV
control latency: 0 events, 0us elapsed, 0.00us avg, min 0us max 0us 0.000us rms
INFO  [mixer_module] Mixer loaded: no
INFO  [mixer_module] Driver instance: 0
Channel Configuration:
Channel 0: value: 0, failsafe: 0, disarmed: 0, min: 0, max: 0
Channel 1: value: 0, failsafe: 0, disarmed: 0, min: 0, max: 0
Channel 2: value: 0, failsafe: 0, disarmed: 0, min: 0, max: 0
Channel 3: value: 0, failsafe: 0, disarmed: 0, min: 0, max: 0
Channel 4: value: 0, failsafe: 0, disarmed: 0, min: 0, max: 0
Channel 5: value: 0, failsafe: 0, disarmed: 0, min: 0, max: 0
Channel 6: value: 0, failsafe: 0, disarmed: 0, min: 0, max: 0
Channel 7: value: 0, failsafe: 0, disarmed: 0, min: 0, max: 0

Sensor 'flow':
name: uavcan_flow

Sensor 'rangefinder':
name: uavcan_rangefinder

Online nodes (Node ID, Health, Mode):

uavcan: cycle time: 1591 events, 53125us elapsed, 33.39us avg, min 16us max 1777us 81.354us rms
uavcan: cycle interval: 1591 events, 2997.98us avg, min 1218us max 4791us 450.188us rms
nsh> free
                   total       used       free    largest  nused  nfree
        Umem:     227840     191984      35856      35664    883     12
        Prog:    2097152    1835008     262144          0     23      2
```

I think it would be better for me to switch to another FC with more ram for now.

I will put my parameters here, just in case it is helpful:

```
x + UAVCAN_ENABLE [716,1608] : 2
x   UAVCAN_ESC_IDLT [717,1609] : 1
x   UAVCAN_LGT_ANTCL [718,1610] : 2
x   UAVCAN_LGT_LAND [719,1611] : 0
x   UAVCAN_LGT_NAV [720,1612] : 3
x   UAVCAN_LGT_STROB [721,1613] : 1
x   UAVCAN_NODE_ID [722,1614] : 1
x   UAVCAN_RNG_MAX [723,1615] : 200.0000
x   UAVCAN_RNG_MIN [724,1616] : 0.3000
x   UAVCAN_SUB_ASPD [725,1617] : 0
x   UAVCAN_SUB_BARO [726,1618] : 0
x   UAVCAN_SUB_BAT [727,1619] : 0
x   UAVCAN_SUB_DPRES [728,1620] : 0
x + UAVCAN_SUB_FLOW [729,1621] : 1
x + UAVCAN_SUB_GPS [730,1622] : 0
x   UAVCAN_SUB_ICE [731,1623] : 0
x   UAVCAN_SUB_IMU [732,1624] : 0
x + UAVCAN_SUB_MAG [733,1625] : 0
x + UAVCAN_SUB_RNG [734,1626] : 1
    UAVCAN_SV_DIS1 [-1,1627] : 500
    UAVCAN_SV_DIS2 [-1,1628] : 500
```

taileron, 2022-01-11:
i think for this ram limited fc you should no longer expect to be able to do vtol and uavcan sensors at the same time. The PX4 code is growing and therefore consuming more and more resources, so I would close this now.

closed 2022-01-11 by taileron