PX4/PX4-Autopilot #16336: error about smooth velocity s curve time

garlinplus, 2020-12-05:
/**
 * @class VelocitySmoothing
 *
 * TODO: document the algorithm
 *    |T1| T2 |T3|
 *     ___
 *   __| |____   __ Jerk
 *            |_|
 *          ___
 *         /   \	 Acceleration
 *   ___/      \___
 *               ___
 *            ;"
 *          /
 *         / 	 Velocity
 *        ;
 *   ----"
 */

As above figure,the velocity smooth process is divide into three segment.We suppose init state is (j0,a0,v0).
T1 end:
a(T1) = a0 +j*T1 
v(T1) = v0 + a0*T1 + 0.5*j*T1^2
T2 end:
a(T2) = a(T1) = a0 +j*T1
v(T2) = v(T1) + a(T1)*T2 
         =  v0 + a0*T1 + 0.5*j*T1^2 + (a0 +j*T1) * T2
         = v0 + a0*(T1+T2) + j*T1*T2 + 0.5*j*T1^2
T3 end
a(T3) = a0
v(T3) = v(T2) + a(T2)*T3 - 0.5*j*T3^2
         = v0 + a0*(T1+T2) + j*T1*T2 + 0.5*j*T1^2 + (a0 +j*T1)*T3 -  0.5*j*T3^2
         = v0 + a0*(T1+T2+T3) + j*T1*(T2+T3) + 0.5*j*(T1^2-T3^2)

  The function can get T1 by two functions,as followed:
float VelocitySmoothing::computeT1(float a0, float v3, float j_max, float a_max)
{
	float delta = 2.f * a0 * a0 + 4.f * j_max * v3;

	if (delta < 0.f) {
		// Solution is not real
		return 0.f;
	}

	float sqrt_delta = sqrtf(delta);
	float T1_plus = (-a0 + 0.5f * sqrt_delta) / j_max;
	float T1_minus = (-a0 - 0.5f * sqrt_delta) / j_max;

	float T3_plus = a0 / j_max + T1_plus;
	float T3_minus = a0 / j_max + T1_minus;

	float T1 = 0.f;

	if (T1_plus >= 0.f && T3_plus >= 0.f) {
		T1 = T1_plus;

	} else if (T1_minus >= 0.f && T3_minus >= 0.f) {
		T1 = T1_minus;
	}

	T1 = saturateT1ForAccel(a0, j_max, T1, a_max);

	return math::max(T1, 0.f);
}

The above function caculate T1 from a0 and delta velocity.
From above equation,If we know a0,v0 and j0(jmax) we can caculate T1 from equation
 v(T1) = v0 + a0*T1 + 0.5*j*T1^2
and dv = v(T1) - v0,so 
 0.5*j*T1^2 + a0*T1 - dv = 0
We can get
T1 = (-a0 +- sqrtf(a0^2 + 2*j*dv) / j
And we can get T3 from
v(T3) = v(T2) + a(T2)*T3 - 0.5*j*T3^2
and dv = v(T3)-v(T2) for this time,so
- 0.5*j*T3^2 + (a0+j*T1)*T3 - dv = 0
We can get
T3 = [-(a0+j*T1) +- sqrtf((a0+j*T1)^2 - 2j*dv)]/(-j)
     = [(a0+j*T1) +- sqrtf((a0+j*T1)^2 - 2j*dv)]/j
because
(a0+j*T1)^2 - 2j*dv = a0^2 + 2*a0*j*T1 + j^2*T1^2 - 2j*dv
= a0^2 + 2*a0*j*T1 + j*(j*T1^2) - 2j*dv
from equation
0.5*j*T1^2 + a0*T1 - dv = 0
we can get
j*T1^2 = 2(dv - a0*T1)
so
(a0+j*T1)^2 - 2j*dv= a0^2 + 2*a0*j*T1 + j*(j*T1^2) - 2j*dv
                    = a0^2 + 2*a0*j*T1 +2j*(dv - a0*T1) - 2j*dv
                    = a0^2
so
T3 = [(a0 + j*T1) +- a0] / j
     = T1 or (T1 + 2a0/j)
If T3 = T1 + 2a0/j
a(T3) = a(T2) - j * T3
         = a0 + j*T1 - j*(T1 + 2*a0/j)
         = -a0
This value is not agree with the first figure,so T3 = T1.But the above function VelocitySmoothing::computeT1 's result is
	float T3_plus = a0 / j_max + T1_plus;
	float T3_minus = a0 / j_max + T1_minus;
It's T3 = T1 + a0 / j,and is it right?
And the second function caculate T1 from Total plan time and delta v as followed:
float VelocitySmoothing::computeT1(float T123, float a0, float v3, float j_max, float a_max)
{
	float a = -j_max;
	float b = j_max * T123 - a0;
	float delta = T123 * T123 * j_max * j_max + 2.f * T123 * a0 * j_max - a0 * a0 - 4.f * j_max * v3;

	if (delta < 0.f) {
		// Solution is not real
		return 0.f;
	}

	float sqrt_delta = sqrtf(delta);
	float denominator_inv = 1.f / (2.f * a);
	float T1_plus = math::max((-b + sqrt_delta) * denominator_inv, 0.f);
	float T1_minus = math::max((-b - sqrt_delta) * denominator_inv, 0.f);

	float T3_plus = a0 / j_max + T1_plus;
	float T3_minus = a0 / j_max + T1_minus;

	float T13_plus = T1_plus + T3_plus;
	float T13_minus = T1_minus + T3_minus;

	float T1 = 0.f;

	if (T13_plus > T123) {
		T1 = T1_minus;

	} else if (T13_minus > T123) {
		T1 = T1_plus;
	}

	T1 = saturateT1ForAccel(a0, j_max, T1, a_max);

	return T1;
}

As I derived this function,I found the function is based on T3 = T1 + a0/j,so is it wrong?

bresch, 2020-12-07:
Hi @garlinplus , I'm not sure if I'm following you. I made this python script to solve the symbolic equations of motion: https://github.com/Auterion/Flight_Control_Prototyping_Scripts/blob/master/trajectory_generator/velocity_trajectory_generator_symbolic.py . 
Could you please have a look and point me to where you see the error?

garlinplus, 2020-12-09:
> Hi @garlinplus , I'm not sure if I'm following you. I made this python script to solve the symbolic equations of motion: [...]
Hi @bresch ,I'm read your python code about  velocity trajectory generator.You are right code is very good!
 I'm make a mistake that I think a(T3) equal with a0,and my derivation is worng if a0 is not zero.So I miss the result a0 +jT1 -jT3 = 0.Thank you very much！

bresch, 2020-12-09:
@garlinplus Ok, all good then. Please let me know if you find anything else strange.

closed 2020-12-09 by bresch