PX4/PX4-Autopilot #16351: Metadata: add support for tagged versions

bkueng, 2020-12-08:
From https://github.com/PX4/PX4-Autopilot/pull/16039, for boards that will use external metadata:
- Push metadata for a released version to S3 to a directory with the tag name
- Use that URL in `component_info.metadata_uri`

dagar, 2020-12-08:
At the moment it's going to be something like

 - master/main
    - vendor (eg px4)
      - board (eg fmu-v5)
         - compressed metadata (airframe, parameters, etc)
 - vX.Y.Z (includes anything tagged alpha, beta, RC, etc)
    - vendor
      - board
         - compressed metadata (airframe, parameters, etc)

One edge case this is glossing over is if we want to handle board variants (specialized builds). 

I'm also still potentially interested in making these metadata packages available by hash for the sake of development builds.

bkueng, 2020-12-09:
>  One edge case this is glossing over is if we want to handle board variants (specialized builds).

I think we have to, otherwise we don't have all metadata for e.g. all v2 build variants.

Also important: the max url length is 70 chars for the mavlink message. I was thinking we can use the board ID instead of the full name.

bkueng, 2021-03-29:
Done in https://github.com/PX4/PX4-Autopilot/pull/17191

closed 2021-03-29 by bkueng