PX4/PX4-Autopilot #16353: Failed and stale ms5611 temperature sensing not detected, result in drifting and wrong altitude

mlaiacker, 2020-12-08:
**Describe problem solved by the proposed feature**
We have a Cube black running modified version 1.8.2. We wanted to do a flight today but noticed unstable relative altitude estimation while still on ground. We found the issue in a heavy drifting pressure_alt. It drifted from -350m to +600m in about 20minutes while the system was heating up but the sensor reported temperature was constant at 43.18 degC. 
This failure was not detected by any checks.

**Describe your preferred solution**
I looked at release 1.11 and found a check for a stale baro sensor but only for the pressure measurement. Maybe it could be an option to add this check also for the temperature measurement. I hope this will not lead to more false positives. I don't know how common this failure mode is but it happened to us so maybe it will happen to others.

**Describe possible alternatives**
Compare to other baro sensors in system and warn for inconsistency like its done with acc and gyro.

**Additional context**
stale temperature sensor on ms5611 baro0
[image]

working temperature sensor on ms5611 baro1
[image]

Resulting pressure mesurements. baro 1 is correct and baro0 is way off!

[image]

Thanks for your support.

dagar, 2020-12-08:
Did this particular ms5611 start working again later?

mlaiacker, 2020-12-08:
Don't know. It is still broken, after several power cycles, no change. But I know it was working at one point. 
But during the last flight in Nov. 30. only one ms5611 got detected. It was the working one so the flight was OK and we only noticed it today after having issues with the baros.
It's a bit strange.

dagar, 2020-12-08:
As much as possible I'd like the drivers themselves to catch these cases and then attempt to reset the actual device.

> Don't know. It is still broken, after several power cycles, no change. 

Does that mean it's still "successfully" starting, but always publishing constant data?

mlaiacker, 2020-12-08:
Yes. Only the published temperature is constant the pressure is noisy as expected but with a huge offset.
I think the sensor itself is broken. It's a MEMS so maybe something mechanical broke inside the sensor.

dagar, 2020-12-08:
The temperature reading we get back from the ms5611 is a 24 bit integer. I would propose we add a simple check in the driver that if the raw (integer) temperature or pressure readings are stuck then the driver stops publishing.

https://github.com/PX4/PX4-Autopilot/blob/8f5f564c05b5a0e12989d6e56642cc0b453ec45d/src/drivers/barometer/ms5611/ms5611.cpp#L261

I had a refactor of this driver on the go that added a simple state machine and reset section. I'll try to revisit this soon and then the driver's internal stale check could trigger the safe reset.

mlaiacker, 2020-12-08:
I hope this sensor failure is not a one in a million event and the check we add will give a false positive one in a thousand and shut off a perfectly good working sensor. I will be happy to test your fix on this broken sensor.

The VTOL the Cube Back is in has around 80 flight hours.

mlaiacker, 2020-12-09:
Today I removed the Cube Black from the airframe and put px4 V1.11 on it. 
Now the broken baro is baro1 and the working one baro0.

Log: https://logs.px4.io/plot_app?log=d62e3f25-7d32-4c97-ae6c-5b0d0138de8b

baro1 temperature constant 43.18 degC
[image]