PX4/PX4-Autopilot #16358: Spektrum DX8G2 doesn't work with PX4

ilNonn0, 2020-12-09:
My Spektrum radio doesn't work on PX4 software.
I am using the DX8 G2 transmitter and the SPM9645 remote satellite receiver. 
According to the documentation, all the Spektrum models are supported but reading the discussion group and the repository issues I found that several people had similar problems (for instance see [here](https://github.com/PX4/PX4-Autopilot/issues/15201)). 

To test my system I temporarily switch to Ardupilot, where everything works fine. Hence I guess the problem lies in the PX4 software, where every time I try to calibrate my radio I got the following message from QGC: "detected 0 radio channels. to operate px4 you need at least 5 channels". (Obviously, the radio has already been manually bound).

I saw that recently there has been some software modification and [decoding improvement](https://github.com/PX4/PX4-Autopilot/pull/15430), but from my side, the problem is still there.  

I checked my radio setup and the PX4 source code, I have noticed (but I am not completely sure about) that the PX4 code (RCinput.cpp and dsm.cpp) manages the DSMX protocol with a frame rate of 11ms. 
My setup seems to use the DSMX protocol with a 22ms frame rate. This is a config that I am not able to change. 
Is it possible that this is the source of error?
Can someone help me and all other people who bought the Spektrum radio system?
Someone from the Maintenance Team to talk with?

Thanks in advance

LorenzMeier, 2020-12-10:
I am looking into this.

ilNonn0, 2020-12-10:
Thanks @LorenzMeier 
I checked again... I don't know if the problem is on the frame rate. Looking the source code seems it is able to manage the DSMX/22ms frame rate configuration. 
Is there anything I can do to report the issue more precisely (log, dmesg, ..)?

LorenzMeier, 2020-12-10:
Could you take a picture of the setup of the autopilot setup? I just want to make sure I understand what is connected to what and which HW you're using.

ilNonn0, 2020-12-10:
Here the pics...
[image]

[image]

The pixhawk is powered by USB cable and there is attached only the satellite receiver.
In the following the dmesg print:
[image]

Is there something wrong with the hardware?
I excluded the possibility since with Ardupilot the setup works...

LorenzMeier, 2020-12-10:
What is the output of `px4io status` on the shell?

ilNonn0, 2020-12-10:
here you are:
[image]

[image]

sorry, I need to instal all the required packet to launch mavlink_shell.py...

LorenzMeier, 2020-12-10:
I have an SPM9645, I will cross-test.

ilNonn0, 2020-12-10:
thank you.
Let m know if I can help.

I have tested several software version, but all the time the radio doesn't work.
I can tell you that Ardupilot runs on ChibiOS. Does it matter?

mcsauder, 2020-12-10:
Hi @ilNonn0 and @LorenzMeier ,  in case it saves you any time, I investigated this briefly in regard to #15201 but didn't have time to actually fix anything.  I spoke briefly with @dagar , but the most likely reason the QGC issued bind command isn't working is because it is calling [here](https://github.com/PX4/PX4-Autopilot/blob/d44e537084cdaf1a2ce001d6865f41b1013c4a9c/src/drivers/px4io/px4io.cpp#L1661) but it likely needs to call [here](https://github.com/PX4/PX4-Autopilot/blob/d44e537084cdaf1a2ce001d6865f41b1013c4a9c/src/drivers/rc_input/RCInput.cpp#L605). I might be wrong because I obviously haven't fixed it, but just in case it saves anyone some time debugging, FWIW.

ilNonn0, 2020-12-11:
Thanks, @mcsauder.
Tomorrow morning I am going to invstigate better the poblem. 
I would like to debug step by step through the code but first I have to rearrange cables and some old electronics in the Lab in order to realize a SWD interface.
By the way, I skipped the bind issue. My problem is that even with the radio already buond (manullay, i.e. without using QGC), I got not channels recognition pressing "Calibrate" button in QGC.

mcsauder, 2020-12-11:
Hi @ilNonn0 , what I found was that the receiver that I used to bind the satellite impacts the serial data stream format and cases the PX4 driver to not communicate properly with the satellite. For example, an AR6210 works fine to bind satellites but as a 6 channel will give you 22ms DSMX and only 6 channel capability.  However, with an AR9020, labeled "with X-plus technology", the satellite will bind just fine as 11ms DSMX but will fail to communicate with the PX4 driver.  If you find another (preferably older) receiver that is SRXL, not SRXL2, or if you could use the native PX4 bind function I suspect your issue will be resolved.

LorenzMeier, 2020-12-11:
Intermediate results: I can reproduce this on my local setup (which used to work and is unchanged). I've already tried a couple of things like downgrading the DSM parser, which didn't yield the desired outcome. I will work this over the weekend. Stay tuned and thanks for the patience!

LorenzMeier, 2020-12-11:
@davids5 I'm also considering the UART layer since the main thing that changed was the NuttX upgrade.

dagar, 2020-12-24:
Could you give this branch a try? https://github.com/PX4/PX4-Autopilot/compare/pr-dsm_bind_always_dsmx_11?expand=1

It defaults to binding as DSMX 11 ms (recommend by Spektrum). https://www.spektrumrc.com/ProdInfo/Files/Remote%20Receiver%20Interfacing%20Rev%20A.pdf

> Selecting the 11ms mode automatically allows either 11ms or 22ms protocols to be used.

ilNonn0, 2020-12-30:
Hi @dagar, sorry for the delay. It's Christmas time and I am at home with a personal laptop not able to build the firmware... 
I struggled a bit to get the .bin file.
I just answer into the link you provided, however, the firmware doesn't work.
I saw your changes, but I think the main problem isn't the binding process since this can also be done "outside" QGC (in my case I used the procedure shown in the Spektrum manual). What do you think?

LorenzMeier, 2020-12-30:
@ilNonn0 Sorry for the delay - I have been out over Christmas, but I'm spending more time on it today. I had first to build the test setup, which I have now.

LorenzMeier, 2020-12-31:
We are making progress, but not finished yet: https://github.com/PX4/PX4-Autopilot/pull/16469

I've also ordered an NX10 so we get a head-start on SRXL2 and potential issues there.

LorenzMeier, 2021-01-01:
Fixed here: https://github.com/PX4/PX4-Autopilot/pull/16470 - works for me now - could you re-test and report back?

mcsauder, 2021-01-01:
@ilNonn0 and @LorenzMeier , I bench tested current master with a Spektrum DX8 G2 and spektrum satellite serial receivers spm4647 and SPM9645. Everything works as expected.  Nice work @LorenzMeier

ilNonn0, 2021-01-01:
Hi, @LorenzMeier great job. 
Now everything works here: Spektrum DX8G2, remote satellite SPM9645 and DSMX 22ms as binding protocol. 
First I have bounded the transmitter with the receiver, then I calibrated the radio Channel successfully.
 
Initially, I had troubles since I triggered the binding process by the Spektrum menu and it doesn't work, later I used the bind button at the radio power on and it immediately works.
In my case, I left the SYS_USE_IO parameter to the default value (1). Putting it to 0 I was not able to bind the radio. 

What was the problem? Didn't the parsing address the DSMX 22ms case?

closed 2021-01-01 by ilNonn0

LorenzMeier, 2021-01-02:
The root cause is a horrible protocol design by Spektrum where some of their receivers communicate without a header and or checksum. We tried to harden the parser, but that required you to use the PX4 binding process (which always worked on FMUv5 and FMUv2/3 if you had your system fully disarmed, but most people have the safety off by default - something that I've now allowed).

Leoiii, 2021-06-17:
I am still getting this error when flashing the latest stable release (1.11.3). The receiver does not enter into a bind mode when prompted on QGroundControl. I have the same setup as above (DX8 receiver and DSMX 9645 satellite receiver). When I flash the developer release of the code then the system appears to work properly and the receiver enters bind mode correctly. However, I'd ideally like to run the latest stable version of firmware on my Pixhawk. Am I missing something that this fix is not part of the latest stable version of the code from QGroundControl?

mcsauder, 2021-06-24:
Hi @Leoiii , have you tried the latest beta (v1.12.0 beta 5)?  I am pretty sure this is fixed now.  Using QGC, just try flashing latest master branch and see if the issue goes away.  If it is gone with that, v1.12.0 release should be good for you.  Hopefully this helps!