PX4/PX4-Autopilot #16359: No thrust scaling factor found, DO NOT FLY

weiyuhe, 2020-12-09:
I was following the instruction on [https://dev.px4.io/master/en/simulation/ros_interface.html](url). When I ran:
roslaunch mavros px4.launch fcu_url:="udp://:14540@127.0.0.1:14557"

I got the following errors:
[FATAL] [1607550676.819332084]: No thrust scaling factor found, DO NOT FLY
[FATAL] [1607550676.819678433]: No system mass found, DO NOT FLY
[FATAL] [1607550676.820004055]: No yaw rate scaling factor found, DO NOT FLY

When I proceed to run:
cd <PX4-Autopilot_clone>
DONT_RUN=1 make px4_sitl_default gazebo
source ~/catkin_ws/devel/setup.bash    # (optional)
source Tools/setup_gazebo.bash $(pwd) $(pwd)/build/px4_sitl_default
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$(pwd)
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$(pwd)/Tools/sitl_gazebo
roslaunch px4 posix_sitl.launch

Gazebo was stuck in "perparing your world". And there are three errors:
ERROR [parameters] ignoring unrecognised parameter 'BAT_CNT_V_CURR'
ERROR [parameters] ignoring unrecognised parameter 'BAT_CNT_V_VOLT'
ERROR [parameters] ignoring unrecognised parameter 'CAL_ACC_PRIME'
ERROR [parameters] ignoring unrecognised parameter 'CAL_GYRO_PRIME'
ERROR [parameters] ignoring unrecognised parameter 'CAL_MAG_PRIME'

In the log file from posix_sitl.launch, I found:
most recent call last):
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 563, in connect
    self.local_endpoint = self.socket.getsockname()
AttributeError: 'NoneType' object has no attribute 'getsockname'

Anyone has idea on why this is happening?

dagar, 2021-01-21:
I think you must be using older PX4 because parameters like CAL_ACC_PRIME/CAL_GYRO_PRIME no longer exist.

Can you give this another try with current PX4?

sunrising101, 2022-07-23:
i have same problem