PX4/PX4-Autopilot #16363: Wrong data in GPS_STATUS

jbeyerstedt, 2020-12-10:
**Describe the bug**
The [GPS_STATUS](https://mavlink.io/en/messages/common.html#GPS_STATUS) message is defined to contain the global satellite ID in the `satellite_prn` array. But the data we receive does not make sense at all (duplicate satellite IDs and IDs, which are not shown in uBlox U-Center).

According to the uBlox protocol description, no PRN values are provided by the NAV-SAT and NAV-SVINFO messages. And there might be a big bug in the parser for these messages (see https://github.com/PX4/PX4-GPSDrivers/issues/63).

This issue should just keep track of the necessary changes to the rest of PX4, when the GPSDriver code is fixed. Because there two possible ways to ensure compatibility between the uBlox driver and the GPS_STATUS message when the PRN field is not filled by the uBlox driver:
- either the MAVLink message should use `satellite_info.svid` when `satellite_info.prn` is empty
- or the ublox driver must copy the `satellite_info.svid` to `satellite_info.prn` (to ensure backwards compatibility).

jbeyerstedt, 2020-12-11:
There is a PR for the driver issue now: https://github.com/PX4/PX4-GPSDrivers/pull/64. For now, I decided to use the second approach (maintaining backwards compatibility ), which does not need any changes in PX4 itself. Just updating the GPS drivers submodule of course.

jbeyerstedt, 2020-12-16:
The PR in the driver repository just got merged, please update the submodule revision to close this issue.

mrpollo, 2020-12-16:
@jbeyerstedt thanks for the PR, I went ahead and sent a PR to update the submodule see #16407 16407

closed 2021-01-06 by jbeyerstedt