PX4/PX4-Autopilot #16368: Dead code in commander

garfieldG, 2020-12-10:
Hi,

after looking at the code I've noticed that the condition_system_sensors_initialized flag is always true.
The only place it can be changed to false is inside this above if, but it can never get there because it will never pass the first condition.
https://github.com/PX4/PX4-Autopilot/blob/72420dfff3fe96dc51386508ecc8fc5a4f453283/src/modules/commander/state_machine_helper.cpp#L147
I'm not entirely sure if it's just leftovers from old versions or maybe the condition_system_sensors_initialized flag should be initialized to false or changed in some other place, but for now, it just seems that the if statement and also condition_system_sensors_initialized flag are just unnecessary.

bresch, 2020-12-10:
It is initialized to false here https://github.com/PX4/PX4-Autopilot/blob/72420dfff3fe96dc51386508ecc8fc5a4f453283/src/modules/commander/Commander.hpp#L392 but then directly to true in the constructor "for now...." https://github.com/PX4/PX4-Autopilot/blob/72420dfff3fe96dc51386508ecc8fc5a4f453283/src/modules/commander/Commander.cpp#L452-L453

mrpollo, 2020-12-15:
Looks like history is hard to find through git blame due to some refactors, @dagar do you remember what the TODO is for? (I'm assuming XXX means TODO)

bkueng, 2022-10-25:
This flag got removed.

closed 2022-10-25 by bkueng