PX4/PX4-Autopilot #16369: Unexpected Mixer Behavior With Yaw and multicopter controller

davidcgregory, 2020-12-10:
**Describe the bug**
Using mc startup script in airframe file:

`sh /etc/init.d/rc.mc_defaults`

And creating a mixer:
`
M:3
O:
S: 0 0   5000  5000        0 -10000 10000
S: 0 1 -5000 -5000        0 -10000 10000
S: 0 2   2500  2500 -2500 -10000 10000
'

servo does not respond to Yaw commands

**To Reproduce**
load airframe with mixer defined as above
Power on drone
change to stabilized mode
send commands via RC stick movement
servo responds to pitch and roll stick commands
no response from yaw stick command

**Expected behavior**
servo responds to stick commands.

**Log Files and Screenshots**
This is a bench test showing the response to roll and pitch stick inputs, and the lack of response to yaw stick inputs
https://review.px4.io/plot_app?log=88c3d100-0806-41f5-a257-02737aaa2ec3

Add screenshots to help explain your problem.

**Drone (please complete the following information):**
-custom drone config and design from scratch, single rotor 4 servos
-hardware is PixHawk4 mini

bresch, 2020-12-11:
`S: 0 2 2500 2500 -2500 -10000 10000` selects the yaw controller output, not the RC channel directly. In your case, the yaw controller did wind-up completely; this is why you don't see the servo responding to the RC yaw stick.
[image]

davidcgregory, 2020-12-11:
I see that the yawrate controller wound up. 

But i dont understand why the ROLL and PITCH setpoints respond to throttle stick inputs but the YAW setpoints did not?

LorenzMeier, 2021-01-01:
Because roll / pitch are prioritized over yaw. You will need to fix your system to re-gain yaw authority.

davidcgregory, 2021-01-01:
The above log is from a test  on the bench with the drone stationary.   I'm having trouble stabilizing the drone around the yaw axis 
and I was trying to verify mixer settings on the bench.    Perhaps the yaw problems in flight are related to the roll/pitch prioritization.

Does the roll/pitch prioritization occur here? https://github.com/PX4/PX4-Autopilot/blob/fd18ba649678a8aa5f1b871b2f6b3d770c1d87d3/src/modules/mc_att_control/mc_att_control_main.cpp

LorenzMeier, 2021-01-01:
There and in the subsequently called mixer. For a heli or multi rotor you need to make sure to use a specific mixer, not the standard mixers

davidcgregory, 2021-01-03:
I'm flying a custom mixer using all simple mixers.  The vehicle flys but I'm unable to stabilize it in yaw.   Does the MC_ATT_CONTROL app limit yaw authority directly, similar to how the thrust saturation logic works?  Or are you simply saying that most mixers will allocate less authority to yaw?

Here is a flight showing the yaw performance (or lack there of)

https://review.px4.io/plot_app?log=10c0af22-eb62-42f4-8bff-cb355e1a9546

bresch, 2021-01-04:
@davidcgregory Why not using the multirotor mixer? It performs much better than simple mixers as it handles motor saturation and axis prioritization properly (the yaw is prioritized over roll/pitch as it is allocated at the end, see for example [here](https://github.com/PX4/PX4-Autopilot/blob/98cff947025d2f5c3e4c69c2f4689a6dae02a2ca/src/lib/mixer/MultirotorMixer/MultirotorMixer.cpp#L290-L291)).

It your log, it seems that there is some yaw authority but none of the controllers are tuned so the drone struggles to keep the right heading and attitude.

davidcgregory, 2021-01-04:
@bresch @LorenzMeier 

Sorry to not be clear about the vehicle configuration.
My config is similar to this: https://hackaday.io/project/174329-spacex-inspired-edf-rocket
Cutaway view of the 4 servo actuators:  https://cdn.hackaday.io/images/796381607539303831.JPG

If you think its a tuning issue, I will keep going.  Its been quite a struggle to get this far, although some if was self inflected as I didn't fully understand all the edge-behavior of the PX4 stack (like the thrust saturation logic).

But to bring back to the original point of this issue:
It would be helpful for me in setting up a new mixer if the servos responded to stick inputs in stabilized mode when disarmed.   To confirm the mixing was correct I actually set it up as a FW first.   Even at this point it would give me some comfort to be able to confirm the yaw behavior is correct for all four servos on the ground.

Alternately, I would be nice to directly simulate or override Pitch/Roll/Yaw Rate commands via console, similar to using "pwm test -c ..."

bresch, 2021-01-04:
@davidcgregory Oh, I see now. One thing to remember is that the collective thrust from the EDF is coupled with the torque produced by the TVC system. Your mixer assumes that the torque is function of the servo deflection and not the thrust, which only works fine a trimmed conditions (hover).
Regarding yaw authority and tuning, did you try to hang the drone on a ceiling, set a thrust close to the hover thrust in acro mode and test the yaw response?