PX4/PX4-Autopilot #16372: ekf2 airspeed check bit on a multicopter

dagar, 2020-12-11:
[image]

https://logs.px4.io/plot_app?log=0ad8c6f0-a7d5-47a1-be91-0da064561eea

bresch, 2020-12-11:
There's a mismatch between the ecl and firmware bitmasks.

The innovation that fails is the yaw observation
[image]

dagar, 2020-12-11:
> should we log 13 booleans instead of a bitmask?

Yes please! It'll burn a bit of flash storing the string for each field (need for listener and logger), but I think it's worth it.

dagar, 2020-12-12:
Started in https://github.com/PX4/PX4-Autopilot/pull/16383.

mrpollo, 2020-12-15:
@dagar where should we track this issue? (which project board?)

dagar, 2020-12-29:
We can consider this fixed by https://github.com/PX4/PX4-Autopilot/pull/16383.

closed 2020-12-29 by dagar