PX4/PX4-Autopilot #16378: Add a gimbal to iris

AliaChe, 2020-12-11:
Hello, 

I'm trying to add a gimbal to the iris model and I'm facing these two main problems :

mrpollo, 2020-12-15:
@Jaeyoung-Lim do you have any hints for @AliaChe ?

Jaeyoung-Lim, 2020-12-16:
@AliaChe 

> With this configuration, the drone control is deteriorated, it has a large response time. So, we can see the drone falling down and then retrying to gain its altitude in nearly 0.5 to 1 second.

I think this is expected and representitive on what will happen in the real world if you add additional payload to the drone - you need to retune the vehicle. 

> The second problem is related to the gimbal controller. Once I remove the sonar model from my configuration, the gimbal starts to move in all directions when the drone is spawn into gazebo and when the drone starts moving, the control performance is not satisfactory.

Are you sure you have all the gimbal control channels configured in the mixer and also the sdf configured correctly?

AliaChe, 2020-12-17:
@Jaeyoung-Lim  

Thanks for your reply.

I agree, I will try retuning the vehicle. However, I have succeeded to add a gimbal to iris without retuning the drone controllers, using ros-eloquent and an older version of PX4-Autopilot (9fcbf18d).

Concerning the gimbal controller, the control channels are configured correctly in the sdf. I'll check the mixer to verify.

AliaChe, 2020-12-17:
@Jaeyoung-Lim 

In fact, the problem of the vehicle control is solved once I reset the motors constant value to 8.54e-06 as in sitl_gazebo 0da08f0 instead of 5.84e-06 as in sitl_gazebo cd8ba25.

LorenzMeier, 2021-01-01:
The gimbal is too heavy for a "realistic" IRIS model. You would need to completely re-design also the trust curves.

closed 2021-01-01 by LorenzMeier