PX4/PX4-Autopilot #16384: MPU9250 and BMP280 I2C support

taileron, 2020-12-12:
in the meantime, does anyone still have an idea how to get the ak8963 magnetometer integrated in the mpu 9250 to work externally I2C together with the BMP280? Only a very small Baro/Mag combination sensor fits on the 75 mm whoop (Mamba F405 Mini MK2/Omnibus F4 Target), this is the only one that is really small enough mechanically.
BMP280 I2C, GPS (Cam M8Q) or ThoneFlow UART 6, DSHOT and DSMX RC are already running successfully, only magnetometer is still missing.
There are 2 drivers, the one outside the Ivensense folder allows I2C but lacks the -M parameter to activate the integrated compass.
Neither the MPU 9250 nor the AK8963 can find the sensor on I2C.
(in PixRacer, the magnetometer, which is disabled by default in the MPU9250, can easily be reactivated with -M).
#5677
#10109

dagar, 2020-12-13:
They standalone ak8963 will work, but only after the mpu9250 is put into i2c passthru mode.

https://github.com/PX4/PX4-Autopilot/pull/15421

mrpollo, 2020-12-15:
@taileron does this help?

taileron, 2020-12-16:
@mrpollo Sorry, so far I have not been able to test it, for Mamba F405 mk2 I have made so many local changes that the original OM F4 code is problematic. I think I need to look into the github features a bit more.
Does it make sense to bring Mamba F405 MK2 as a new target to github, but then how does the changed MPU9250 driver get in there, or is it possible to change it with just a few lines of code?

taileron, 2020-12-18:
@mrpollo the driver seems to start but then mag goes into TOUT:
(the module pin AD0 has to be low otherwise BMP280 doesn´t start anymore so mpu9250 must be 0x68)
if mpu9250_i2c -X -b 1 -a 0x68 -f 400 -R 6 start; then
 	sleep 1 
 	ak8963 -X -R 4 start
 	bmp280 -X start
fi

dmesg:
mpu6000 #0 on SPI bus 1 (devid=0x21)
ERROR [SPI_I2C] Bug: driver mpu9250_i2c does not pass the I2C address to I2CSPIDriverBase
mpu9250_i2c #0 on I2C bus 2 (external, equal to '-b 1')
bmp280 #0 on I2C bus 2 (external, equal to '-b 1')
ERROR [sensors] Mag #0 fail:  TIMEOUT!
WARN  [PreFlightCheck] Preflight Fail: Accel #1 uncalibrated
WARN  [PreFlightCheck] Preflight Fail: Gyro #1 uncalibrated
WARN  [PreFlightCheck] Preflight Fail: Accels inconsistent - Check Cal

(rotation not corrected)
sensors status
INFO  [sensors] selected gyro: 2162698 (0)
INFO  [data_validator] validator: best: 0, prev best: 0, failsafe: NO (0 events)
INFO  [data_validator] sensor #0, prio: 100, state: OK
INFO  [data_validator] 	val:   0.0135, lp:   0.0054 mean dev:   0.0000 RMS:   0.0033 conf:   1.0000
INFO  [data_validator] 	val:   0.0059, lp:  -0.0001 mean dev:  -0.0000 RMS:   0.0030 conf:   1.0000
INFO  [data_validator] 	val:  -0.0048, lp:  -0.0045 mean dev:   0.0000 RMS:   0.0010 conf:   1.0000
INFO  [data_validator] sensor #1, prio: 100, state: TOUT
INFO  [data_validator] 	val:  -0.0147, lp:  -0.0804 mean dev:  -0.0003 RMS:   0.5125 conf:   0.0000
INFO  [data_validator] 	val:   0.0196, lp:   0.2156 mean dev:  -0.0027 RMS:   1.0582 conf:   0.0000
INFO  [data_validator] 	val:  -0.0115, lp:  -0.0300 mean dev:   0.0031 RMS:   0.3201 conf:   0.0000

INFO  [sensors] selected accel: 2162698 (0)
INFO  [data_validator] validator: best: 0, prev best: 0, failsafe: NO (0 events)
INFO  [data_validator] sensor #0, prio: 100, state: OK
INFO  [data_validator] 	val:  -0.3020, lp:  -0.3326 mean dev:  -0.0005 RMS:   0.0204 conf:   1.0000
INFO  [data_validator] 	val:   0.8947, lp:   0.8236 mean dev:  -0.0001 RMS:   0.0203 conf:   1.0000
INFO  [data_validator] 	val:  -9.8834, lp:  -9.8704 mean dev:   0.0003 RMS:   0.0314 conf:   1.0000
INFO  [data_validator] sensor #1, prio: 100, state: OK
INFO  [data_validator] 	val:   0.3629, lp:   0.2111 mean dev:  -0.0008 RMS:   1.0860 conf:   0.0000
INFO  [data_validator] 	val:   9.5235, lp:  95.2628 mean dev:  -1.0127 RMS: 516.7733 conf:   0.0100
INFO  [data_validator] 	val:  -1.9008, lp: -18.6376 mean dev:   0.1816 RMS: 101.8295 conf:   0.0200

INFO  [sensors] selected mag: 0 (0)
INFO  [data_validator] validator: best: -1, prev best: -1, failsafe: NO (0 events)

INFO  [vehicle_air_data] selected barometer: 4158993 (0)
INFO  [data_validator] validator: best: 0, prev best: 0, failsafe: NO (0 events)
INFO  [data_validator] sensor #0, prio: 75, state: OK
INFO  [data_validator] 	val: 100252.4531, lp: 100253.0313 mean dev:  -1.5176 RMS:   1.2020 conf:   1.0000
INFO  [data_validator] 	val:  33.5338, lp:  33.5278 mean dev:   0.0374 RMS:   0.0265 conf:   1.0000
INFO  [data_validator] 	val:   0.0000, lp:   0.0000 mean dev:   0.0000 RMS:   0.0000 conf:   1.0000

taileron, 2021-09-20:
The hmc5883 module was damaged in a crash. Now I would have to rely on the compass in the MPU9250 again ...
with current master, 9250 in bypass mode, the MPU9250/BMP280 module now starts inclusive mag with:
if mpu9250_i2c -X -b 2 -f 400 -a 0x68 -R 28 start; then
	sleep 1 # wait for mpu9250 to be configured with bypass enabled
	ak8963 -X -b 2 -R 18 start
	bmp280 -X -b 2 start
fi

but still with a lot of:
ERROR [sensors] Mag #0 fail:  TIMEOUT!

Is there a way to disable the 9250´s IMU while only using the mag and how can I reduce or avoid this mag errors?

dagar, 2021-09-20:
@taileron take a look at the ICM20948 which has a standalone helper (icm20948_i2c_passthrough) to put the device into I2C passthru mode (put's the AK magnetometer directly on the bus).

taileron, 2021-09-21:
In this way it seems to work after all - 9250 IMU stops while the built-in mag Ak9863 keeps running via the bypass:

if mpu9250_i2c -X -b 2 -a 0x68 -R 28 start; then
	sleep 1 # wait for mpu9250 to be configured with bypass enabled
	ak8963 -X -b 2 -R 18 start
	sleep 1
	mpu9250_i2c stop
	bmp280 -X -b 2 start
fi

dagar, 2021-09-21:
> In this way it seems to work after all - 9250 IMU stops while the built-in mag Ak9863 keeps running via the bypass:
>
> if mpu9250_i2c -X -b 2 -a 0x68 -R 28 start; then
> sleep 1 # wait for mpu9250 to be [...]

I haven't spent much time with them, but the Invensense sensors on I2C don't seem to be very stable, at least not when you're trying to get high rate data out of them. Probably solvable with some time, but not worth the effort with nearly all boards using them on SPI.

Can you replicate `icm20948_i2c_passthrough` for the mpu9250?

taileron, 2021-09-28:
I'm afraid I can't really do the replicate thing. When the sensor finally starts, it is flawlessly flyable. The only problem is that very often the driver just won't load, and it takes a few reboots until the mag is useable.