PX4/PX4-Autopilot #16386: FAILED: make px4_sitl gazebo

Theo-Zhang, 2020-12-13:
Ubuntu 20.04
gazebo11

FAILED: external/Stamp/sitl_gazebo/sitl_gazebo-configure 
cd /home/zhang/PX4-Autopilot/build/px4_sitl_default/build_gazebo && /usr/bin/cmake -DCMAKE_INSTALL_PREFIX=/usr/local -DSEND_ODOMETRY_DATA=ON -GNinja /home/zhang/PX4-Autopilot/Tools/sitl_gazebo && /usr/bin/cmake -E touch /home/zhang/PX4-Autopilot/build/px4_sitl_default/external/Stamp/sitl_gazebo/sitl_gazebo-configure
ninja: build stopped: subcommand failed.
make: *** [Makefile:226：px4_sitl] 错误 1

Theo-Zhang, 2020-12-13:
zhang@zhang:~/PX4-Autopilot$ make px4_sitl gazebo
[0/5] Performing configure step for 'sitl_gazebo'
-- install-prefix: /usr/local
-- cmake build type: RelWithDebInfo
-- Using C++17 standard
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found suitable version "1.71.0", minimum required is "1.58") found components: system thread filesystem 
-- Found DART: /usr/include (Required is at least version "6.6") found components: dart 
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found suitable version "1.71.0", minimum required is "1.40.0") found components: thread system filesystem program_options regex iostreams date_time 
-- Found Protobuf: /usr/lib/x86_64-linux-gnu/libprotobuf.so;-lpthread (found version "3.6.1") 
-- Looking for ignition-math6 -- found version 6.7.0
-- Searching for dependencies of ignition-math6
-- Looking for OGRE...
-- Found Ogre Ghadamon (1.9.0)
-- Found OGRE: optimized;/usr/lib/x86_64-linux-gnu/libOgreMain.so;debug;/usr/lib/x86_64-linux-gnu/libOgreMain.so
-- Looking for OGRE_Paging...
-- Found OGRE_Paging: optimized;/usr/lib/x86_64-linux-gnu/libOgrePaging.so;debug;/usr/lib/x86_64-linux-gnu/libOgrePaging.so
-- Looking for OGRE_Terrain...
-- Found OGRE_Terrain: optimized;/usr/lib/x86_64-linux-gnu/libOgreTerrain.so;debug;/usr/lib/x86_64-linux-gnu/libOgreTerrain.so
-- Looking for OGRE_Property...
-- Found OGRE_Property: optimized;/usr/lib/x86_64-linux-gnu/libOgreProperty.so;debug;/usr/lib/x86_64-linux-gnu/libOgreProperty.so
-- Looking for OGRE_RTShaderSystem...
-- Found OGRE_RTShaderSystem: optimized;/usr/lib/x86_64-linux-gnu/libOgreRTShaderSystem.so;debug;/usr/lib/x86_64-linux-gnu/libOgreRTShaderSystem.so
-- Looking for OGRE_Volume...
-- Found OGRE_Volume: optimized;/usr/lib/x86_64-linux-gnu/libOgreVolume.so;debug;/usr/lib/x86_64-linux-gnu/libOgreVolume.so
-- Looking for OGRE_Overlay...
-- Found OGRE_Overlay: optimized;/usr/lib/x86_64-linux-gnu/libOgreOverlay.so;debug;/usr/lib/x86_64-linux-gnu/libOgreOverlay.so
-- Looking for ignition-math6 -- found version 6.7.0
-- Looking for ignition-transport8 -- found version 8.1.0
-- Searching for dependencies of ignition-transport8
-- Found Protobuf: /usr/lib/x86_64-linux-gnu/libprotobuf.so;-lpthread (found suitable version "3.6.1", minimum required is "3") 
-- Config-file not installed for ZeroMQ -- checking for pkg-config
-- Checking for module 'libzmq >= 4'
--   Found libzmq , version 4.3.2
-- Checking for module 'uuid'
--   Found uuid, version 2.34.0
-- Looking for ignition-msgs5 -- found version 5.4.0
-- Searching for dependencies of ignition-msgs5
-- Looking for ignition-math6 -- found version 6.7.0
-- Checking for module 'tinyxml2'
--   Found tinyxml2, version 6.2.0
-- Looking for ignition-msgs5 -- found version 5.4.0
-- Looking for ignition-common3 -- found version 3.7.0
-- Searching for dependencies of ignition-common3
-- Looking for dlfcn.h - found
-- Looking for libdl - found
-- Searching for <ignition-common3> component [graphics]
CMake Error at /usr/share/cmake-3.16/Modules/CMakeFindDependencyMacro.cmake:47 (find_package):
  Could not find a configuration file for package "ignition-common3-graphics"
  that exactly matches requested version "3.7.0".

  The following configuration files were considered but not accepted:

    /usr/lib/x86_64-linux-gnu/cmake/ignition-common3-graphics/ignition-common3-graphics-config.cmake, version: 3.6.0
    /lib/x86_64-linux-gnu/cmake/ignition-common3-graphics/ignition-common3-graphics-config.cmake, version: 3.6.0

Call Stack (most recent call first):
  /usr/lib/x86_64-linux-gnu/cmake/ignition-common3/ignition-common3-config.cmake:203 (find_dependency)
  /usr/lib/x86_64-linux-gnu/cmake/gazebo/gazebo-config.cmake:209 (find_package)
  CMakeLists.txt:47 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/zhang/PX4-Autopilot/build/px4_sitl_default/build_gazebo/CMakeFiles/CMakeOutput.log".
See also "/home/zhang/PX4-Autopilot/build/px4_sitl_default/build_gazebo/CMakeFiles/CMakeError.log".
FAILED: external/Stamp/sitl_gazebo/sitl_gazebo-configure 
cd /home/zhang/PX4-Autopilot/build/px4_sitl_default/build_gazebo && /usr/bin/cmake -DCMAKE_INSTALL_PREFIX=/usr/local -DSEND_ODOMETRY_DATA=ON -GNinja /home/zhang/PX4-Autopilot/Tools/sitl_gazebo && /usr/bin/cmake -E touch /home/zhang/PX4-Autopilot/build/px4_sitl_default/external/Stamp/sitl_gazebo/sitl_gazebo-configure
ninja: build stopped: subcommand failed.
make: *** [Makefile:226：px4_sitl] Error 1

Theo-Zhang, 2020-12-13:
Can anybody help?

ijulanday, 2020-12-14:
Bumping this as I'm also having an issue on Ubuntu 20.04 with gazebo11.

Specifically:
`
CMake Error at /usr/share/cmake-3.16/Modules/CMakeFindDependencyMacro.cmake:47 (find_package):
  Could not find a configuration file for package "ignition-common3-graphics"
  that exactly matches requested version "3.7.0".

The following configuration files were considered but not accepted:

    /usr/lib/x86_64-linux-gnu/cmake/ignition-common3-graphics/ignition-common3-graphics-config.cmake, version: 3.6.0
    /lib/x86_64-linux-gnu/cmake/ignition-common3-graphics/ignition-common3-graphics-config.cmake, version: 3.6.0
`

This was while attempting to follow the [Plugins 101 tutorial](http://gazebosim.org/tutorials/?tut=plugins_hello_world), when trying to run cmake.

antonio-sc66, 2020-12-14:
Hi @Theo-Zhang, I've also had some problems, I think the best option right now if you have Ubuntu 20.04 is to use Docker (https://github.com/PX4/PX4-containers). For SITL Gazebo the right image should be: https://hub.docker.com/r/px4io/px4-dev-simulation-bionic. If case you're planing on using ROS and MAVROS you can check https://hub.docker.com/r/px4io/px4-dev-ros-melodic. There are also ROS2 images available

Jaeyoung-Lim, 2020-12-15:
@Theo-Zhang @antonio-sc66 @ijulanday The solution is to install the following package
```
apt install ignition-rendering3
```

Theo-Zhang, 2020-12-15:
> @Theo-Zhang @antonio-sc66 @ijulanday The solution is to install the following package
> 
> ```
> apt install ignition-rendering3
> ```

Thanks for your help！

ijulanday, 2020-12-15:
> @Theo-Zhang @antonio-sc66 @ijulanday The solution is to install the following package
> 
> ```
> apt install ignition-rendering3
> ```

Worked for me as well, thank you!

closed 2020-12-15 by Jaeyoung-Lim

SpencerHehl, 2021-01-04:
I tried installing and was unable to find that package. I ended up using the instructions at the following link to perform the install that appears to have worked (or at least gotten me farther in the install): https://ignitionrobotics.org/api/rendering/4.1/installation.html

lejan, 2021-04-26:
This worked for me:
sudo sh -c 'echo "deb http://packages.osrfoundation.org/gazebo/ubuntu-stable `lsb_release -cs` main" > /etc/apt/sources.list.d/gazebo-stable.list'
wget http://packages.osrfoundation.org/gazebo.key -O - | sudo apt-key add -
sudo apt update
sudo apt-get install libignition-common3-graphics

Waldri, 2021-08-05:
Following:

sudo apt-get update
sudo apt-get -y install wget lsb-release gnupg

sudo sh -c 'echo "deb http://packages.osrfoundation.org/gazebo/ubuntu-stable `lsb_release -cs` main" > /etc/apt/sources.list.d/gazebo-stable.list'

wget http://packages.osrfoundation.org/gazebo.key -O - | sudo apt-key add -

sudo apt-get update
sudo apt-get install libignition-rendering<#>-dev

Done!

alvgaona, 2021-09-15:
> @Theo-Zhang @antonio-sc66 @ijulanday The solution is to install the following package
> 
> ```
> apt install ignition-rendering3
> ```

On Ubuntu 20.04 it's:

```bash
sudo apt install libignition-rendering3
```

sankethj, 2021-11-07:
> I am also facing the same issue and i am not able to solve this issue using all the above written commands. does anyone have any other solution ? ![Screenshot from 2021-10-25 00-41-25](https://user-images.githubusercontent.com/57715378/138609118-03547e4f-b5af-4d79-a212-0dab0c327259.ppaste your error log?!

> I am also facing the same issue and i am not able to solve this issue using all the above written commands. does anyone have any other solution ? [image] [

`sudo apt install libignition-rendering3`

this solved for me

Qhijane, 2024-04-23:
> Ref

hello，When I execute “make px4_sitd_default Gazebo”， I use the same method as you to install the “sudo aptinstall libignition-rendering3” command, but the error still occurs: “/usr/include/sign/msgs5/sign/msgs/world_modify. pb.h: In member function 'void sign:: msgs: WorldModify: set-allocated_cloned_uri (std:: string *)':
/Usr/include/designation/msgs5/designation/msgs/world_modify. pb.h:438:15: error: 'struct Google:: protobuf:: internal:: ArenaStringPtr' has no member named 'SetAllocatedNoArena'; Did you mean 'SetAllocated'?
438 | cloned_uri_ SetAllocatedNoArena (&:: Google:: protobuf:: internal:: GetEmptyStringAlreadyInited(), cloned_uri);
|^~~~~~~~~~~~~~~~~~~
|SetAllocated
Ninja: build stopped: subcommand failed
FAILED: external/Stamp/sitl_gazebo class/sitl_gazebo class build
Cd/home/SLby/PX4/PX4 Autopilot/build/px4_sitL_default/build_gazebo class&&&/usr/bin/make -- build/home/SLby/PX4/PX4 Autopilot/build/px4_sitL_default/build_gazebo class -- j 18
Ninja: build stopped: subcommand failed
Make: * * * [Makefile: 227: px4_sitL_default] Error 1”
Is there any other solution?