PX4/PX4-Autopilot #16387: MPU9250 Magnetometer Disabled by Default?

Pskoog3, 2020-12-13:
Is it correct that the magnetometer of MPU9250 is disabled by default because it has some performance issue and is not required by any currently supported boards?

How to enable it.

dagar, 2020-12-13:
The mpu9250 magnetometer is disabled by default because having the i2c slave enabled (ak8963 magnetometer) messes up the scheduling. You can see this if you start the mpu9250 with and without `-M` and look at the `vehicle_imu` perf counters. The mpu9250 is providing gyro at 8 kHz and accel at 4 kHz, with the mag enabled it will be off by "some" undocumented amount.

mrpollo, 2020-12-15:
@Pskoog3 please open a new issue if you need anything else. Thank you.

closed 2020-12-15 by mrpollo