PX4/PX4-Autopilot #16389: Cube Orange: paraeters reset to default values

RamDG, 2020-12-15:
Hi,
I am facing an issue  similar to :
https://github.com/ArduPilot/ardupilot/issues/13134
https://discuss.ardupilot.org/t/cube-orange-randomly-reset-parameter-to-default/52588

when using  cube orange with kore  board.

I Don't  know if this issue has been resolved on px4 or not..!!!!
Please suggest .....

dagar, 2020-12-15:
What specifically you were doing when the parameters were lost? Has it happened more than once? Which version of PX4 exactly?

RamDG, 2020-12-15:
It  is happening  randomly  after 2-3 flights.  On  restarting the  board parameters are resetting to default.  it happened 4 times. 
Also, I tried resetting the boot loader  by setting SYS_BL_UPDATE parameter and now the board is not being detected. This also happened on 2 boards (cubeOrange only). board is booting up fine but USB not detected  and telemetry not connecting. 

any suggestion...?

dagar, 2020-12-15:
> On restarting the board parameters are resetting to default. it happened 4 times. and i am using 1.11.2 stable version.

What could be enlightening is to do some basic debugging specifically on the first boot where they appear to be gone. For example in NuttX run `dmesg` to see if the FRAM failed to start or failed to restore the parameters. Then next I'm wondering if you're careful to restart (without saving any parameters) is anything restored on the following bootup?

> Also, I tried resetting the boot loader by setting SYS_BL_UPDATE parameter and now the board is not being detected. 

Now being detected by QGC or at all (eg `dmesg` output)? It's possible it's working, but the USB device string naming changed.

dagar, 2020-12-15:
@amikhalev @JacobCrabill have either of you experienced a loss of parameters on a Cube Orange? I haven't, but so far have only used mine for bench testing.

RamDG, 2020-12-15:
> What could be enlightening is to do some basic debugging specifically on the first boot where they appear to be gone. For example in NuttX run `dmesg` to see if the FRAM failed to start or failed to [...]

i did flying  and landed . I replaced the batteries and found that the parameters got reset. But it is happening  randomly.

> Not being detected by QGC or at all (eg `dmesg` output)? It's possible it's working, but the USB device string naming changed.

Now the board is not detected at all .... both the board are new. both the boards on which i updated the boot-loader are powering up but not being detected at all (neither on windows nor on ubuntu). also there is no telemetry connection.

Now i cant connect both the boards, will have to fix the boot-loader issue first. To do the debugging.

RamDG, 2020-12-15:
Here are the files from SD card
[fault_2020_12_15_10_26_40.log](https://github.com/PX4/PX4-Autopilot/files/5697373/fault_2020_12_15_10_26_40.log)
[bootlog.txt](https://github.com/PX4/PX4-Autopilot/files/5697374/bootlog.txt)

dagar, 2020-12-15:
> Here are the files from SD card:
> [fault_2020_12_15_10_26_40.log](https://github.com/PX4/PX4-Autopilot/files/5697373/fault_2020_12_15_10_26_40.log)
> [bootlog.txt](https://github.com/PX4/PX4-Autopilot/files/5697374/bootlog.txt)

Interesting, let's see if this hard fault is the culprit.

``` Console
[hardfault_log] -- 2020-12-15-10:26:40 Begin Fault Log --
System fault Occurred on: 2020-12-15-10:26:40
 Type:Hard Fault in file:armv7-m/arm_hardfault.c at line: 134 running task: hpwork
 FW git-hash: aa244a098c39a2b21b82fd9cd5950ed566293fdc
 Build datetime: Dec 15 2020 15:54:47
 Build url: localhost 
 Processor registers: from 0x3800097c
 r0:0x380005a4 r1:0x081d5bbb  r2:0x38000a68  r3:0x00000000  r4:0x00000001  r5:0x00000000 r6:0x000000f0 r7:0x00000000
 r8:0x24001938 r9:0x00000080 r10:0xffffffff r11:0x01ff9cfe r12:0x00000000  sp:0x38000a50 lr:0x0815f08f pc:0x0815f098
 xpsr:0x21000000 basepri:0x000000f0 control:0x00000004
 exe return:0xffffffe9
 Fault status registers: from NVIC
 cfsr:0x00008200 hfsr:0x40000000  dfsr:0x00000000  mmfsr:0x00000000  bfsr:0x00000000 afsr:0x00000000 abfsr:0x00000000 
 IRQ stack: 
  top:    0x24001000
  sp:     0x24000f78 Valid
  bottom: 0x24000e00
  size:   0x00000200
Invalid Stack! (Corrupted TCB)  Stack base:  24000e00 Stack size:  00000200
 User stack: 
  top:    0x38000ad8
  sp:     0x38000a50 Valid
  bottom: 0x380005e4
  size:   0x000004f4
Invalid Stack! (Corrupted TCB)  Stack base:  380005e4 Stack size:  000004f4
Interrupt sp memory region, stack pointer lies within stack
```

PX4 side the only thing left running in NuttX's high priority work queue (HPWORK) is the shutdown hooks. 
The param save is one of the few things that actually hold the shutdown lock. https://github.com/PX4/PX4-Autopilot/blob/aa244a098c39a2b21b82fd9cd5950ed566293fdc/src/lib/parameters/parameters.cpp#L982-L986

JacobCrabill, 2020-12-15:
@dagar the fix from ArduPilot (which was actually to the bootloader, so may be relevant) is noted here: https://github.com/ArduPilot/ardupilot/issues/13134#issuecomment-597936386, and here: https://github.com/ArduPilot/ardupilot/pull/13780

The issue was a floating pin on FRAM, leading to reset of all values saved to FRAM, including parameter values.  Don't know if the PX4 bootloader has the same bug.

amikhalev, 2020-12-15:
I have experienced parameter loss a few times, although I assumed it was just any issue of my hardware.

That `hpwork` hardfault is probably #16133

JacobCrabill, 2021-02-12:
Just saw an update about this issue in the weekly ArduPilot forum digest; see here: https://discuss.ardupilot.org/t/plane-4-0-release/47188/434

mrpollo, 2021-03-03:
@RamDG hey, can you try the latest PX4 master and check if this fixes your problem?

RamDG, 2021-03-14:
@mrpollo  yeah sure will give it a try

mrpollo, 2021-04-28:
@RamDG hey were you able to give this a quick test again? can you maybe check with current master or the latest [beta-4 release](https://github.com/PX4/PX4-Autopilot/releases/v1.12.0-beta4)?

Saaadat, 2021-06-17:
Has this issue been resolved?

harrisonmg, 2021-09-23:
@mrpollo Could you point me to the commit you think might fix this issue? I would like to pull this change into my fork of PX4. Thanks!

mnumanuyar, 2022-06-09:
any update on this ?

mnumanuyar, 2022-06-09:
> @mrpollo yeah sure will give it a try

did it solved the issue ?

mnumanuyar, 2022-06-09:
@RamDG is this issue resolved

julianoes, 2023-04-03:
As far as I'm aware this still happens, however, thanks to the param backup on SD card by @dagar it gets hardly noticed these days.