PX4/PX4-Autopilot #16391: Logging of actuator_ouputs -> PWM limited to 80Hz

Lilienthal-GF, 2020-12-15:
**Describe the bug**
The actuator_output topic can only be logged with 80Hz.

**To Reproduce**
Steps to reproduce the behavior:
1. Flash PX4-Autopilot v1.11.2
2. Configure IMU_RATE and PWM_RATE to 400Hz
[image]
3. Configure SD-Logging Profile to High Rate
[image]
4. Log flight and extract measurements from ulog file

**Expected behavior**
Measurement should contain the signal actuator_outputs_0 with a rate of 400 Hz -> dt = 0.0025s.

**Log Files and Screenshots**
Plotting the time differences between consecutive ulog messages and the corresponding histogramm shows actutator_outputs to be logged at 80Hz -> 0.0125

[image]

**Drone (please complete the following information):**
- Holybro S500 with Holybro pixhawk4

dagar, 2020-12-15:
Did you reboot after updating `SDLOG_PROFILE`?

Lilienthal-GF, 2020-12-15:
Thank you for the quick response.

I did reboot the vehicle. In fact tested recalibrating the sensors and ESCs and complete air frame resets followed by the described calibration and a reboot. What i didn't do so far is test on a number of pixhawks.

dagar, 2020-12-15:
Can you share the full log? https://logs.px4.io/

dagar, 2020-12-15:
In addition to sharing the full log you could check a few things through the mavlink shell to see how fast things are actually running.

``` Console

nsh> uorb top

nsh> work_queue status

nsh> logger status
```

Lilienthal-GF, 2020-12-15:
I uploaded a measurement 

https://logs.px4.io/plot_app?log=7fdae510-e504-4021-b603-1621ce6ff967

For the measurement the quadcopter was armed without rotors and moved by hand. So the attitude does not respond to the actuator_outputs.

Here is the response to nsh> uorb top
[image]
[image]

and to nsh> work_queue status
[image]

Lilienthal-GF, 2020-12-15:
I'm sorry I forgot to do the nsh> logger status, but even so it looks like the rate_ctrl loop are running at the correct rate.

dagar, 2020-12-15:
@Lilienthal-GF actuator_outputs is only being published at ~ 50 Hz.

[image]

This is because on the Pixhawk 4 the MAIN outputs are actually going out to the px4io over serial, a separate stm32f1 processor.

Reading back the output is throttled to about 50 Hz here. https://github.com/PX4/PX4-Autopilot/blob/aa244a098c39a2b21b82fd9cd5950ed566293fdc/src/drivers/px4io/px4io.cpp#L923-L934

You can lift that limit in the code, but be careful to check the corresponding change in cpu usage.

Alternatively if you really care about getting the full data you might be better off putting the outputs on AUX where the main processor simply directly writes PWM and publishes `actuator_outputs` at corresponding to each actual update. You can even use things like oneshot or dshot if you want to remove additional latency (and your ESCs support it).

mrpollo, 2020-12-15:
@Lilienthal-GF looks like @dagar answered your question, feel free to re-open if you don't agree or open a new issue if you find something else. Thank you!

closed 2020-12-15 by mrpollo

Lilienthal-GF, 2020-12-16:
Thanks for answering my question. I think the issue can stay closed, although i do have a small follow up question.

I was not aware of this distribution of the functionality on the two controllers. The reason for trying to get the PWMs is that we would like to know the outputs of the quadcopter including the limiting and scaling in the mixer for system identification purposes.

So far, i was working under the assumption that the actuator_outputs get published from the mixer_module, i.e.

https://github.com/PX4/PX4-Autopilot/blob/97ab8277ed16a548ee2c933c3c56b6360b2d145b/src/lib/mixer_module/mixer_module.cpp#L385-L391

https://github.com/PX4/PX4-Autopilot/blob/97ab8277ed16a548ee2c933c3c56b6360b2d145b/src/lib/mixer_module/mixer_module.cpp#L414-L421

https://github.com/PX4/PX4-Autopilot/blob/97ab8277ed16a548ee2c933c3c56b6360b2d145b/src/lib/mixer_module/mixer_module.cpp#L428-L439

Can you point me to some documentation, where the interaction of the rate controller, the mixer, and the driver is explained in more details?