PX4/PX4-Autopilot #16425: Mavlink's COMMAND_LONG message sending false values

Safranek42, 2020-12-18:
**Describe the bug**
MAVLink message COMMAND_LONG sends false values when using VEHICLE_CMD_DO_PARACHUTE, and sending commands too fast (less than about 2 seconds) after each other.

**Steps to reproduce the behavior:**
I am using vehicle_command uORB message, and send VEHICLE_CMD_DO_PARACHUTE message with a parameter set to 0/1 on the first index. The MavlinkStreamCommandLong function in src/modules/mavlink/mavlink_messages.cpp is converting vehicle_command uORB topic to mavlink message, so it is broadcasted and can be seen in QGC's MAVLink Inspector under the tab COMMAND_LONG. When I gave command via the mavlink console (I previously set up Commander.cpp to have a command for sending the needed uORB message) it switched the mavlink message's value to the desired value. Then it is switching back and forth between the values a couple of times, and in the end it stays on the correct (commanded) number. (On the image after 3min 1sec I gave command "0", but no command was given after that. Yet, param1 jumps to 1 and to 0 again.) It only happens when I am giving new commands too fast, I have to wait 1.5-2 seconds between them to avoid this problem.

**Expected behavior**
The message should just switch once per every command.

[image]

**Drone (please complete the following information):**
- Holybro Durandal flight controller
- PX4 1.11 stable version

**Additional context**
The uORB message was correct according to the logfile, it was only the broadcasted mavlink message that showed this behaviour.
I could solve this bug with deleting `MavlinkCommandSender::instance().check_timeout(_mavlink->get_channel());` from line [533](https://github.com/PX4/PX4-Autopilot/blob/39c585ac97cf476b648555d5823a432e2a46e909/src/modules/mavlink/mavlink_messages.cpp#L533) from mavlink_messages.cpp. It works sufficiently now.
However there could be a better solution as I don't know exactly what this part of the code is doing.
The code that I am using to command new vehicle_command message from the mavlink console:

```
if(!strcmp(argv[1], "send_parachute")) {
  if (argc>2)
  {
	  if (!strcmp(argv[2], "open"))
	  {
		  send_vehicle_command(vehicle_command_s::VEHICLE_CMD_DO_PARACHUTE, 1);
	  }
	  else if (!strcmp(argv[2], "close"))
	  {
		  send_vehicle_command(vehicle_command_s::VEHICLE_CMD_DO_PARACHUTE, 0);
	  }
	  else
	  {
		  PX4_ERR("Not supported argument. Supported arguments:\nopen\nclose.");
	  }
  }
  else
  {
	  PX4_ERR("Missing argument: open/close.");
  }
  
  return 0;
}
```

dayjaby, 2020-12-18:
Is the VEHICLE_CMD_DO_PARACHUTE being acknowledged with a COMMAND_ACK? If not, the same command might be sent up to 4 times with increasing values in the "confirmation field". According to your picture exactly that happens here.

Safranek42, 2020-12-18:
Yes, that might be the case.
As I understand now, I need to publish vehicle_command_ack message from the same place with VEHICLE_RESULT_ACCEPTED value to have a successful broadcast of the new value in vehicle_command, right? (And other parameters set like timestamp, command, target_system of course.)

dayjaby, 2020-12-18:
Usually, the recipient for the COMMAND_LONG/DO_PARACHUTE should answer with an COMMAND_ACK. Is some other part of the PX4 firmware the recipient or is it some external module where you cant modify the code?

You could also do other approaches: modify send_vehicle_command to set the confirmation field to 255 for DO_PARACHUTE and disable the re-sending of COMMAND_LONGs if vehicle_command.confirmation is 255.

Safranek42, 2020-12-21:
Thanks for your help @dayjaby!

I am trying to send to an external module, and I am currently testing in QGroundControl because it is easy to see mavlink messages there.
However when I am sending the DO_PARACHUTE message, and QGroundControl receives it, also COMMAND_ACK is shown in the MAVLink Inspector, with "command" field as the command I sent (208 in the case of DO_PARACHUTE), and all other parameters as 0. The same happens for flight mode change with `commander mode *` (and the same behavior as described before). Is that the acknowledgement sent by QGC (and then there is some problem capturing it in PX4) or is it something QGC receives? (As the target_system field is set to 0 it should be a broadcast message and should be captured by PX4 regardless of its actual system target number/system ID.)

Deleting the `MavlinkCommandSender::instance().check_timeout(_mavlink->get_channel());` line from mavlink_messages.cpp does what you suggested, and does not resend the message if it doesn't get acknowledgement if I am seeing it right. (For COMMAND_LONG messages, regardless of the command id.)

dayjaby, 2021-01-18:
Hey, sorry for late reply. QGroundControl does not acknowledge anything. What is the result field of the COMMAND_ACK message in QGC? Is your command id also visible in the uORB topic vehicle_command_ack (you can test with `listener vehicle_command_ack` in mavlink console in QGC)?