PX4/PX4-Autopilot #16427: VTOL standard flight to review

dagar, 2020-12-18:
https://review.px4.io/plot_app?log=4240e1a9-5f40-4e5b-be73-812f59a072e3

https://px4.slack.com/archives/C39C4E709/p1608315970050900

dagar, 2020-12-18:
[image]

dagar, 2020-12-18:
Previous good flight. https://review.px4.io/plot_app?log=b8cd65dc-fda0-4490-9178-69567272c632

dagar, 2020-12-18:
That's weird, the log doesn't have `tecs_status`.

dagar, 2020-12-18:
I don't think there's any bug here, this is a STABILIZED mode flight that stalled.

[image]

dagar, 2020-12-18:
I think I jumped the gun on this one, the roll departure actually happens during transition (FW -> MC).

[image]

Why is FW roll and pitch actuation at maximum during the transition?

dayjaby, 2021-01-02:
The rampup time for MC motors seems to be a bit too long, which causes pitch/roll instability. I assume your MC motors are not strong enough to handle your VTOL rolling, especially not if the thrust is still ramping up.

sfuhrer, 2021-01-04:
The ramp up of the MC actuation is coupled with the airspeed measurement - and that happens to be still quite high during the back-transition when the roll angle diverges.  

During the previous "healthy" flight, the airspeed even doesn't drop below the blending airspeed when the transition is finished, meaning it will keep the MC actuators at idle during the whole transition.
[image]

The high airspeed measurement is (at least partially) due to the sensor being badly calibrated (ASPD_SCALE param would need to be set to about 0.8).  
[image]

I also wonder if the transition was done into the wind or with tailwind. From what I see in the log it's rather with the wind then against.

I will investigate where the multiple quadchute messages are coming from, it should just be one. Further, I will work on the airspeed scale factor estimation --> it should update the scale factor automatically and not via a user input/manually started calibration flight.

ThomasRigi, 2021-01-07:
@sfuhrer the quadchute logic is broken for back transitions. This function: https://github.com/PX4/PX4-Autopilot/blob/master/src/modules/vtol_att_control/vtol_att_control_main.cpp#L197 . The logic only works for front transitions and FW flight, but not back transitions. (which kind of explains why the quadchute triggering function is called `abort_front_transition`...)

A triggered QC sets `_abort_front_transition` to true. But since here it was during a manual back transition `to_fw` is false and resets the quadchute failsafe. 

Also, https://github.com/PX4/PX4-Autopilot/blob/master/src/modules/vtol_att_control/standard.cpp#L99 needs to be adapted I think