PX4/PX4-Autopilot #16436: Error in ```make px4_sitl_default gazebo```

AmrapaliKhandare, 2020-12-22:
After running command
```make px4_sitl_default gazebo``` 
i am getting the following error :
```pxh> [Err] [GuiIface.cc:124] Cannot mix incompatible Qt library (version 0x50905) with this library (version 0x50c05)```
```
INFO  [ecl/EKF] 532000: reset position to last known position
INFO  INFO  [ecl/EKF] 532000: reset velocity to zero
INFO  [ecl/EKF] 532000: reset position to last known position
INFO  [ecl/EKF] 532000: reset velocity to zero
[ecl/EKF] 532000: reset position to last known position
INFO  [ecl/EKF] 532000: reset velocity to zero
INFO  [ecl/EKF] 532000: reset position to last known position
INFO  [ecl/EKF] 532000: reset velocity to zero
INFO  [ecl/EKF] 536000: reset position to last known position
INFO  [ecl/EKF] 536000: reset velocity to zero
INFO  [ecl/EKF] 540000: reset position to last known position
INFO  [ecl/EKF] 540000: reset velocity to zero
INFO  INFO  [ecl/EKF] 804000: GPS checks passed[ecl/EKF] 
INFO  808000: GPS checks passedINFO  [ecl/EKF] 
812000: GPS checks passed
[ecl/EKF] 808000: GPS checks passed
INFO  [ecl/EKF] 808000: GPS checks passed
INFO  [ecl/EKF] 808000: GPS checks passed
INFO  [ecl/EKF] INFO  INFO  [ecl/EKF] 6848000: EKF aligned, (baro hgt, IMU buf: 12, OBS buf: 9)
[ecl/EKF] 6848000: EKF aligned, (baro hgt, IMU buf: 12, OBS buf: 9)
6848000: EKF aligned, (baro hgt, IMU buf: 12, OBS buf: 9)
INFO  [ecl/EKF] 6848000: EKF aligned, (baro hgt, IMU buf: 12, OBS buf: 9)
INFO  [ecl/EKF] 6848000: EKF aligned, (baro hgt, IMU buf: 12, OBS buf: 9)
INFO  [ecl/EKF] 6852000: EKF aligned, (baro hgt, IMU buf: 12, OBS buf: 9)
INFO  [ecl/EKF] 7232000: reset position to GPS
INFO  [ecl/EKF] 7232000: reset velocity to GPS
INFO  [ecl/EKF] 7232000: starting GPS fusion
INFO  [ecl/EKF] 7240000: reset position to GPS
INFO  [ecl/EKF] 7240000: reset velocity to GPS
INFO  [ecl/EKF] 7240000: starting GPS fusion
INFO  [ecl/EKF] 7236000: reset position to GPS
INFO  [ecl/EKF] 7236000: reset velocity to GPS
INFO  [ecl/EKF] 7236000: starting GPS fusion
INFO  [ecl/EKF] 7236000: reset position to GPSINFO  [ecl/EKF] 7236000: reset position to GPS
INFO  [ecl/EKF] 7236000: reset velocity to GPS
INFO  [ecl/EKF] 7236000: starting GPS fusion

INFO  [ecl/EKF] 7236000: reset velocity to GPS
INFO  [ecl/EKF] 7236000: starting GPS fusion
INFO  [ecl/EKF] 7236000: reset position to GPS
INFO  [ecl/EKF] 7236000: reset velocity to GPS
INFO  [ecl/EKF] 7236000: starting GPS fusion
INFO  [tone_alarm] home set
ERROR [tone_alarm] notify negative
```

Jaeyoung-Lim, 2020-12-22:
@AmrapaliKhandare 

> [Err] [GuiIface.cc:124] Cannot mix incompatible Qt library (version 0x50905) with this library (version 0x50c05)

You have Qt dependency conflicts. Can you run gazebo separately?

AmrapaliKhandare, 2020-12-23:
No. When I run 
```gazebo``` command. Gazebo window doesn't open.
After running the command
 ```gazebo --verbose```
I get same Qt dependency error. Following is the error
```
Gazebo multi-robot simulator, version 9.16.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
Gazebo multi-robot simulator, version 9.16.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 10.10.5.42
[Msg] Loading world file [/usr/share/gazebo-9/worlds/empty.world]
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 10.10.5.42
[Err] [GuiIface.cc:124] Cannot mix incompatible Qt library (version 0x50905) with this library (version 0x50c05)
[Wrn] [Publisher.cc:185] Queue limit reached for topic /gazebo/default/physics/contacts, deleting message. This warning is printed only once.
```

LorenzMeier, 2020-12-29:
You will need to reinstall your Gazebo environment - this error is related to your system setup, not to PX4.

closed 2020-12-29 by LorenzMeier