PX4/PX4-Autopilot #16440: Automatic airspeed scale calibration for FW and VTOL

moreba1, 2020-12-23:
**Describe problem solved by the proposed feature**
Flying in different locations need new airspeed calibration because of air density change with AMSL altitude. so we need to calibrate the airspeed scale in new locations. but doing it manually is difficult and may be dangerous for users.

**Describe your preferred solution**
After takeoff  In a loiter with constant altitude and for a duration, airspeed calibration activated then autopilot measure average of groundspeed in every round and use for airspeed calibration.

LorenzMeier, 2020-12-28:
We are calculating the true airspeed already based on the current altitude. What use case do you have that would require additional calibration?

dagar, 2020-12-28:
The airspeed scale factor is still often worthwhile for things like pitot position error. Currently it's not safe to keep airspeed scale calibration enabled at all times (ASPD_SCALE_EST), but it's something @sfuhrer and I have discussed recently. Either an explicit calibration flight (large loiter) or monitor a normal flight to consider committing the estimated TAS scale if it converged, everything checks out, and other properties were satisfied (eg enough turning).

moreba1, 2020-12-29:
> We are calculating the true airspeed already based on the current altitude. What use case do you have that would require additional calibration?

I calibrated the airspeed scale at 1000 m AMSL altitude, but in the second flight at 2000 m AMSL altitude it was uncalibrated.

sfuhrer, 2021-01-04:
> I calibrated the airspeed scale at 1000 m AMSL altitude, but in the second flight at 2000 m AMSL altitude it was uncalibrated.

Could you share logs from these two flights?

And yes, I think we should work towards having the estimation enabled always, with checks in place to not create dangerous situations.

moreba1, 2021-01-05:
> Either an explicit calibration flight (large loiter)

This is good. calibration of airspeed scale in a loiter in first of each flight.

moreba1, 2021-01-13:
[image]

moreba1, 2021-01-14:
[image]

sfuhrer, 2021-01-19:
Hi @moreba1, the controller will always try to keep the IAS (resp. CAS) constant. CAS is passed into TECS, and there converted to a TAS setpoint, as that's what TECS internally needs. Also the tecs_status.airspeed_sp is in TAS. So it's normal that FW_AIRSPD_TRIM is not equal to tecs_status.airspeed_sp. 
This PR https://github.com/PX4/PX4-Autopilot/pull/16585 should make it more clear for the future. 
I think it makes sense though to display CAS/EAS in the HUD of the groundstation, as if you fly manually, that is the airspeed feedback you're most concerned for.

moreba1, 2021-01-19:
>  the controller will always try to keep the IAS

1- if FW_AIRSPD_TRIM and airspeed in HUD are not the same when operator will be confused.
2. EAS2TAS is different in each flight, so TAS will different in each flight. Designers want to set TAS constant in all drone flights
3. If CAS displayed in HUD when we will set scale calibration. we will be confused because the average of min and max of groundspeed in loiter equal to TAS, not CAS. 

I think  FW_AIRSPD_TRIM and HUD airspeed must set to TAS.

sfuhrer, 2021-01-19:
>     1- if FW_AIRSPD_TRIM and airspeed in HUD are not the same when operator will be confused.

FW_AIRSPD_TRIM is a very safety related system setting, that no operator should need to touch. Only developers. And they should know the difference between IAS/CAS and TAS. If you want to have user-facing airspeed setpoints, then I would recommend using change speeds commands in the groundstation. 

>2. EAS2TAS is different in each flight, so TAS will different in each flight. Designers want to set TAS constant in all drone flights

It will only be different if you fly at different altitudes. I don't see that problem of having a changing TAS depending on the current altitude, that's standard for any sort of planes. 

>3. If CAS displayed in HUD when we will set scale calibration. we will be confused because the average of min and max of groundspeed in loiter equal to TAS, not CAS.

I don't really get your question. The scale (from IAS to CAS) can be estimated automatically, or then set manually by comparing groundspeed and TAS while loitering (note that the scale is also applied to TAS). 

>I think FW_AIRSPD_TRIM and HUD airspeed must set to TAS.

I disagree with the FW_AIRSPD_TRIM part, but would agree that having TAS displayed in the HUD would be beneficial (but only if we keep CAS).

moreba1, 2021-01-19:
> It will only be different if you fly at different altitudes. I don't see a problem in having a changing TAS depending on the current altitude, that's standard for any sort of planes.

Flight performance (l/d) depends on the angle of attack and AOA depends on TAS, not IAS, so developers want to maintain TAS constant. 
Also, the range of ground speed needs to TAS and wind , if TAS be more max ground speed will be more and it is not desirable for aerial imaging.