PX4/PX4-Autopilot #16456: Java runtime exception

lweingart, 2020-12-28:
Dear team,

I just followed the instructions to install everything on macOS following the youtube video on your website:
https://www.youtube.com/watch?v=tMbMGiMs1cQ

meaning, among other things I installed AdoptOpenJDK14.

Everything seems to have run smoothly, but when I ran
```
make px4_sitl jmavsim
```
it ended up with a BUILD SUCCESSFUL, directly followed by an Exception, and then nothing anymore.
Here is the result
```
______  __   __    ___
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 0
Info: found model autostart file as SYS_AUTOSTART=10016
INFO  [param] selected parameter default file eeprom/parameters_10016
[param] parameter file not found, creating eeprom/parameters_10016
  SYS_AUTOCONFIG: curr: 0 -> new: 1
  SYS_AUTOSTART: curr: 0 -> new: 10016
  BAT_N_CELLS: curr: 0 -> new: 4
  CAL_ACC0_ID: curr: 0 -> new: 1310988
  CAL_ACC1_ID: curr: 0 -> new: 1310996
  CAL_ACC2_ID: curr: 0 -> new: 1311004
  CAL_GYRO0_ID: curr: 0 -> new: 1310988
  CAL_GYRO1_ID: curr: 0 -> new: 1310996
  CAL_GYRO2_ID: curr: 0 -> new: 1311004
  CAL_MAG0_ID: curr: 0 -> new: 197388
  CAL_MAG1_ID: curr: 0 -> new: 197644
  CBRK_SUPPLY_CHK: curr: 0 -> new: 894281
  COM_RC_IN_MODE: curr: 0 -> new: 1
  EKF2_REQ_GPS_H: curr: 10.0000 -> new: 0.5000
  EKF2_MULTI_IMU: curr: 0 -> new: 3
  SENS_IMU_MODE: curr: 1 -> new: 0
  EKF2_MULTI_MAG: curr: 0 -> new: 2
  SENS_MAG_MODE: curr: 1 -> new: 0
  SDLOG_MODE: curr: 0 -> new: 1
  SDLOG_PROFILE: curr: 1 -> new: 131
  SDLOG_DIRS_MAX: curr: 0 -> new: 7
  SENS_BOARD_X_OFF: curr: 0.0000 -> new: 0.0000
  SENS_DPRES_OFF: curr: 0.0000 -> new: 0.0010
  TRIG_INTERFACE: curr: 4 -> new: 3
  COM_CPU_MAX: curr: 90.0000 -> new: -1.0000
  IMU_INTEG_RATE: curr: 200 -> new: 250
  NAV_ACC_RAD: curr: 10.0000 -> new: 2.0000
  RTL_RETURN_ALT: curr: 60.0000 -> new: 30.0000
  RTL_DESCEND_ALT: curr: 30.0000 -> new: 10.0000
  RTL_LAND_DELAY: curr: -1.0000 -> new: 0.0000
  PWM_MAX: curr: 2000 -> new: 1950
  PWM_MIN: curr: 1000 -> new: 1075
  GPS_UBX_DYNMODEL: curr: 7 -> new: 6
* SYS_AUTOCONFIG: curr: 1 -> new: 0
INFO  [dataman] Unknown restart, data manager file './dataman' size is 11798680 bytes
INFO  [simulator] Waiting for simulator to accept connection on TCP port 4560
Buildfile: /Users/lweingart/Developer/droning/git/Firmware/Tools/jMAVSim/build.xml

make_dirs:
    [mkdir] Created dir: /Users/lweingart/Developer/droning/git/Firmware/Tools/jMAVSim/out/production/jMAVSim

compile:
    [javac] Compiling 76 source files to /Users/lweingart/Developer/droning/git/Firmware/Tools/jMAVSim/out/production/jMAVSim

create_run_jar:
      [jar] Building jar: /Users/lweingart/Developer/droning/git/Firmware/Tools/jMAVSim/out/production/jmavsim_run.jar

copy_res:
     [copy] Copying 17 files to /Users/lweingart/Developer/droning/git/Firmware/Tools/jMAVSim/out/production

BUILD SUCCESSFUL
Total time: 2 seconds
Exception in thread "main" java.lang.UnsupportedClassVersionError: me/drton/jmavsim/Simulator has been compiled by a more recent version of the Java Runtime (class file version 59.0), this version of the Java Runtime only recognizes class file versions up to 58.0
	at java.base/java.lang.ClassLoader.defineClass1(Native Method)
	at java.base/java.lang.ClassLoader.defineClass(ClassLoader.java:1017)
	at java.base/java.security.SecureClassLoader.defineClass(SecureClassLoader.java:151)
	at java.base/java.net.URLClassLoader.defineClass(URLClassLoader.java:514)
	at java.base/java.net.URLClassLoader$1.run(URLClassLoader.java:422)
	at java.base/java.net.URLClassLoader$1.run(URLClassLoader.java:416)
	at java.base/java.security.AccessController.doPrivileged(AccessController.java:691)
	at java.base/java.net.URLClassLoader.findClass(URLClassLoader.java:415)
	at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:589)
	at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:522)
	at java.base/java.lang.Class.forName0(Native Method)
	at java.base/java.lang.Class.forName(Class.java:427)
	at org.eclipse.jdt.internal.jarinjarloader.JarRsrcLoader.main(JarRsrcLoader.java:59)
```

Any help would be greatly appreciated.
Have a nice day

Best Regards,
Laurent

LorenzMeier, 2020-12-28:
Can you run `make clean` and run again?

lweingart, 2020-12-28:
I just did, with the exact same result

LorenzMeier, 2020-12-28:
[image]

I just ran it successfully. Could you try `make distclean` and try again? What is the output of `java --version`? Mine is:

`openjdk 15 2020-09-15
OpenJDK Runtime Environment AdoptOpenJDK (build 15+36)
OpenJDK 64-Bit Server VM AdoptOpenJDK (build 15+36, mixed mode, sharing)`

lweingart, 2020-12-28:
Hi, so:
```
java --version
openjdk 14.0.2 2020-07-14
OpenJDK Runtime Environment AdoptOpenJDK (build 14.0.2+12)
OpenJDK 64-Bit Server VM AdoptOpenJDK (build 14.0.2+12, mixed mode, sharing)
```

This probably explains it, I should update to version 15 probably.

Otherwise, `make distclean` and then `make px4_sitl jmavsim` again and still the same exception.
I will try and install a more recent java version then.

lweingart, 2020-12-28:
Alright, so I updated to adoptopenJDK 15 and now it works fine indeed.

Thank you very much for your help and for your responsiveness !

Cheers,
Laurent

LorenzMeier, 2020-12-28:
My pleasure!

closed 2020-12-28 by LorenzMeier