PX4/PX4-Autopilot #16459: Issue with MavlinkFTP of pixhawk1

1AwesomeDeveloper, 2020-12-29:
hello all, I am trying using PX4 Pro Stable Release v1.11.2 on PX4 FMU v2, and I am trying to download a file via Mavlink FTP.
But when I try to request a file via Mavlink FTP the board doesnot repond to this operation, and I get a timeout error. Can anyone please tell me that why this is happening, and is Mavlink FTP available for this old version of Pixhawk ?

LorenzMeier, 2020-12-29:
No, it is not available, since the board is flash constrained. You might be lucky though and your manufacturer might have mis-classified its FMUv3 as a v2 and you can convert it using the bootloader update:
https://docs.px4.io/master/en/config/firmware.html#bootloader

closed 2020-12-29 by LorenzMeier

1AwesomeDeveloper, 2020-12-29:
[image]
Tried bootloader update FMUv3 was installed but didn't worked for my FTP Problem...still the board is not responding

LorenzMeier, 2020-12-29:
Could you share a screenshot of the firmware upgrade?

1AwesomeDeveloper, 2020-12-29:
[image]
here it is

LorenzMeier, 2020-12-29:
Thanks! Let me quickly cross-check!

LorenzMeier, 2020-12-29:
Which QGroundcontrol version are you using?

LorenzMeier, 2020-12-29:
I checked and on my setup MAVLink FTP is enabled: 
[image]

1AwesomeDeveloper, 2020-12-29:
I updated the firmware from stable version of QGroundControl v4.0.10. 
I forked the main repo to my github (https://github.com/1AwesomeDeveloper/qgroundcontrol/tree/Spacejam_NPNT) and here I am doing changes. Applogies I don't know how to check the version of QGC I am working on

1AwesomeDeveloper, 2020-12-29:
when I try to check, from debug build I get this 
[image]

1AwesomeDeveloper, 2020-12-29:
here is how you can get an idea what I am trying to do.
https://github.com/mavlink/qgroundcontrol/issues/9275

LorenzMeier, 2020-12-29:
The URL of the file is incorrect - any file on the SD card is on `/fs/microsd/key.txt`, NOT `/key.txt`

LorenzMeier, 2020-12-29:
Make sure you can see the file in the MAVLink shell - `ls /fs/microsd/key.txt` should list it.

1AwesomeDeveloper, 2020-12-29:
I tried to run on Mavlink Console but it shows 
[image]

LorenzMeier, 2020-12-29:
Use a QGC daily build please so we're sure you have the latest and something proven (it works for me).

1AwesomeDeveloper, 2020-12-29:
I have key.txt in my sd card 
[image]
But when I look through mavlink console it doesn't show me 
[image]

1AwesomeDeveloper, 2020-12-29:
Sir, Now I am using the stable build.

1AwesomeDeveloper, 2020-12-29:
Sir by default my SD card has LOGS directory (note the caps), but in the Mavlink Console I have log(in small letters), I guess this is the issue but don't know how to fix it
Any Idea ?

LorenzMeier, 2020-12-29:
The screenshot from your SD card shows that you don't have the file `key.txt` in the root of the SD card. You put it into a folder and so it can't be found on the path that you're attempting. Please make sure to check the file path and your steps very carefully. The issue tracker for an open source project is not intended for 1:1 debugging support.

I'm more than happy to resolve any real issues.

1AwesomeDeveloper, 2020-12-29:
> Make sure you can see the file in the MAVLink shell - `ls /fs/microsd/key.txt` should list it.

Earlier it was not showing but I it showed after mounting the SD card on the PX4 filesystem using mount -t vfat /dev/mmcsd0 /fs/microsd

1AwesomeDeveloper, 2020-12-29:
> The screenshot from your SD card shows that you don't have the file `key.txt` in the root of the SD card. You put it into a folder and so it can't be found on the path that you're attempting. Please [...]

thanks a lot. I will ask on forums for 1:1 debugging support.

1AwesomeDeveloper, 2020-12-29:
> o it can't be found on the path that you're attempting

in root path also it didn't worked

LorenzMeier, 2021-01-01:
I will be using this for mixer configurations myself in the next days and will post my working branch here. You might just need a working example to work from.