PX4/PX4-Autopilot #16464: Intermittent SITL failures

LorenzMeier, 2020-12-30:
@julianoes @dagar I did dig a little bit through the current intermittent SITL issues and what I see is that:

- It always happens on the first test (so its initialization)
- It happens with MAVSDK and with MAVROS (example: https://github.com/PX4/PX4-Autopilot/runs/1623874905?check_suite_focus=true#step:10:226)
- It seems network connection / simulator module / mavlink related (see the message from MAVROS below):
[image]

dagar, 2020-12-30:
One thing to be aware of is regular mavlink instances aren't covered by PX4 lockstep. It could still be possible to starve the connection to MAVSDK while the simulator and PX4 (sensors, estimator, controllers, output) are still working. I'm not yet suggesting this is a problem we're encountering.

LorenzMeier, 2020-12-31:
Another one: "Waiting for system to be ready" - I've seen weird behavior locally as well: https://github.com/PX4/PX4-Autopilot/pull/16466/checks?check_run_id=1630000477#step:17:178

dagar, 2021-01-01:
Here's a fresh one. https://github.com/PX4/PX4-Autopilot/pull/16473/checks?check_run_id=1634086648

[image]