PX4/PX4-Autopilot #16474: undesirable back-transition behavior.

yuthikasagarage, 2021-01-02:
when using just the vtol-transition mission item in qgc, or flipping the switch from fw to mc mode stabilized mode, the curved path gets executed with a rolling behavior in the back transition

this is solved in the pr-vtol-rtl-fixes when using vtol transition and land mode, but however it still persists on the above mentioned conditions.

**Alternate Solution**
https://github.com/PX4/PX4-Autopilot/blob/d502292d0731a38c513dc1224c135b6f1965586e/src/modules/vtol_att_control/standard.cpp#L283
in the current master, _fw_virtual_att_sp->roll_body diverges from zero after some time in the back-transition, can we make it a hard zero as we don't want the aircraft to roll in the back-transition anyway?

**To Reproduce**
to reproduce, in a mission set a vtol-transition mission item at the end of the mission. the behavior can be observed.

**Expected behavior**
The vtol should transition to mc_mode in a straight line without going in a curved path with a roll.

**Log Files and Screenshots**
https://review.px4.io/plot_app?log=1e1bb5d2-5260-40da-bd7e-46fadf55fc6c

**Additional context**

LorenzMeier, 2021-01-02:
Here is the PR: https://github.com/PX4/PX4-Autopilot/pull/16475

sfuhrer, 2021-01-06:
Alternative solution, reduces the overshoots: https://github.com/PX4/PX4-Autopilot/pull/16499