PX4/PX4-Autopilot #16478: RPi3: ERROR [gps] ERR: -1 (tcsetattr)

Pskoog3, 2021-01-03:
I am using px4 build for Raspberry Pi 3 with custom sensors. The px4 starting is fine but after some time I get this error…
ERROR [gps] ERR: -1 (tcsetattr)
What is this error and its cause? The GPS is locks well and has a 3D fix. The GPS module is ublox LEA-6H with ubx output.

dagar, 2021-01-03:
That's weird, are you sure you have a lock and see valid data before this error message?

https://github.com/PX4/PX4-Autopilot/blob/8c71ecd97e3da679ae9a4c2ff22565be5bf60c87/src/drivers/gps/gps.cpp#L561-L563

If it's easily reproducible you could start by printing error (and strerrr(errno)).

Pskoog3, 2021-01-09:
The error is not consistent... sometimes it works for long without error and sometimes the error starts appearing when px4 is started. 

Perhaps the problem is related to the power of RPi... powering everything from a single source causes the RPi power led (red) to blink. I have separated the power of RPi and telemetry from rest of the sensors and GPS and so far it is working. But sometimes the error do appear at the very start when GPS is actually not locked.