PX4/PX4-Autopilot #16480: ranger finder valid check

garlinplus, 2021-01-04:
As followed,when _time_last_valid_us is timeout,the updateStuckCheck can be executed.But in updateValidity function, only variable _is_stuck is false, the varible _time_last_valid_us can be set false.
So the varible _is_stuck can not be set false?
void SensorRangeFinder::updateStuckCheck()
{
	// Check for "stuck" range finder measurements when range was not valid for certain period
	// This handles a failure mode observed with some lidar sensors
	if (((_sample.time_us - _time_last_valid_us) > (uint64_t)10e6)) {

		// require a variance of rangefinder values to check for "stuck" measurements
		if (_stuck_max_val - _stuck_min_val > _stuck_threshold) {
			_stuck_min_val = 0.0f;
			_stuck_max_val = 0.0f;
			_is_stuck = false;

		} else {
			if (_sample.rng > _stuck_max_val) {
				_stuck_max_val = _sample.rng;
			}

			if (_stuck_min_val < 0.1f || _sample.rng < _stuck_min_val) {
				_stuck_min_val = _sample.rng;
			}

			_is_stuck = true;
		}
	}
}

void SensorRangeFinder::updateValidity(uint64_t current_time_us)
{
	updateDtDataLpf(current_time_us);

	if (isSampleOutOfDate(current_time_us) || !isDataContinuous()) {
		_is_sample_valid = false;
		return;
	}

	// Don't run the checks unless we have retrieved new data from the buffer
	if (_is_sample_ready) {
		_is_sample_valid = false;

		if (_sample.quality == 0) {
			_time_bad_quality_us = current_time_us;

		} else if (current_time_us - _time_bad_quality_us > _quality_hyst_us) {
			// We did not receive bad quality data for some time

			if (isTiltOk() && isDataInRange()) {
				updateStuckCheck();

				if (!_is_stuck) {
					_is_sample_valid = true;
					_time_last_valid_us = _sample.time_us;
				}
			}
		}
	}
}

bresch, 2021-01-04:
When `_is_stuck` is `true`, the `_time_last_valid_us` is not updated anymore. This means that `if (((_sample.time_us - _time_last_valid_us) > (uint64_t)10e6))` (in `updateStuckCheck`) will stay `true` and update the `_stuck_[min|max]_val` at every loop.
At some point, if the range measurement is moving enough, the `if (_stuck_max_val - _stuck_min_val > _stuck_threshold)` check will pass and `_is_stuck` will become `false`, the sample is declared valid and `_time_last_valid_us` is updated.

Am I missing anything? 

There is a [unit test](https://github.com/PX4/PX4-ECL/blob/78ce46f00777f07d24c31bdc5e2edb203be2c0f4/test/test_SensorRangeFinder.cpp#L206-L236) for this logic that could be extended to show the part where `_is_stuck` goes from `true` to `false`. You could play with it to see if it really works and show the problem.

closed 2021-01-05 by priseborough

garlinplus, 2021-01-06:
> extended to show th

> When `_is_stuck` is `true`, the `_time_last_valid_us` is not updated anymore. This means that `if (((_sample.time_us - _time_last_valid_us) > (uint64_t)10e6))` (in `updateStuckCheck`) will stay `true` and update the `_stuck_[min|max]_val` at every loop.
> At some point, if the range measurement is moving enough, the `if (_stuck_max_val - _stuck_min_val > _stuck_threshold)` check will pass and `_is_stuck` will become `false`, the sample is declared valid and `_time_last_valid_us` is updated.
> 
> Am I missing anything?
> 
> There is a [unit test](https://github.com/PX4/PX4-ECL/blob/78ce46f00777f07d24c31bdc5e2edb203be2c0f4/test/test_SensorRangeFinder.cpp#L206-L236) for this logic that could be extended to show the part where `_is_stuck` goes from `true` to `false`. You could play with it to see if it really works and show the problem.
> 
> _Edit_: I extended the unit test here: [PX4/PX4-ECL#954](https://github.com/PX4/PX4-ECL/pull/954)
Hi,@bresch.
The problem is that when _is_stuck is false,the valid time _time_last_valid_us is always updated,so the condition
(_sample.time_us - _time_last_valid_us) > (uint64_t)10e6) will not be executed,and then _is_stuck  can not be set true.
But when _is_stuck  is true,the timeout condition is correct and will be trig.

bresch, 2021-01-06:
> The problem is that when _is_stuck is false,the valid time _time_last_valid_us is always updated,so the condition
(_sample.time_us - _time_last_valid_us) > (uint64_t)10e6) will not be executed,and then _is_stuck can not be set true.
But when _is_stuck is true,the timeout condition is correct and will be trig.

Yes, but that's fine, we only want to check if the sensor is stuck when the EKF stops fusing the range finder data for some time. As the comment says:
> // Check for "stuck" range finder measurements when range **was not valid for certain period**
// This handles a failure mode observed with some lidar sensors