PX4/PX4-Autopilot #16483: pitch setpoint drops just after the back-transition

yuthikasagarage, 2021-01-04:
**Describe the bug**

there's a sudden drop in the pitch setpoint just after the back transition of the standard vtol in the master. i think the initial setpoint after the back transition is set to zero  just after the back-transition, i just looked at the value of _v_att_sp.pitch_body in the vtol_att_control_main.cpp file in case mode:ROTARY_WING

**To Reproduce**
Steps to reproduce the behavior:
you can load this plan and reproduce the same with standard_vtol
[plan.zip](https://github.com/PX4/PX4-Autopilot/files/5765173/plan.zip)

**Expected behavior**
pitch setpoint should not drop immediately just after the transition, it should come down gradually.

**Log Files and Screenshots**
https://review.px4.io/plot_app?log=19538338-d46f-4c67-a3ca-11e7d1566eb9

Add screenshots to help explain your problem.
[image]

**Additional context**
I think when changing from the flight_mode it zeros the pitch setpoint on full rotary_wing mode.

dagar, 2021-01-04:
It's the re-initialization of full MC mode when coming out of transition. The vehicle still has forward velocity, but it's pitched back. When MC is enabled it's initialized with that velocity and you then get a corresponding attitude setpoint for a multicopter at that velocity (eg pitched forward).

sfuhrer, 2021-01-05:
Would there be an easy way to make the initialization of the MC velocity controller smoother? E.g. limiting the rate of the pitch setpoint change? Or initialize with a 0 velocity setpoint? Sounds a bit hacky to me.. 
Meanwhile, note that in the example here there was a big pitch overshoot (25° instead of 15°) at the end of the back transition, which made the jump in pitch even bigger. 
[image]

bresch, 2021-01-05:
@sfuhrer You could set an initial acceleration feedforward to control the initial pitch setpoint but it might not be easy to find the correct value.